FreeCAD/FreeCAD #11824: [Problem] Sketcher: OVP tools not working when hovering external geometry lines

maxwxyz, 2023-12-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

@PaddleStroke OVP tools (tested with line, circle and rectangle) seem not to work when trying to snap/autoconstrain them to external geometry lines. They snap to an arbitrary position.

Demo with rectangle tool (all OVP enabled):

https://github.com/FreeCAD/FreeCAD/assets/6246609/ca8d11ad-5615-4894-8a00-41d86617f26f

Demo file, see last Sketch001 as an example:

[demo.zip](https://github.com/FreeCAD/FreeCAD/files/13761789/demo.zip)

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35449 (Git)
Build type: Release
Branch: main
Hash: f3fa2ba9c9b1999e5c32d3ae544481b6e17553bb
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.24.9
  * CurvedShapes 1.0.5
  * Curves 0.6.19
  * Defeaturing 1.2.1
  * fasteners 0.4.75
  * FEMbyGEN 2.0.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.3.13
  * woodworking 0.21.1.33694
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-12-24:
@PaddleStroke

maxwxyz, 2023-12-26:
it's still present in yesterdays weekly build but I've narrowed it down: Its only behaving weird if you reference an external line that was created when selecting an external face. This creates a very long line as a reference and then the geometry snaps weirdly. It has nothing to do with @PaddleStroke OVP I guess, as Polyline and BSpline are also snapping wrongly:

https://github.com/FreeCAD/FreeCAD/assets/6246609/f3165e9e-98b9-45b8-8656-8b98d93efb67

[image]
[image]

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35456 (Git)
Build type: Release
Branch: main
Hash: c91a35961f7ef42f1b64eb11ca963d6fbc0b42da
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.24.9
  * CurvedShapes 1.0.5
  * Curves 0.6.19
  * Defeaturing 1.2.1
  * fasteners 0.4.75
  * FEMbyGEN 2.0.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.3.13
  * woodworking 0.21.1.33694
```

PaddleStroke, 2024-01-05:
Ah so I know what is happening. You are snapping to the line center.
As the line is really big (-10000 +10000) the snap zone to snap to the middle of the line is very big. Here on your screen the whole visible line is within the 'snap to middle' zone.

This is because the snap to middle of line is using a % of line length. Maybe 2 or 5%.

Maybe one way to fix this would be to do something like this : 
instead of 
`middleSnapLength = lineLength * 0.05`
We do :
`middleSnapLength = lengthOfLineSegmentVisibleOnScreen * 0.05)`

Now the question is, how do we find lengthOfLineSegmentVisibleOnScreen.

maxwxyz, 2024-01-05:
Usually when it snaps it creates an constraint, right? It snaps in this case but does not create a constraint or dimension so why should it snap there in the first place? Snap to the line creates a point on line constraint which is fine.

PaddleStroke, 2024-01-05:
I've been asked to snap to line middles before.
It's not creating a auto-constraint but it should actually. It should create a symmetric constraint. But no one has done it yet it's an open Feature request.

maxwxyz, 2024-01-09:
@PaddleStroke what I've noticed:

1. Auto snap for the middle of a line is currently present for most tools, but not PolyLine. 
2. It snaps to the middle of lines and arcs
3. In the case of a line, a symmetry constraint could be created --> will this work with your double click to select loop PR, as Symmetry deletes the Point on Line constraint?
4. In case of snap to the middle of an arc, a symmetry constraint is not feasible --> construction geo would be needed.

Should these points be addressed or rather fix the middle snap to 
1. consider visible edge length to get a relative range of the middle depending on the zoom level
2. don't snap to the middle (as no constraints are created)

Separately concerning this issue:
Selecting a face from an external reference shouldn't create a 20.000 long external line but one that is the height of the face?

PaddleStroke, 2024-01-09:
3 Double click only select edges forming a wire. Constraints does not matter, only that the start/endpoint of a edge is at the same position as another point.
4. yes

> Selecting a face from an external reference shouldn't create a 20.000 long external line but one that is the height of the face?

+1 I guess it was hard to find start/end of the line so they used 10 000. With projections it should be doable to proejct the face onto the sketch plane to get lines then check the max / min of those lines.

maxwxyz, 2024-01-09:
3. so in the following case, on the left, all edges would be selected, and on the right only either the closed rectangle or the two lines wire: 
[image]

maxwxyz, 2024-01-14:
https://github.com/FreeCAD/FreeCAD/blob/ce1a43ed2cdc3e7688feecfedd49df26e4234043/src/Mod/Sketcher/App/SketchObject.cpp#L7262C1-L7305C2

maxwxyz, 2024-02-23:
https://github.com/FreeCAD/FreeCAD/issues/10704 related

maxwxyz, 2024-11-29:
Fixed by https://github.com/FreeCAD/FreeCAD/pull/17736

closed 2024-11-29 by maxwxyz