FreeCAD/FreeCAD #11825: [Problem] no automatic quantity conversion: `App::anyToQuantity` does not handle `string` case

Andrei-Pozolotin, 2023-12-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

PROBLEM:

currently it is impossible to handle automatic quantity conversion in property expressions
due to:  `App::anyToQuantity` does not handle `string` case
https://github.com/FreeCAD/FreeCAD/blob/8d51d78ce7e44526be6a84daa6dfa2a62c7a1896/src/App/Expression.cpp#L520

VERIFICATION:

* add new document
* add new object (say spreadsheet)
* add new text property: `App::PropertyString` called  `size_text` = `10 mm`
* add new data property: `App::PropertyLength` called  `size_quantity`
* set property expression for the `size_quatity` = `size_text`
* click "recompute"
* observe error message: `<Exception> Failed to convert to Quantity`

<details>
<summary><b>sample stack trace</b></summary>

```
libFreeCADApp.so!App::anyToQuantity(const App::any & value, const char * msg) (/home/work/source/git/FreeCAD_VS/src/App/Expression.cpp:535)
libFreeCADApp.so!App::isAnyEqual(const App::any & v1, const App::any & v2) (/home/work/source/git/FreeCAD_VS/src/App/Expression.cpp:577)
libFreeCADApp.so!App::PropertyExpressionEngine::execute(App::PropertyExpressionEngine * const this, App::PropertyExpressionEngine::ExecuteOption option, bool * touched) (/home/work/source/git/FreeCAD_VS/src/App/PropertyExpressionEngine.cpp:694)
libFreeCADApp.so!App::Document::_recomputeFeature(App::Document * const this, App::DocumentObject * Feat) (/home/work/source/git/FreeCAD_VS/src/App/Document.cpp:3127)
libFreeCADApp.so!App::Document::recompute(App::Document * const this, const std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > & objs, bool force, bool * hasError, int options) (/home/work/source/git/FreeCAD_VS/src/App/Document.cpp:2856)
libFreeCADApp.so!App::DocumentPy::recompute(App::DocumentPy * const this, PyObject * args) (/home/work/source/git/FreeCAD_VS/src/App/DocumentPyImp.cpp:584)
libFreeCADApp.so!App::DocumentPy::staticCallback_recompute(PyObject * self, PyObject * args) (/home/work/source/git/FreeCAD_VS/build/debug/src/App/DocumentPy.cpp:1949)
libpython3.11.so.1.0!cfunction_call(PyObject * func, PyObject * args, PyObject * kwargs) (/home/user0/.cache/debuginfod_client/fa6a19d79d578b6fa7c904274aae9deec2331d44/source##usr##src##debug##python##Python-3.11.6##Objects##methodobject.c:553)
libpython3.11.so.1.0!_PyObject_MakeTpCall(PyThreadState * tstate, PyObject * callable, PyObject * const * args, Py_ssize_t nargs, PyObject * keywords) (/home/user0/.cache/debuginfod_client/fa6a19d79d578b6fa7c904274aae9deec2331d44/source##usr##src##debug##python##Python-3.11.6##Objects##call.c:214)
libpython3.11.so.1.0!_PyEval_EvalFrameDefault(PyThreadState * tstate, _PyInterpreterFrame * frame, int throwflag) (/home/user0/.cache/debuginfod_client/fa6a19d79d578b6fa7c904274aae9deec2331d44/source##usr##src##debug##python##Python-3.11.6##Python##ceval.c:4760)
libpython3.11.so.1.0!_PyEval_EvalFrame(int throwflag, struct _PyInterpreterFrame * frame, PyThreadState * tstate) (/home/user0/.cache/debuginfod_client/fa6a19d79d578b6fa7c904274aae9deec2331d44/source##usr##src##debug##python##Python-3.11.6##.##Include##internal##pycore_ceval.h:73)
libpython3.11.so.1.0!_PyEval_Vector(PyThreadState * tstate, PyFunctionObject * func, PyObject * locals, PyObject * const * args, size_t argcount, PyObject * kwnames) (/home/user0/.cache/debuginfod_client/fa6a19d79d578b6fa7c904274aae9deec2331d44/source##usr##src##debug##python##Python-3.11.6##Python##ceval.c:6425)
libpython3.11.so.1.0!PyEval_EvalCode(PyObject * co, PyObject * globals, PyObject * locals) (/home/user0/.cache/debuginfod_client/fa6a19d79d578b6fa7c904274aae9deec2331d44/source##usr##src##debug##python##Python-3.11.6##Python##ceval.c:1140)
libpython3.11.so.1.0!run_eval_code_obj(PyThreadState * tstate, PyCodeObject * co, PyObject * globals, PyObject * locals) (/home/user0/.cache/debuginfod_client/fa6a19d79d578b6fa7c904274aae9deec2331d44/source##usr##src##debug##python##Python-3.11.6##Python##pythonrun.c:1710)
libpython3.11.so.1.0!run_mod(mod_ty mod, PyObject * filename, PyObject * globals, PyObject * locals, PyCompilerFlags * flags, PyArena * arena) (/home/user0/.cache/debuginfod_client/fa6a19d79d578b6fa7c904274aae9deec2331d44/source##usr##src##debug##python##Python-3.11.6##Python##pythonrun.c:1731)
libpython3.11.so.1.0!PyRun_StringFlags(const char * str, int start, PyObject * globals, PyObject * locals, PyCompilerFlags * flags) (/home/user0/.cache/debuginfod_client/fa6a19d79d578b6fa7c904274aae9deec2331d44/source##usr##src##debug##python##Python-3.11.6##Python##pythonrun.c:1601)
libFreeCADBase.so!Base::InterpreterSingleton::runString[abi:cxx11](char const*)(Base::InterpreterSingleton * const this, const char * sCmd) (/home/work/source/git/FreeCAD_VS/src/Base/Interpreter.cpp:249)
libFreeCADGui.so!Gui::Command::_runCommand(const char * file, int line, Gui::Command::DoCmd_Type eType, const char * sCmd) (/home/work/source/git/FreeCAD_VS/src/Gui/Command.cpp:725)
libFreeCADGui.so!Gui::Command::_doCommand(const char * file, int line, Gui::Command::DoCmd_Type eType, const char * sCmd) (/home/work/source/git/FreeCAD_VS/src/Gui/Command.cpp:677)
libFreeCADGui.so!StdCmdRefresh::activated(StdCmdRefresh * const this, int iMsg) (/home/work/source/git/FreeCAD_VS/src/Gui/CommandDoc.cpp:1483)
```
</details>

Note: this is a contrived example, actual use case is quantity extraction from complicated `App::PropertyPythonObject`

### Full version info

```shell
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35451 (Git)
Build type: None
Branch: main
Hash: 8d51d78ce7e44526be6a84daa6dfa2a62c7a1896
Python 3.11.6, Qt 5.15.11, Coin 4.0.1, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Andrei-Pozolotin, 2024-01-02:
@Roy-043 @luzpaz : please let me know - will you accept a PR for this issue?  for example:

https://github.com/FreeCAD/FreeCAD/compare/main...random-cadder:FreeCAD:issue_11825

note: there are other places in the code with similar hacks, reflecting a lack of "stand-alone parser", i.e.:
https://github.com/FreeCAD/FreeCAD/blob/ef15fbfea6878ccfb0d83b5b45ba3047f4f98ac1/src/Tools/plugins/widget/customwidgets.cpp#L577

<details>
<summary>
and here is a basic test case:  <code>issue_11825.FCMacro</code>
</summary>

```python
import FreeCAD

print(f"### https://github.com/FreeCAD/FreeCAD/issues/11825 ###")

obj_type = "Spreadsheet::Sheet"
obj_name = "issue 11825"

prop_data_name = "Size_Quantity"
prop_data_type = "App::PropertyLength"

prop_text_name = "Size_String"
prop_text_type = "App::PropertyString"
expr_data_path = f".{prop_data_name}"
expr_data_rule = f".{prop_text_name}"

def verify_parse_operation(prop_text_value:str) -> bool:
    this_doc = FreeCAD.activeDocument()

    try:
        this_obj = this_doc.addObject(obj_type, obj_name)

        this_obj.addProperty(prop_data_type, prop_data_name)
        this_obj.addProperty(prop_text_type, prop_text_name)

        setattr(this_obj, prop_text_name, prop_text_value)
        this_obj.setExpression(expr_data_path, expr_data_rule)

        this_doc.recompute()  # may produce error message on console

        data_value = getattr(this_obj, prop_data_name)
        text_value = getattr(this_obj, prop_text_name)
        data_quant = FreeCAD.Units.parseQuantity(text_value)

        print(f"{data_value=} {text_value=} {data_quant=}")
        return data_value == data_quant

    except Exception as parse_error:

        print(f"{parse_error=}")
        return False

    finally:
        this_doc.removeObject(this_obj.Name)

assert verify_parse_operation("") is False
assert verify_parse_operation("10 mm") is True

```
</details>

Roy-043, 2024-01-03:
I can't make that call, I know nothing about the code involved. Maybe @luzpaz knows who the right person is.

Andrei-Pozolotin, 2024-01-04:
@luzpaz : please let me know - "who the right person is"?

yorikvanhavre, 2024-01-08:
> @Roy-043 @luzpaz : please let me know - will you accept a PR for this issue?

PRs are always welcome! Make sure you reference this issue in it.

At first sight your code at https://github.com/FreeCAD/FreeCAD/compare/main...random-cadder:FreeCAD:issue_11825 seems good, but it's also not my favourite area, maybe @wwmayer or @chennes could have a look?

closed 2024-01-11 by chennes

wwmayer, 2024-02-08:
What is supposed to happen if an invalid quantity is used?
`verify_parse_operation("10 §§§")`

verify_parse_operation correctly returns False but the Size_Quantity property has a value of "10 mm". IMO, this behaviour is inconsistent.

Andrei-Pozolotin, 2024-02-08:
@wwmayer : Werner: thank you for checking this out.

* you are right, and this is pending to be reverted:
https://github.com/FreeCAD/FreeCAD/pull/11937
* as discussed here with @pieterhijma Pieter:
https://github.com/FreeCAD/FreeCAD/pull/11971
* and proposed as "proper solution"  here:
https://github.com/FreeCAD/FreeCAD/pull/12225

If you can, please review or share your opinion on PR https://github.com/FreeCAD/FreeCAD/pull/12225 , thank you.

wwmayer, 2024-02-09:
I haven't yet checked the links you posted.

It seems there is a bug in the quantity parser of the core system.
When I write `App.Units.parseQuantity("1$")` then it correctly raises an exception but when I write `App.Units.parseQuantity("1§")` then it incorrectly returns 1.0.

PR #11937 (2024-01-11): Resolve #11825 - no automatic quantity conversion: App::anyToQuantity
Resolve #11825 - no automatic quantity conversion:` App::anyToQuantity`
commit 6a54591b (2024-01-11): Resolve #11825 - no automatic quantity conversion: `App::anyToQuantity`