FreeCAD/FreeCAD #11826: [Problem] Sketcher: OVP dimension lines and display issues

maxwxyz, 2023-12-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

@PaddleStroke I have an issue with the dimensions in the current weekly build regarding the OVP, not sure if it is also related to the issues https://github.com/FreeCAD/FreeCAD/issues/11603 or https://github.com/FreeCAD/FreeCAD/issues/11824 
- using tools with OVP does not display the correct position of the dimension lines, and dragging them around is flipped / works in the opposite direction
- also the general decimal place setting is used for the first display of the dimensions while the tool is active (I enter 12 and it displays 12,0000) and after finishing the line it is displayed as 12 again

Demo:

https://github.com/FreeCAD/FreeCAD/assets/6246609/f1de29e6-05c1-42b8-a2a3-402b1d0762ee

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35449 (Git)
Build type: Release
Branch: main
Hash: f3fa2ba9c9b1999e5c32d3ae544481b6e17553bb
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.24.9
  * CurvedShapes 1.0.5
  * Curves 0.6.19
  * Defeaturing 1.2.1
  * fasteners 0.4.75
  * FEMbyGEN 2.0.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.3.13
  * woodworking 0.21.1.33694
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-12-26:
Is this still an issue in that case ?

maxwxyz, 2023-12-26:
Still present in yesterdays weekly build 2023-12-25:
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35456 (Git)
Build type: Release
Branch: main
Hash: c91a35961f7ef42f1b64eb11ca963d6fbc0b42da
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.24.9
  * CurvedShapes 1.0.5
  * Curves 0.6.19
  * Defeaturing 1.2.1
  * fasteners 0.4.75
  * FEMbyGEN 2.0.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.3.13
  * woodworking 0.21.1.33694
```

FlachyJoe, 2023-12-28:
Hi,
I'm the culprit and I'm trying to fix this.

PR #11854 (2024-01-05): Sketcher: fix one point distance
fix #11826
closed 2024-01-05 by chennes