FreeCAD/FreeCAD #11830: [Problem] HighDPI on Windows shows small labels on UI elements

danergo, 2023-12-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm using my computer on 175% display scaling.

FreeCAD is mostly rendering correctly, but many UI labels are not scaled correctly:

[image]

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/World (en_001)
Installed mods: 
  * fasteners 0.4.71
  * lattice2 1.0.0
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-12-25:
You created an issue, however, you have forgotten to mention the most important things in your post.

- [ ] What is your screen resolution?
- [ ] How many screens/monitors do you have? What are their parameters?
- [ ] How many video cards do you have?
- [ ] Do you connect displays occasionally or not? (Laptop users tend to connect and disconnect external displays)
- [ ] Does other _Qt_-based software work correctly?

- [ ] Have _FreeCAD_ worked fine before (like, older versions) and suddenly stopped working? Or you installed _FreeCAD_ for the first time (or bought a monitor for the first time)?

----

It's not connected to the topic, but, perhaps, could be valuable in the future

- [ ] What drivers are installed?
- [ ] Does _OpenGL_ software work correctly?

- [ ] How does it work on _Linux_?

danergo, 2023-12-25:
Sorry, I restarted my computer, and it eventually got back to normal. Windows11.

closed 2023-12-25 by danergo