FreeCAD/FreeCAD #11831: Odd filet issue causing problems in 3d print slicer

greggjaskiewicz, 2023-12-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Filet is causing STL to be all mangled , confusing slicers 
[image]
[image]

Basically the last Filet operation, Filet002 is causing odd issue in STLs. It looks all good in FreeCAD. 

See attached files. 
[uszker.zip](https://github.com/FreeCAD/FreeCAD/files/13766125/uszker.zip)
Issue exists both in 0.21.1 and latest nightly build

### Full version info

```shell
[code]
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * AnimationFreeCAD 1.0.0 (Disabled)
  * A2plus 0.4.61
  * sheetmetal 0.3.13
  * freecad_metal_workbench 0.0.1
  * Curves 0.6.19
[/code]
```

### Subproject(s) affected?

File formats

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

greggjaskiewicz, 2023-12-25:
It's actually Fillet001 , not 002

Roy-043, 2023-12-25:
Fillet001 produces an invalid shape. I am not surprised by that. You need to revise you model. I suggest you lower the bottom of the conical shape. You can then make a fillet for tops of the beams and a separate fillet for the bottom of the conical shape. See the attached image (where only 3 edges have been filleted).

This is an [OCC](https://wiki.freecad.org/OpenCASCADE) (FreeCAD's geometry kernel) issue and not something that FreeCAD can fix. I am therefore closing this.

[image]

closed 2023-12-25 by Roy-043