FreeCAD/FreeCAD #11833: Path: Unable to Add ToolBits in ToolBit Library Editor (macOS)

JohnOCFII, 2023-12-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Following the recent Concrete dog Path tutorial at https://blog.freecad.org/2023/11/24/tutorial-getting-started-with-the-path-workbench-part-one/ I was frustrated in the inability to add a tool on my macOS primary system.  I ended up installing the same weekly build on a Debian Linux (Bookworm) installation and did not have issues.

Digging deeper, I ran into a few separate issues.  One is related to tool names translating to OS-agnostic names caused issues with Tool names such as `1/8"` but I created that ticket separately.  Even with very OS-acceptable names, every time I would try to exit from creating a new toolkit I'd receive the following message in the Report View: 
```
13:07:14  /Users/osh/Library/Application Support/FreeCAD/Macro/Bit/7mm_Endmill.fctb
13:07:48  Traceback (most recent call last):
13:07:48    File "/Applications/Other Apps/FreeCAD-35414.app/Contents/Resources/Mod/Path/Path/Tool/Gui/BitEdit.py", line 244, in updateTool
13:07:48      if self.tool.Label != label:
13:07:48  ReferenceError: Cannot access attribute 'Label' of deleted object
```

Digging yet deeper, I'm starting to wonder if the FreeCAD Path Tool Editor needs to be modified to work with the Extended Attribute on more recent versions of macOS.  https://osxdaily.com/2018/05/03/view-remove-extended-attributes-file-mac/

When looking at the directory of files, I do see that each file seems to have the quarantine attribute:

```
(base)osh@OshMBPro Macro % cd Bit
(base)osh@OshMBPro Bit % ls -lta
total 80
-rw-r--r--   1 osh  staff  341 Dec 25 13:07 7mm_Endmill.fctb
drwxr-xr-x  12 osh  staff  384 Dec 25 13:07 .
-rw-rw-r--@  1 osh  staff  337 Dec 25 13:06 5mm_Endmill.fctb
drwxr-xr-x   8 osh  staff  256 Dec 25 13:06 ..
-rw-rw-r--@  1 osh  staff  199 Dec 25 13:06 5mm_Drill.fctb
-rw-rw-r--@  1 osh  staff  242 Dec 25 13:06 6mm_Ball_End.fctb
-rw-rw-r--@  1 osh  staff  276 Dec 25 13:06 6mm_Bullnose.fctb
-rw-rw-r--@  1 osh  staff  316 Dec 25 13:06 60degree_Vbit.fctb
-rw-rw-r--@  1 osh  staff  320 Dec 25 13:06 45degree_chamfer.fctb
-rw-rw-r--@  1 osh  staff  195 Dec 25 13:06 probe.fctb
-rw-rw-r--@  1 osh  staff  307 Dec 25 13:06 slittingsaw.fctb
-rw-rw-r--@  1 osh  staff  292 Dec 25 13:06 5mm-thread-cutter.fctb
(base)osh@OshMBPro Bit %       
(base)osh@OshMBPro Bit % xattr 5mm_Endmill.fctb 
com.apple.quarantine
```

I've attached a screen shot showing that the `7mm_Endmill` IS added, but it can't be selected without error.
[image]

I'm not sure what the resolution is, but I think I've found the root of one of my difficulties using FreeCAD Path.  

Happy to provide any additional information!

### Full version info

```shell
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35414 (Git)
Build type: Release
Branch: main
Hash: 759930e3aab711355d2f0895f641a8657897af48
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```
```

### Subproject(s) affected?

Path

### Anything else?

[BestCylYet.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/13767094/BestCylYet.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

JohnOCFII, 2023-12-26:
FYI - Placing the 3 directories in my /user/Documents/ tree didn't change behavior.

```
18:13:18  /Users/osh/Documents/FreeCAD/PathTools/Bit/7mm_Endmill.fctb
18:14:00  Traceback (most recent call last):
18:14:00    File "/Applications/Other Apps/FreeCAD-35414.app/Contents/Resources/Mod/Path/Path/Tool/Gui/BitEdit.py", line 244, in updateTool
18:14:00      if self.tool.Label != label:
18:14:00  ReferenceError: Cannot access attribute 'Label' of deleted object
```
Also -- manually removing the `quarantine bit` from each file using `xattr -c` in the Shape, Bit, and Library directories did not seem to change behavior.  Perhaps I'm barking up the wrong tree.

JohnOCFII, 2025-02-04:
This remains an issue.

tcurdt, 2025-02-04:
I have my Library outside of Application Support and no problems. Did you try that?
That said Application Support should be fine, too.

If quarantine is removed I don't see how it should be any different.
Did you really get rid of all flags?

JohnOCFII, 2025-02-04:
Huh -- @tcurdt got me to try this again.  And creating tools works fine today.  I'll close this ticket.

closed 2025-02-04 by JohnOCFII