FreeCAD/FreeCAD #11834: [Problem] Draft: error when changing scale of a clone - Pad: gp_GTrsf::Trsf() - non-orthogonal GTrsf

furgo16, 2023-12-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Create a part via a sketch and a pad on the PartDesign workbench
2. Switch workbench to Draft
3. Create a (Draft) clone
4. Select the newly created clone
5. Change its scale property for one of the axis.
6. Stacktrace with error `Pad: gp_GTrsf::Trsf() - non-orthogonal GTrsf`. Change is not applied

[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35456 (Git) Snap 875
Build type: Release
Branch: main
Hash: c91a35961f7ef42f1b64eb11ca963d6fbc0b42da
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * Assembly4 0.50.6
```

### Subproject(s) affected?

Draft

### Anything else?

Notes:
- https://github.com/FreeCAD/FreeCAD/issues/11819 seems to describe a crash when changing scale, but on imported objects.
- https://forum.freecad.org/viewtopic.php?style=1&t=78527 mentions the same error, but I'm unsure if it's related to this issue
- File to reproduce the issue: [Playpen - draft.zip](https://github.com/FreeCAD/FreeCAD/files/13767263/Playpen.-.draft.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-12-25:
This is a duplicate of https://github.com/FreeCAD/FreeCAD/issues/9651 OCC 7.7.x should not be used for production work, OCC 7.6.3 is the recommended version.

Roy-043, 2023-12-28:
Closing this as duplicate.

closed 2023-12-28 by Roy-043