FreeCAD/FreeCAD #11835: crash in the nightly

greggjaskiewicz, 2023-12-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Crash during fillet/chamfer operation, applying changes mentioned in https://github.com/FreeCAD/FreeCAD/issues/11831

[freecad_crash.txt](https://github.com/FreeCAD/FreeCAD/files/13771689/freecad_crash.txt)

### Full version info

```shell
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35449 (Git)
Build type: Release
Branch: main
Hash: f3fa2ba9c9b1999e5c32d3ae544481b6e17553bb
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * AnimationFreeCAD 1.0.0 (Disabled)
  * A2plus 0.4.61
  * sheetmetal 0.3.13
  * freecad_metal_workbench 0.0.1
  * Curves 0.6.19
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-12-28:
Without step-by-step instructions this issue makes no sense. Please provide them.

Roy-043, 2024-01-04:
Since step-by-step instructions are still missing I am closing this.

closed 2024-01-04 by Roy-043

greggjaskiewicz, 2024-01-04:
probably dup of #11836