FreeCAD/FreeCAD #11836: chamfering crash

greggjaskiewicz, 2023-12-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Reproducible, see attached video and stacktrace.
Attaching project that caused it. 

[another_crash.txt](https://github.com/FreeCAD/FreeCAD/files/13771722/another_crash.txt)

https://github.com/FreeCAD/FreeCAD/assets/553583/87c7e4e5-a13c-40b2-aac3-088c6f98d59e

[uszka_maker_26_Dec_2024_13_02.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/13771723/uszka_maker_26_Dec_2024_13_02.FCStd.zip)

### Full version info

```shell
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35449 (Git)
Build type: Release
Branch: main
Hash: f3fa2ba9c9b1999e5c32d3ae544481b6e17553bb
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * AnimationFreeCAD 1.0.0 (Disabled)
  * A2plus 0.4.61
  * sheetmetal 0.3.13
  * freecad_metal_workbench 0.0.1
  * Curves 0.6.19
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

JohnOCFII, 2023-12-27:
Same behavior.  A fillet here will also crash with a segmentation fault.

```
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35456 (Git)
Build type: Release
Branch: main
Hash: c91a35961f7ef42f1b64eb11ca963d6fbc0b42da
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```

Roy-043, 2023-12-28:
This is an issue on Windows as well. No crash on that OS, but an Access violation.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35449 (Git)
Build type: Release
Branch: main
Hash: f3fa2ba9c9b1999e5c32d3ae544481b6e17553bb
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
```

bgbsww, 2023-12-31:
Chamfer crashes on Ubuntu too.  Fillet does not, but logs an exception, likely because of https://github.com/FreeCAD/FreeCAD/blob/a662fbb2ff267fb8dd56fcc3d1ec55a5968b7f5a/src/Mod/Part/App/FeatureFillet.cpp#L54C1-L56C7 which doesn't exist in chamfer.

It looks to me like we've detected the problem https://github.com/FreeCAD/FreeCAD/blob/a662fbb2ff267fb8dd56fcc3d1ec55a5968b7f5a/src/App/DocumentObject.cpp#L152 but we then go on and try to recompute the object anyway, which leads to the crash.

A very specific fix would be done in the execute() methods of Part::Chamfer and Part::Fillet.  It also may be that we should just stop recomputing in DocumentObject as soon as detect the bad link, but that could have pretty widespread impact.

I tried a build with         `return new App::DocumentObjectExecReturn("Invalid links in object");` after the warning at line 153, and it did not crash on the test file, and it did pass all the python and c++ tests, but this needs experienced eyes weighing in to confirm this is the correct approach.

maxwxyz, 2024-10-07:
It works on the stable version so this is a regression.
```
[code]
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.48
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)
[/code]

```
Added blocker label for discussion.

maxwxyz, 2024-10-12:
@wwmayer any idea why this crashes?

wwmayer, 2024-10-12:
It's a bug in OCC:
```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f0a9a1a5520]
#1  0x7f0a83e8bcd4 in BRep_Tool::CurveOnSurface(TopoDS_Edge const&, opencascade::handle<Geom_Surface> const&, TopLoc_Location const&, double&, double&, bool*) from /lib/x86_64-linux-gnu/libTKBRep.so.7+0x184
#2  0x7f0a83e8bde1 in BRep_Tool::CurveOnSurface(TopoDS_Edge const&, TopoDS_Face const&, double&, double&, bool*) from /lib/x86_64-linux-gnu/libTKBRep.so.7+0xb1
#3  0x7f0a83e9b6eb in BRepAdaptor_Curve2d::Initialize(TopoDS_Edge const&, TopoDS_Face const&) from /lib/x86_64-linux-gnu/libTKBRep.so.7+0x9b
#4  0x7f0a6ad62f75 in ChFi3d_Builder::StartSol(opencascade::handle<ChFiDS_Spine> const&, opencascade::handle<BRepAdaptor_HSurface>&, gp_Pnt2d&, opencascade::handle<BRepAdaptor_HCurve2d>&, double&, opencascade::handle<ChFiDS_SurfData> const&, bool, int, opencascade::handle<BRepAdaptor_HSurface>&, opencascade::handle<BRepAdaptor_HCurve2d>&, bool&, bool&, bool&, bool&, opencascade::handle<BRepAdaptor_HSurface>&, gp_Pnt2d&, bool, TopoDS_Vertex const&) const from /lib/x86_64-linux-gnu/libTKFillet.so.7+0xe85
#5  0x7f0a6ad679cc in ChFi3d_Builder::PerformSetOfSurfOnElSpine(opencascade::handle<ChFiDS_HElSpine> const&, opencascade::handle<ChFiDS_Stripe>&, opencascade::handle<BRepTopAdaptor_TopolTool>&, opencascade::handle<BRepTopAdaptor_TopolTool>&, bool) from /lib/x86_64-linux-gnu/libTKFillet.so.7+0x6ec
#6  0x7f0a6ad6fabc in ChFi3d_Builder::PerformSetOfKGen(opencascade::handle<ChFiDS_Stripe>&, bool) from /lib/x86_64-linux-gnu/libTKFillet.so.7+0x10c
#7  0x7f0a6ad72f29 in ChFi3d_Builder::PerformSetOfSurf(opencascade::handle<ChFiDS_Stripe>&, bool) from /lib/x86_64-linux-gnu/libTKFillet.so.7+0x69
#8  0x7f0a6ad3ad95 in ChFi3d_Builder::Compute() from /lib/x86_64-linux-gnu/libTKFillet.so.7+0x3a5
#9  0x7f0a6ae98ceb in BRepFilletAPI_MakeChamfer::Build() from /lib/x86_64-linux-gnu/libTKFillet.so.7+0x2b
#10  0x7f0a78584e62 in PartDesign::Chamfer::execute() from /home/user/Projects/rel-FreeCAD-0-21/build/Mod/PartDesign/_PartDesign.so+0x82e
```

maxwxyz, 2024-10-13:
@chennes so this cannot be a blocker when there is an issue in OCC, right?

pieterhijma, 2025-08-31:
I can reproduce the crash on basically each FreeCAD/OCCT combination, also with OCC 7.6.3, so I checked why @maxwxyz comes to the conclusion that it works in the stable version. So, I used the exact same version as Max:

```
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
```
Although FreeCAD doesn't crash, the operation does not succeed and the following error is thrown:
```
11:29:49  <Exception> Access violation
11:29:49  Recompute failed!
```

I think on Windows, we simply have a handler that catches segfaults. So, although this is a serious issue and indeed one that originates from OCCT, there is still a segfault on Windows but it is handled better in Windows than on Linux. This means that it is not a regression in my opinion. @maxwxyz, could you please remove that label.

ElementW, 2026-08-25:
No longer crashes since 2eb4dc7 / PR #27854 but throws a stacktrace in the message log and report view on `26.3.0dev.20260819` with OCC 7.8.1.

On faf5feae9d2d7b0d9bb81ee9a9fda266ed38ae2c and either OCC 7.9.3 or 8.1.0.dev1-3d097a0328, throws the more usual `BRep_API: command not done` error instead.

@maxwxyz I think we can consider this resolved either way?

closed 2026-08-25 by maxwxyz