FreeCAD/FreeCAD #11839: Path: Erroneous Warning about incorrect Unit Schema for CNC usage (Debian Linux)

JohnOCFII, 2023-12-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Recently I've noticed that when initially launching FreeCAD and then switching to Path workbench on Linux (issue does not occur on macOS), I receive a pop-up dialog about the need to select one of the correct unit schema's for CNC work.  The issue is, I've already selected and saved one of the correct options.  (`Metric, Small parts, & CNC`, in my case).

This does not appear to cause any issues, but wanted to see if others are seeing this behavior too.
[image]

[TestBox.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/13774098/TestBox.FCStd.zip)

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35456 (Git) AppImage
Build type: Release
Branch: main
Hash: c91a35961f7ef42f1b64eb11ca963d6fbc0b42da
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```
```

### Subproject(s) affected?

Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-12-27:
@JohnOCFII Is this the singular setting that doesn't get saved, all your toolbars stay in position etc? Have you tried changing a colour (purely for testing purposes) in Preferences for example in Sketcher and clicking OK. Restart FreeCAD and check the colour has been retained. I use Debian 12 but mine is Python 3.11.x and recently some panels and toolbars have been misbehaving using the same build revision as my Linux Mint 20.3 partition but that has been stable to the best of my observations.

JohnOCFII, 2023-12-28:
> @JohnOCFII Is this the singular setting that doesn't get saved, all your toolbars stay in position etc?

The thing is -- the changed settings DO get saved -- it is whatever check the Path Workbench is doing seems to be failing and not seeing that I've already chosen an acceptable unit schema.

But -- it was a good suggestion - so I did confirm that two other settings in the `General` Preferences were changed and saved successfully.

Syres916, 2023-12-28:
I was wondering why would @JohnOCFII attach a FreeCAD file for a system/platform issue. Then it occurred to me that around the start of October the Units Schema was made document 'aware' sorry can't think of the correct terminology. So the message from Path-InitGui.py is looking at your system Units Schema not the document.

@sliptonic does this check need to be revised to be document level rather than system level or am I completely barking up the wrong tree?

JohnOCFII, 2023-12-28:
> I was wondering why would @JohnOCFII attach a FreeCAD file for a system/platform issue.

:) I mainly added a FreeCAD file, even though the issue isn't file specific, because I've often been asked for one regardless.

Syres916, 2023-12-28:
@JohnOCFII  Could you please post a screenshot of your equivalent for the Path Preferences - Advanced, for both your Mac and your Debian system, this is mine:

[image]

JohnOCFII, 2023-12-28:
Mine both look the same as yours.
[image]

[image]

Syres916, 2023-12-28:
Come on, the one checkbox in both of yours that isn't ticked is the very one that needs to be - _Suppress feed rate warning_ that is the `WarningSuppressVelocity` to stop the dialog box from appearing when the system units don't match one of three that have minutes in them.

JohnOCFII, 2023-12-28:
> Come on, the one checkbox in both of yours that isn't ticked is the very one that needs to be - _Suppress feed rate warning_ that is the `WarningSuppressVelocity` to stop the dialog box from appearing when the system units don't match one of three that have minutes in them.

You are correct.  My setting did NOT match yours.  Good to have a second set of eyes.  Thank you.  But it does not explain why I seem to need that item selected on Debian, but it isn't required on macOS.  On the Mac, FreeCAD seems to correctly see that I have selected an appropriate unit schema, but on Debian, it seems to ignore that correct selection.

Syres916, 2023-12-28:
If you have selected the schema in the bottom right hand corner of the 3D View then as I stated above it is **_document_** level not **_system_** level, I'm really going to have to step back as this is just going round in circles. The code is **not** faulty, I've tested it on my Debian 12.4 box and it works exactly as it should which is at **_system_** level. In some ways I wish that units schema change had been rejected because there has been limited communication and I can see it causing major headaches when the full release is made.

JohnOCFII, 2023-12-28:
> If you have selected the schema in the bottom right hand corner of the 3D View then as I stated above it is **_document_** level not **_system_** level, I'm really going to have to step back as this is just going round in circles.

Thank you for your time.  It definitely helped me.  You were again correct -- for some reason, while the *system* level preferences were correctly set to Small Metric, that particular test document was not.  Once I changed that document-level setting, I stopped seeing the warning.

Weird.  But at this point, I agree, not worth chasing, other than to perhaps modify that warning text to suggest looking at document as well as system preferences.

Again, thank you very much!

[image]

closed 2023-12-28 by JohnOCFII