FreeCAD/FreeCAD #11842: [Problem] Sketcher: moving cursor while renaming constraints exit renaming dialog

maxwxyz, 2023-12-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When renaming constraints (dimensions) in the task dialog list of constraints by right clicking and choosing `Rename`in the context menu, you can rename the constraint. As soon as the cursor is moved 1px the renaming exits so it is not possible to rename the constraints.

Demo:

https://github.com/FreeCAD/FreeCAD/assets/6246609/e84548a6-0580-4dee-8df0-9898e4a6b0bd

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35456 (Git)
Build type: Release
Branch: main
Hash: c91a35961f7ef42f1b64eb11ca963d6fbc0b42da
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.24.9
  * CurvedShapes 1.0.5
  * Curves 0.6.21
  * Defeaturing 1.2.1
  * fasteners 0.4.75
  * FEMbyGEN 2.1.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.3.13
  * woodworking 0.21.1.33694
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rpronto, 2025-03-06:
Hi! Can I work on this issue? It looks interesting. If there are any specific guidelines or if someone is already working on it, please let me know. Thanks!

maxwxyz, 2025-03-06:
Sure, thanks for your help!

maxwxyz, 2025-03-09:
Just tell us if you need any help. You can find resources in the https://freecad.github.io/DevelopersHandbook/gettingstarted/ or the wiki

PR #20458 (2025-03-31): Sketcher: Prevent renaming interruption when hovering task elements.
When renaming a constraint in the task panel, the renaming process was interrupted if the mouse moved over the task elements area. This was caused by two signals being emitted, triggering 'TaskSketcherElements::onListWidgetElementsItemEntered' and 'TaskSketcherElements::onListWidgetElementsMouseMoveOnItem', which changed focus and canceled the renaming.  
The fix adds a check at the beginning of both functions to ensure that if a 'QLineEdit' is focused (indicating an active renaming process), the function exits early, prevent [... cut]
commit 1d4a0936 (2025-03-31): Sketcher: Prevent renaming interruption when hovering task elements. (#20458)

* Sketcher: Prevent renaming interruption when hovering task elements.

Fixes #11842.

* Sketcher: Refactored input focus check based on PR feedback (PR #20458)

* Update src/Mod/Sketcher/Gui/TaskSketcherElements.cpp

---------
closed 2025-03-31 by hyarion