FreeCAD/FreeCAD #11848: [Problem] Dxf Export of Sketch is Not Correct

WandererFan, 2023-12-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This applies to the c++ dxf exporter in the Import module.

When exporting a sketch to dxf, if the sketch is not on the XY plane, the
result is not correct.  Lines and curves that should be connected have gaps.

The exporter code in dxf.cpp (based on HeeksCad exporter) makes assumptions that
the objects to be exported lie on the XY plane.

In order to produce a valid dxf, the sketch should be projected onto its base
plane first.  This is essentially what happens when TechDraw exports a sketch, and
these exports are correct when viewed in LibreCAD.

This issue was first discussed in the Polish forum here: 
https://forum.freecad.org/viewtopic.php?style=4&t=83203
and here:
https://forum.freecad.org/viewtopic.php?style=4&t=82640

The project first approach could be applied to non-sketch objects, but it is 
not clear how to determine a projection direction in this case.

### Full version info

```shell
Any.
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-12-28:
> The exporter code in dxf.cpp (based on HeeksCad exporter) makes assumptions that
> the objects to be exported lie on the XY plane.

but to just clarify, it is documented as so https://wiki.freecad.org/Draft_DXF#C++_exporter though that isn't a reason to not improve it.

kaktusus, 2023-12-28:
If it doesn't work
Isn't it worth giving the case a second look?

Roy-043, 2023-12-28:
Duplicate: #11224
Closing this.

closed 2023-12-28 by Roy-043

kaktusus, 2023-12-28:
Better merge this [old report](url=https://github.com/FreeCAD/FreeCAD/issues/11224) into this one.

Roy-043, 2023-12-28:
I see no reason for that. This report contains no additional information.

kaktusus, 2023-12-28:
But forum posts do. Including the solution.
Also the developer concerned has a good plan to solve the problem. :wink: