FreeCAD/FreeCAD #11861: [Problem] Gui::PrefUnitSpinBox requires restart after unit system change

Roy-043, 2023-12-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After changing the unit system in the preferences FreeCAD must be restarted for Gui::PrefUnitSpinBoxes to display the new units

### Full version info

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35449 (Git)
Build type: Release
Branch: main
Hash: f3fa2ba9c9b1999e5c32d3ae544481b6e17553bb
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-12-31:
I have to say I can't decide whether this is by design from https://github.com/FreeCAD/FreeCAD/pull/9521 or a regression. Maybe @AgCaliva 's input would be of use.

AgCaliva, 2024-01-03:
As far as i remember my code did not required to restart something. Anyway im not contributing again untill XMLReader its kinda deprecated. My suggestions and changes got ignored. I dont think im required.

Roy-043, 2024-01-03:
I am labeling this as a regression.

commit 1dad1b37 (2024-03-22): Fixes #11861: Gui::PrefUnitSpinBox requires restart after unit system change

This fixes a regression of PR #11266 that dropped the case where neither the option 'Ignore project unit system and use the default' is set
nor an active document exists. In this case nothing happens even if the user changed the unit system.
closed 2024-03-22 by wwmayer