FreeCAD/FreeCAD #11862: [Problem] Exporting data from CSV converts US units to mm

0x8000-0000, 2023-12-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In my project I have a spreadsheet containing various dimensions, using "Building US" unit system. Values are expressed in feets, inches and 1/8ths of an inch. When exporting the data, the values are all converted to mm, which is inconvenient.

### Full version info

```shell
[code]
OS: Debian GNU/Linux 12 (bookworm) (LXDE/LXDE)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2025-02-23:
This issue and #11863 should be considered more carefully. I think these are somewhat confusing feature requests. If I set the default unit system to _US customary_, _Spreadsheet_ imports and shows units in the latter system correctly, while if I set to _Standard_, it converts on-the-fly into _Standard_. _Spreadsheet_ always converts units into _Standard_ when exporting a CSV file. I think this behavior is intended, since it is based on the expression mechanism that _Spreadsheet_ merely uses.

I think this and #11863 should be closed in favor to #16861 or rephrased into something more solid, for example, “_Spreadsheet_ must not convert units into _Standard unit system_ when exporting a CSV file”, but why? This is intended by design.

FEA-eng, 2026-10-16:
Seems to be a feature request, actually.