FreeCAD/FreeCAD #11864: [Problem] FreeCAD.exe Application Error 0x64effee6

kwgagel, 2023-12-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD randomly crashes, sometimes it happens multiple times in a row before it is stable again. Windows Event Viewer Application records Event 1000, Application Error with these details:
`Faulting application name: FreeCAD.exe, version: 0.21.1.0, time stamp: 0x64effee6
Faulting module name: CoreMessaging.dll, version: 10.0.22621.2506, time stamp: 0x11fe18fb
Exception code: 0xe0464645
Fault offset: 0x0000000000083ea9
Faulting process id: 0x0x33CC
Faulting application start time: 0x0x1DA3C32DF589427
Faulting application path: C:\Program Files\FreeCAD 0.21\bin\FreeCAD.exe
Faulting module path: C:\WINDOWS\SYSTEM32\CoreMessaging.dll
Report Id: 1a93bb21-0649-4c0e-adaf-0f36cf901d27
Faulting package full name: 
Faulting package-relative application ID: `

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Anything else?

I am just learning how to use it and following some YouTube tutorials. I'm not doing anything complicated. I'll keep trying to identify steps that trigger the crash, but I haven't seen a pattern of the cause so far. While I have a mouse and a touchpad, I only use the mouse.

My Computer
`Processor	Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz   3.70 GHz
Installed RAM	64.0 GB
Product ID	00330-81478-63673-AA867
System type	64-bit operating system, x64-based processor
Pen and touch	Pen support
`

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-12-31:
Have you got a newish NVidia card by any chance, if so, the version of Qt (even a more recent one) is not liking the combination of hardware/driver etc? Windows 10 with exactly the same hardware is a high chance of being fine. An alternative is to use WSL2 and use the Linux version of FreeCAD in that as a workaround.

kwgagel, 2023-12-31:
Here is the NVIDIA Information:
[NVIDIA System Information 12-31-2023 15-54-40.txt](https://github.com/FreeCAD/FreeCAD/files/13802382/NVIDIA.System.Information.12-31-2023.15-54-40.txt)

Syres916, 2024-01-01:
Not come across too many users with a Quadro K2200 so the best I can advise is (with FreeCAD closed) find the file `opengl32sw.dll` in C:\Program Files\FreeCAD 0.21\bin and rename it to `opengl32.dll`. This has the effect of making the OpenGL software based rendering rather than hardware. There will be a minor performance impact but it should be more stable.

I personally use an old GeForce GTX670 using hardware acceleration and it's worked right from FreeCAD version 0.16 through to the latest daily build on both Windows and Linux but it was luck rather than judgement.

adrianinsaval, 2024-04-11:
are you still experiencing this with the latest weekly build?

luzpaz, 2024-07-26:
@kwgagel :point_up:

kwgagel, 2024-07-26:
I've hardly been on the computer lately. I'm not likely to get back to this until September.

luzpaz, 2024-07-27:
Note: follow up in September with kwgagel

luzpaz, 2024-09-10:
@Palingenesis can you install the development build and see if you reproduce ?  
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Palingenesis, 2024-09-10:
haha, sorry about the laugh.  
I have just posted on the forum this: [https://forum.freecad.org/viewtopic.php?t=90493](https://forum.freecad.org/viewtopic.php?t=90493)  
I mention that I once tried the weekly and had difficulty going back to a project with an earlier version.
I would like to use the weeklies, there are some nice improvements, but I would like to know if it going to fix things first.
It was early in the year when I last tried and it did not fix things then.
This is why I asked.

I could try a stand alone instance, but I find it gets worse the more it accesses the cache on larger projects.
So I would have to make a new large project.
I put as much as I can on my E drive, which helps, but the FreeCAD cache is still on C.
Personally I think it is windows 11 new security features about access, but I probably talking out of my arse with that.

Give me a few days, I will see if I can try a weekly without messing up may current setup.

Palingenesis, 2024-09-10:
Thank you, I will check that out, The latest weekly failed, same issue.
I ran it stand alone-ish, it still uses user info run from a folder.
I copied a project I had to a new folder so my original remains untouched, and worked on it.
Just had a thought, I use the link to my standard parts, perhaps most people don't use.
Any ways I don't do the whole project in one file most times parts are created in separate files and are linked in another file.
So I may have a file that has a part drawn in it and it has screws linked from my parts folder. then I may have a file that has the assembly where the part is put together using links to all the parts in there files.
I mention this because a lot of time it crashes is when editing the part names in the link of the part. That's why I mentioned cache before.
One other reason for mentioning it is:
First thing I did is make a copy of a bigger project to test the candidate release, the copy when loaded came up with a relative (sort of) path for a part it could not find. That is it was looking for a part link relative to the new copy, not an absolute link which gives the full path to the actual (true) file location.
This could be due to it been a copy of a project and the error got in with the previous release, I don't know.

The only way to fully test it is to start from scratch with a new project.
This will take time, so fingers crossed, I will come back if there is still issues.
I do believe there is some issue between cache and file paths with links, some times I will have to load the part files before the assembly file if I don't do a restore and do a clean instead.

I hope that made some sense, I have been chasing my tail for a while now and I tend to get mixed up.
Thanks for the link.
Tim

luzpaz, 2024-09-10:
What assembly workbench do you use ?

luzpaz, 2024-09-10:
Also any chance you can zip up some of these files and drag/drop them into github here for testers ?

Palingenesis, 2024-09-11:
I don't use an assembly work bench to assemble a project.
I use Part Design Workbench mainly, unless I need to do something it doesn't do.

This is how it usually goes:
Part "A" file => Create Part "A" => Create Body.
Assembly File => Link to Part "A".
Part "B" file => Create Part "B" => Create Body.
Assembly File => Link to Part "B".
They are links to parts in the assembly file so that they can be edited in the assembly file.
The file I have linked will show what I do.

I need more time with the release candidate for FreeCAD 1.0. It has not crashed yet, I will use it on a copy of my current project and see how it goes.
My current project could do with a lot of cleaning up so I will see how it handles that.

As for FreeCAD_weekly-builds-38643-conda-Windows-x86_64-py311 it was a fail, still the same issue.
CoreMessaging.dll
I used a smaller project for this, one that I don't mind sharing.

It usually shuts down when I change link names, in the Model Tree. Although the application could be just sat there one minute and gone the next.
The project I was testing the weekly on I have zipped. But I can only attach images here. I have put it on my server.
[TIM02.zip](https://www.timsnet.co.uk/stuff/TIM02.zip)

Palingenesis, 2024-09-14:
Update, The [the-first-release-candidate-of-freecad-1](https://blog.freecad.org/2024/09/10/the-first-release-candidate-of-freecad-1-0-is-out/) also fails the same.  
**Take this with a pinch of salt.** (I could be talking out of my Ars.)
Windows 10 had no issues.
I think the issue is Windows 11, with all it's security upgrades cause, assess problems to files.
If anyone with windows 11 wants to test the file(s) I uploaded, the test would be to change all the names of the links.
If enough people do the test it would help swing the path in the correct direction.

It's a shame I could not get FreeCAD to run inside Visual Studio 2022, may be I could have been more help.
I can run it in the Linux emulator OK without issue in Windows 11, this is another reason I think it is a file access issue.

luzpaz, 2024-09-14:
@chennes can you test on windows 11 ? https://github.com/FreeCAD/FreeCAD/issues/11864#issuecomment-2343554769

Palingenesis, 2024-09-14:
This is my problem, ever since I upgrades to Windows 11, FreeCAD  crashes.
Moving as much as I could from C drive to my second drive seemed to make it more stable.
I also use KiCAD. I had to do the same with that. KiCAD also had access violations when it was on C drive.
I had no issues when I was running Windows 10.

luzpaz, 2024-09-14:
> Moving as much as I could from C drive to my second drive seemed to make it more stable.
I also use KiCAD. I had to do the same with that. KiCAD also had access violations when it was on C drive.

adrianinsaval, 2024-09-14:
that sounds like your system is damaged rather than the applications. I can run freecad with no obvious issues on windows 11. You posted a bunch of files, which in particular is giving you trouble and doing what? I tried some for couple minutes and did not see a crash.

> OS: Windows 11 build 22621
> Word size of FreeCAD: 64-bit
> Version: 1.0.0RC1.38642 (Git)
> Build type: Release
> Branch: (HEAD detached at 1.0rc1)
> Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
> Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
> Locale: C/Default (C) [ OS: Spanish/Paraguay (es_PY) ]
> Installed mods: 
>   * AirPlaneDesign
>   * CfdOF 1.25.1
>   * CfdOF.backup1708023105.1879435 1.21.7 (Disabled)
>   * CurvedShapes 1.0.0
>   * Curves 0.6.14
>   * DynamicData 2.34.0
>   * fasteners 0.5.10
>   * FCGear 1.0.0
>   * FC_SU
>   * IconThemes
>   * OpenTheme 2024.9.1
>   * Plot 2022.4.17
>   * SelectorToolbar
>   * sheetmetal 0.4.1

Palingenesis, 2024-09-15:
> that sounds like your system (or your hard rive) is damaged rather than the applications. I can run freecad with no obvious issues on windows 11. You posted a bunch of files, which in particular is giving you trouble and doing what? I tried some for couple minutes and did not see a crash.
> 

When you tried the project files I posted. Did you edit the linked models from the main file (_New Track.FCStd)?
Alter the names? change some sketches?
If I load the model I posted and do not edit, there is no issue.
I can also create a new part with no issue.
Editing linked models after is the issue.

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * kicadStepUpMod 10.23.1

If I have have hard drive issues (SSD), should I not get issues with other programs on my machine?
My be I should do a video?

Palingenesis, 2024-09-16:
Did a Video.
The first part of the video in compatibility mode, is long, to show it is stable in compatibility mode.
Go close to the end of the video to see what happens when I take it out of compatibility mode.
No hard drive issues.
I hope to enjoy FreeCAD again one day when someone figures it out. My other option is to go back to Windows 10.
[Showing my FreeCAD issue](https://www.youtube.com/watch?v=GT-nwwyvLJM&t=21s)

Stimmenhotel, 2024-09-20:
I have the same problem (I think).

I have installed the latest stable and replaced it with the dev version.
Last crash happened right after starting, going to "2D-Draft" from the "New File" Panel and starting a new line.
Trying it a second time, no crash.

This happens very randomly, and also only for me, my colleagues have no problem, all on Windows 11.
The Laptops are managed by cooperation, if this makes a difference.

Sometimes I feel like working slower will help not to crash. Btw, I am no professional, so all my models are very basic. Also it will crash right after drawing the first lines.

I am using a NVIDIA GPU too.
Specs:
Lenovo P1 Gen3
Intel I7-10850H
16GB RAM
NVIDEA Quadro T2000 with Max-Q Design (acc. to `dxdiag`), 4GB VRAM
=> Driver Version 538.18 (just checked v550 may be latest)

Freecad Versions Info:
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git)
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.47

From windows event histroy:
AppName FreeCAD.exe 
  AppVersion 1.1.0.0 
  AppTimeStamp 66e49e04 
  ModuleName CoreMessaging.dll 
  ModuleVersion 10.0.22621.3672 
  ModuleTimeStamp 41404c38 
  ExceptionCode e0464645 
  FaultingOffset 000000000008405d 
  ProcessId 0x478 
  ProcessCreationTime 0x1db0b2e4fd41943 
  AppPath C:\Users\XXXXXX\AppData\Local\Programs\FreeCAD 0.21\bin\FreeCAD.exe 
  ModulePath C:\WINDOWS\SYSTEM32\CoreMessaging.dll 
  IntegratorReportId 8792ea75-d28e-4050-891c-01600dd1929f 
  PackageFullName  
  PackageRelativeAppId

Stimmenhotel, 2024-09-20:
Just updated GPU drivers to v553 and FreeCAD to latest dev, no problems so far.
Until now no other crashes. Don't mean it is fixed, but because of the approaching weekend, I wanted to update this.
I will keep an eye on the problem, but I am also not often using the tool.

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git)
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.47

Palingenesis, 2024-09-20:
>@[Stimmenhotel](https://github.com/Stimmenhotel)
> "Btw, I am no professional,"
>
>Just updated GPU drivers to v553 and FreeCAD to latest dev, no problems so far. Until now no other crashes. Don't mean it is fixed, but because of the approaching weekend, I wanted to update this. I will keep an eye on the problem, but I am also not often using the tool.

Just wondering how well you know/use FreeCAD?
The time it usually crashes with me is: When I am using "Links" to other models in other files.
Editing names and paths to the file I am linking to.
If I am just making a single model in one file, it is fairly stable.

CarloSartori, 2024-11-18:
I have the same problem of the first post
here the informazion of my feecad

`OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39176 (Git) Conda
Build type: Release
Branch: main
Hash: 45d3864942c6928cec7dc18f95b1f222485756e5
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * CADExchanger
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * offline-documentation 1.0.0-alpha
  * Render 2024.9.25
  * sheetmetal 0.4.26
`

my graphic card is Nvidia T1200 Laptop GPU

here the error I get on windows on the last crash
`
Nome dell'applicazione che ha generato l'errore: freecad.exe, versione: 1.1.0.0, timestamp: 0x6733aa94
Nome del modulo che ha generato l'errore: CoreMessaging.dll, versione: 10.0.22621.4391, timestamp: 0x373137e2
Codice eccezione: 0xe0464645
Offset errore 0x000000000008405d
ID processo che ha generato l'errore: 0x0x1130
Ora di avvio dell'applicazione che ha generato l'errore: 0x0x1DB39A5E0E3AA41
Percorso dell'applicazione che ha generato l'errore: C:\Program Files\FreeCAD 1.0 RC\bin\freecad.exe
Percorso del modulo che ha generato l'errore: C:\WINDOWS\SYSTEM32\CoreMessaging.dll
ID segnalazione: 2d18f464-4bf3-4d8f-9163-15cfc5ac82b0
Nome completo pacchetto che ha generato l'errore: 
ID applicazione relativo al pacchetto che ha generato l'errore: 
`

I really hope you can solve this issue. The random crash is a nightmare

Thank you
Carlo

Palingenesis, 2025-02-17:
I still have had the same issue until two days ago.
Two days ago I swapped out my graphics card for my previous graphics card.
I have been running FreeCAD 1.0.0 since the swap and has not crashed.
I have put the issue down to the graphics card I had installed.
For me this issue seems to be resolved.

nt4boy, 2025-03-04:
I'm new to Freecad, and have an upto date new Desktop with an Nvidia RTX2000.
Same issue still persists for me, and yes, It drive you mad to simply disappear from the screen, just before you remember to save...

Faulting application name: freecad.exe, version: 1.0.0.0, time stamp: 0x673bb0ab
Faulting module name: CoreMessaging.dll, version: 10.0.26100.3194, time stamp: 0x7f002d46
Exception code: 0xe0464645
Fault offset: 0x000000000002ae53
Faulting process id: 0x8538
Faulting application start time: 0x1DB8D1C547D5255
Faulting application path: C:\Program Files\FreeCAD 1.0\bin\freecad.exe
Faulting module path: C:\WINDOWS\SYSTEM32\CoreMessaging.dll
Report Id: c8ff4354-5b41-4c2e-97c8-d3a8d04c6305
Faulting package full name: 
Faulting package-relative application ID: 

Fault bucket 1702813254681423060, type 4
Event Name: APPCRASH
Response: Not available
Cab Id: 0

Problem signature:
P1: freecad.exe
P2: 1.0.0.0
P3: 673bb0ab
P4: CoreMessaging.dll
P5: 10.0.26100.3194
P6: 7f002d46
P7: e0464645
P8: 000000000002ae53
P9: 
P10: 

Attached files:
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.31637114-696b-4a92-88f4-4fb3fa2107dc.tmp.dmp
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.f939e3dc-c537-4277-942f-4f0504b479d7.tmp.WERInternalMetadata.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.6241f1f3-ac5e-4d49-9d25-e05c4c607801.tmp.csv
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.175fed8c-5a45-490b-abef-b7b20e7bb176.tmp.txt
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.953ffb33-b835-4b4f-97c9-22716675edbe.tmp.xml

These files may be available here:
\\?\C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_freecad.exe_5a7d3be4a632905148caa2ae2ca096ecd75bfb2_e14d5101_31c435a8-588b-472c-8f1e-a046fbf41986

Analysis symbol: 
Rechecking for solution: 0
Report Id: c8ff4354-5b41-4c2e-97c8-d3a8d04c6305
Report Status: 268435456
Hashed bucket: 5ceee44b4fcef1ab67a19ba21c1cecd4
Cab Guid: 0

Could get the dump files if useful to anyone.

nt4boy, 2025-03-04:
The Nvidia driver was dated 7/24, so put in one dated 25/2/25 and so far so good.

Palingenesis, 2025-03-04:
I forgot to say what my Graphic cards are.
Titan X Pascal = FreeCAD Crashes.
GTX 1080 = OK, no issues with FreeCAD.

I don't know why, but I am happy now with the GTX 1080.
This is with the latest FreeCAD v1.0

CarloSartori, 2025-06-10:
I'm not sure if I already posted this before, but I solved the issue by checking this option:
"Disabilita ottimizzazioni schermo intero" on "compatibilità" on the property of the freecad.exe file 
Sorry for the Italian but I think the english version is "Fullscreen optimization uncheck" but I'm not sure.
Hope this help

closed 2025-08-25 by github-actions[bot]