FreeCAD/FreeCAD #11865: [Problem] Roundtrip of setting the background colour of a spreadsheet's cell raises `TypeError`

sphh, 2024-01-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you use the following code on a spreadsheet cell, which has the default background colour 
```python3
spreadsheet.setBackground('A1', spreadsheet.getBackground('A1'))
```
raises
```python3
TypeError: Tuple required.
```

This happens, because `spreadsheet.getBackground(cell)` returns `None` for a cell, which background colour is the default background. It would be good, if `spreadsheet.setBackground(cell, None)` resets the cell's background to the default colour.

### Full version info

```shell
[code]
OS: Linux Mint 21.2 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * DynamicData 2.54.0
  * workbench_central_configuration
[/code]
```

### Subproject(s) affected?

Spreadsheet

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-09:
@sphh can you please test with the latest weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2025-03-18:
@sphh can you test on 1.1dev ?

sphh, 2025-03-19:
I would love to check it, but honestly this happened such a long time ago, that I have no clear idea anymore, how I came across this bug. Would need a bit of time to figure this out …