FreeCAD/FreeCAD #11867: [Problem] Setting the precision of `Gui::QuantitySpinBox` has no effect

sphh, 2024-01-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

For setting an integer value in a `QuantitySpinBox` I want to remove all digits after the decimal point with the following Python code:
```python3
spinbox = FreeCADGui.UiLoader().createWidget('Gui::QuantitySpinBox')
spinbox.setProperty('rawValue', 1)
spinbox.setProperty('decimals', 0)
```
But this still displays `1.00` in the spinbox, as it is done without setting the `decimals` property at all.

### Full version info

```shell
[code]
OS: Linux Mint 21.2 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * DynamicData 2.54.0
  * workbench_central_configuration
[/code]
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-01-01:
Gui::QuantitySpinBox are floats so of course they use the global decimals value, you should use a standard DoubleSpinBox which you can specify the decimals and it doesn't take the FreeCAD global setting. Whether I'd class this as a problem, no, a user education issue to be honest.

sphh, 2024-01-01:
I can see your point and this is certainly a workaround.

But if I want to set the number of decimal values to let's say to 6:
```python3
spinbox.setProperty('decimals', 0)
```
The value in the spin box is still displayed as `1.00` and not `1.000000`. Is this the indented behaviour? Does the `Gui::QuantitySpinBox` _always_ use the global decimals value? Why? Why can I then set the number of digits after the decimal point? Shouldn't that throw an error, if it does not have any effect?

Syres916, 2024-01-01:
That is the intended behaviour of Gui::QuantitySpinBox, I'm not here to answer the why's as I'm not a decision maker. If you want it changing then you are welcome to raise a Pull Request explaining the reasons and impact assessment on existing UX and code and the maintainers will either accept, advise an alternative or reject it based on the information you provide.

sphh, 2024-01-01:
As I said, I do not object this behaviour. I only said, that setting the number of digits after the decimal point should raise an error (with a sensible explanation), if it is ignored. Personally I consider this a bug/problem.

I am able to raise an exception in Python code but the `Gui::QuantitySpinBox` is written in C++, so a PR with that effect is unfortunately out of my scope …

sphh, 2024-09-12:
Thanks @PaddleStroke, that is very constructive remark!

I did a very quick and superficial check and they use a `Gui::PrefSpinBox` which again is derived from the standard QSpinBox` (and `PrefWidget`).

I will try it, whenever I find time to dig out that old code of mine.

Still I believe, that an exception should be thrown and it should be mentioned in the documentation, that setting the number of decimals does not have any effect, so I leave this open to be addressed.

PaddleStroke, 2024-09-12:
I have just checked (I have trouble with decimals as well).
And we actually have a "setDecimals" function in QuantitySpinbox that we introduced with Abdullah when we worked on the sketcher's widget. But it's only in cpp. I don't think we have a python binding for this.

closed 2025-06-24 by github-actions[bot]

luzpaz, 2025-06-24:
@sphh any idea if this improved in 1.0.x or 1.1.x ?

luzpaz, 2025-07-21:
@sphh ping

FEA-eng, 2025-09-20:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```