FreeCAD/FreeCAD #11873: [Problem] During C++ DXF import, the Group Layers option is not well handled

kpemartin, 2024-01-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

During C++ DXF import, if "group layers" is selected, the shape contents of each layer are combined into a single shape, which is named using the layer name, and placed as a top-level object in the document. Non-shape contents (text, dimensions) are placed as individual objects at the top level since they can't be merged.

If the layer contains text etc. both the text and the merged shape should be placed in a Layer in the document. Only if there is just a single merged shape would that shape appear as a top-level object, duplicating existing behaviour.

If this option is not selected, such objects remain as individual un-merged shapes, but are still placed at the root level of the document. They should be placed in an appropriate Layer within the document, along with any text and dimensions within that layer.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35389 +97 (Git)
Build type: Debug
Branch: Issue11394
Hash: fcbd1105d64a945f242b8356fd72119f388b4747
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Canada (en_CA)
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kpemartin, 2024-01-01:
There are three options here whose interaction is unclear:
Join geometry
Group layers into blocks
Use layers

In particular "Group layers into blocks" uses the confusing term "block" and also just seems to be a combination of Join geometry and Use layers.

I feel this is one option too many. Things would be better served by:
`Preserve Layers` tells the importer to keep objects in the layers as expressed in the DXF file; otherwise layer structure will be discarded (though layer assignment will still be used for resolution of BYLAYER attributes)
`Join geometry` tells the importer to combine objects like lines, arcs, etc. into compound objects while still preserving their individual attributes (colour, line type, layer assignment if `Preserve layers` is selected)

| Preserve Layers | Join geometry | Use colors | Shapes | Placement | Colour |
|---|---|---|---|---|---|
| off | off | off | Remain as separate objects | Top-level drawing objects | Default colour |
| off | off | on | Remain as separate objects | Top-level drawing objects | From DXF data |
| off | on | off | Merged by line type | Top-level drawing objects | Default colour |
| off | on | on | Merged by line type and colour | Top-level drawing objects | From DXF data |
| on | off | off | Remain as separate objects | Layer in drawing | Default colour |
| on | off | on | Remain as separate objects | Layer in drawing | From DXF data |
| on | on | off | Merged by line type and layer | Layer in drawing if more than one merged object, otherwise single top-level object named using layer name  | Default colour |
| on | on | on | Merged by line type, colour, and layer | Layer in drawing if more than one merged object, otherwise single top-level object named using layer name | From DXF data |

yorikvanhavre, 2024-01-08:
Excellent plan! I agree layers containing texts should be handled much better. Grouping everything in a layer is a good idea indeed.

Another possible option, but slower, could be to offer to create shapestring objects for texts, so they can be compounded together with the all-in-one shape. But for documents with a large number of texts, this might be very slow

commit 0dcfe945 (2024-02-06): Handle all combos of "group into blocks" "use DXF colors" "use layers"
Fixes #11873 this was the primary goal of these changes

Fixes (partially) #11874 the parts of a polyline are combined as a
compound object (shape) but it would be preferable for them to be made
into a wire (if possible)

Fixes #11872 Objects in a block definition are now kept separately
until the block is inserted, in which case the inserted objects are
subject to all the other options regarding combining.

Fixes (partially, review required) #11871 Text and dimensions are now
kept as part of the block definition and are pl [... cut]