FreeCAD/FreeCAD #11875: [Problem] C++ DXF Import without merging loses object type

kpemartin, 2024-01-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When the C++ DXF importer is used without the "merge layers" option, the individual entites still import as "shapes" rather than retaining their identities as lines, arcs, etc.

This interferes with further programmatic manipulation of the drawing, as well as the information shown in the Properties panel.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35389 +97 (Git)
Build type: Debug
Branch: Issue11394
Hash: fcbd1105d64a945f242b8356fd72119f388b4747
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Canada (en_CA)
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-01-08:
The reason this was not done before is to gain speed, since this would mean creating Draft objects. However, the Part module now has several native 2D object types (lines, rectangles, etc..)... That might be an interesting option here, the performance would be very similar to dumb shapes I believe

kpemartin, 2024-01-08:
There is an option to import as "draft objects/simple part shapes/sketches" which could be used to control this. Still won't do "sketches" though.

yorikvanhavre, 2024-01-08:
Indeed importing DXF objects as sketches would be something totally different. Not sure how interesting that is, though, as it would add a lot of unnecessary functionality, since DXF files have no constraints anyway.

But here too there is also a Draft tool to convert to sketch, that could be used after import...

kpemartin, 2024-01-08:
I don't really have any use for sketches either. I'm not really sure what they do that is useful that nothing else does. One of my early impressions with FreeCAD was that there seemed to be several ways to do essentially the same thing. It took me a while to figure out the "workbenches" concept too, right now my mental model of these is that they are different subsets of the large selection of commands available.

yorikvanhavre, 2024-01-12:
workbenches are really a GUI concept. When working directly with code, they make less sense indeed, you could simply see them as modules or libraries

luzpaz, 2024-03-15:
What's the status here ?

kpemartin, 2024-03-15:
This is still partially at the design stage, in terms of the interaction of the various import options, and I have to look into whether the interpretation of the options I would be taking is compatible with how the legacy importer views the options.

The import options have a three-way (radio button) selection of "simple shapes", "draft objects" or "sketch objects" but the new importer treats "simple shapes" and "draft objects" the same, in both cases using draft objects only when necessary (text and dimensions) so the way to select whether you get the functionality suggested in this issue would be to select "draft objects", in which case the code would make draft objects preferentially.

The problem (in my mind) is that the "merge shapes" and "join geometry" options, which are separate checkboxes but can only apply to "shapes" and not Draft or Sketch objects, are really just a continuation of this radio button set:
Sketch Objects
Draft Objects
(individual) simple shapes
joined shapes (i.e. join them if they connect end-to-end)
merged shapes (merge all mergeable shapes into a single shape)

Changing the options form to show things this way would make more of the options mutually exclusive and I want to review the legacy importer to ensure it does not try to do something original when more that one of these is specified.

Actually implementing the creation of Draft objects is just grunt work of adding the code to invoke the various Python Draft.make* methods, and perhaps a few function signature changes. This could be done without changing the options but then the code has to guard against nonsense option combinations that would be forbidden by the modified radio button set. Although given that the settings can be code-generated nothing actually prevents nonsense combinations so these have to be handled gracefully anyway...

Unfortunately it is tax time again, and I'm also looking for a new job, so I don't really have time to look at this just now. I do not have any code changes that would be disrupted by other commits on the master branch so I have no risk of a merge from hell.

kpemartin, 2025-01-22:
This appears to be somewhat of a duplicate of #16543

kpemartin, 2025-02-04:
I've got much of this done.
I still have to see if the Draft mesh type is appropriate for importing a 3D mesh "polyline". The legacy importer seems to have code to do this but the C++ importer does not handle these entities properly at all. The case of the Poly-face mesh also has to be considered but I doubt there is a Draft object for this.

I'm struggling with the SPLINE entities. The simple-objects import uses Geom_BSplineCurve if the entity has control points (I suspect it always does), otherwise Geom_Interpolate to make a Bézier curve if the entity has fit points (it doesn't always).

There do not seem to be Draft objects that do either of these reliably. The Draft BSpline object takes fit points but does not have support for start and end tangents, and fit points are not always available in DXF. The Draft BezCurve object takes its control points as groups of (degree) points each defining a Bézier segment but with corners at the join points. This does not produce at all the same shape as the NURBS (B-Spline) curve in the original application.

The legacy importer does not seem to handle SPLINE entities at all. I may just leave these as the non-Draft objects because there is insufficient support for general BSplines (uniform or otherwise). Proper support would require `ControlPoints`, `Weights`, `Poles`, and `Degree` properties in `Draft.BSpline`. Setting these would make `Points` (the fit points) unavailable (`None` perhaps) since the curve is not longer what was fitted to the points.

kpemartin, 2025-06-09:
This is still in my worklist and associated with other work.

furgo16, 2025-08-11:
Indeed, I believe most of this should have been addressed by https://github.com/FreeCAD/FreeCAD/pull/22251 - I didn't mark it as closing this issue as I wasn't sure about its scope, and also because the PR did not handle splines.