FreeCAD/FreeCAD #11879: [Problem] Viewing sketches in different files at the same time

jes, 2024-01-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I often want to look at a sketch in a different file, to take reference dimensions from it to apply to a sketch that I'm making. I can switch between files either with the tabs at the bottom of the 3D area, or with both open at the same time with "Windows -> Tile".

But when I open a sketch in another file, it closes the sketch that was open in the first file, which means I can't look at my reference sketch while working on my new sketch.

### Full version info

```shell
[code]
OS: Ubuntu 20.04.5 LTS (Regolith:GNOME-Flashback:GNOME/regolith)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Curves 0.4.2
  * Render 2022.4.0
[/code]
```

### Subproject(s) affected?

None

### Anything else?

I found this forum thread from 2014: https://forum.freecad.org/viewtopic.php?t=5583 which says this is infeasible - but this is about *editing* 2 sketches at the same time. I just want the sketch in the other file to remain *visible* while I work on my sketch.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-01-02:
_If_ you must have two sketches open at the same time then you need to run two instances of FreeCAD open with a file open in each. There is no reason for having a read-only Sketcher mode so viewing is editing.

Now if you want to use a proper modelling workflow then you need to educate yourself in the _MasterSketch_ workflow which uses _Sketch-CarbonCopy_ at it's heart and can use Link a sketch in a different file (though I personally never use Link). Naming constraints with useful terms is fundamental to not having any need to open a Sketch as the names are available in the Properties tab and in the Auto complete of Expressions.

I strongly suggest joining the Forum if you haven't already as it's a much more responsive place for user education questions.

maxwxyz, 2024-01-05:
Maybe duplicate of https://github.com/FreeCAD/FreeCAD/issues/5580

jes, 2024-01-05:
It's not exactly a duplicate of #5580 because I don't want to edit them at the same time, I just want to be able to look at one while I edit the other.

pierreporte, 2024-01-05:
It is possible to copy a sketch from a file to an other file. You can then use it as reference with `Sketcher_External` ore modify it.

maxwxyz, 2024-01-06:
I guess there is currently no difference between editing and viewing a sketch as the sketch is active at both times. Currently only one sketch can be active.

pierreporte, 2025-03-02:
I think that #11914 supersedes this issue.

closed 2025-03-02 by maxwxyz