FreeCAD/FreeCAD #11880: [Problem] The file dialog is very slow

jes, 2024-01-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Whenever I need to use a file dialog (e.g. open/save/import/export) in FreeCAD, it seems to take a few seconds to load in the contents of every directory I browse through. It sometimes loads in half the files instantly and then after a little while it updates and all the rest appears.

This does not occur in any other application that I use.

### Full version info

```shell
[code]
OS: Ubuntu 20.04.5 LTS (Regolith:GNOME-Flashback:GNOME/regolith)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Curves 0.4.2
  * Render 2022.4.0
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-01-02:
All I can suggest is implementing the Dialog/DontUseNativeDialog (boolean) parameter as per https://wiki.freecad.org/Fine-tuning

furgo16, 2024-03-15:
I've noticed this happening on folders that have a huge number of files which are supported by FreeCAD. By default, at least the `File > Open` dialog's filter is preset to open all supported file formats. My guess is that it is taking it a while to go through all the files, although I don't quite understand why, since it would seem it's a simple filtering process.

Setting the filter in the dialog to `FCStd` noticeably speeds up the process.

jes, 2024-03-15:
Thanks but it is slow for me even on directories that don't have very many files in.

luzpaz, 2024-03-15:
> All I can suggest is implementing the Dialog/DontUseNativeDialog (boolean) parameter as per https://wiki.freecad.org/Fine-tuning

@jes did you try this workaround ?

jes, 2024-03-15:
It didn't make any difference at all, to the extent that I wasn't sure it did anything.

lninio, 2024-07-23:
Hi! I have this problem too. Also FreeCAD freezes at the stage of opening the Open dialog boxes / Save / Save as / Import / Export. When I run it as a superuser (sudo), it works fine. 'Save a file' operation may take a long time as 20 minutes, because any click in dialog take about 3-4 minutes.
[image]
When I click in FreeCAD window somewhere, the a have this message. Label is "FreeCAD does not respond".

ldeninski, 2024-10-25:
@lninio, @jes I had the same issue after some fiddling with strace I isolated my problem to (probably) preview generation. The file dialog tries to open every file in the directory you are browsing to, or in case of new dialog to your home folder. If there is a .json file in the directory the dialog freezes. In the strace log I can see that freecad opens every file in the folder and reads some part of it. In the case of json files it just opens reads and then freezes. The workaround is to remove.. when possible all json files from your home folder and to avoid browsing into directories that contain one as freecad will freeze again if it sees one. I am not sure at which developer I can direct this findings. If anyone needs a strace log I can add one.

ldeninski, 2024-10-25:
It is possible this is related to #10099 and #5853

ldeninski, 2024-10-28:
@adrianinsaval here's the info from about

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Defeaturing 1.2.2
  * 3D_Printing_Tools
  * Design456 0.0.1
  * MeshRemodel 1.10.32
  * A2plus 0.4.68
  * Assembly4 0.50.15
  * Behave-Dark-Colors 0.1.1
  * Curves 0.6.50
  * DynamicData 2.68.0
  * fasteners 0.5.29
  * Manipulator 1.5.7
  * ProDarkThemePreferencePack 1.92.0
  * freecad.gears 1.0.0
```

Toggling `DontUseNativeDialog` have no visual effect for me and freecad freezes when seeing json file with both `true` and `false`.

tillfoo, 2024-12-17:
same here with FreeCAD 1.0 on Manjaro KDE.
`DontUseNativeDialog` `true`/`false` doesn't help.
Removing .json files from the relevant directories works as a workaround.

tarman3, 2024-12-17:
Can not reproduce

<details>
  <summary>FreeCAD 1.0.0.39109 (Git) makepkg, Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods:lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.1; Silk 0.1.5; Curves 0.6.51

```
</details>

<details>
  <summary>FreeCAD 1.1.0dev.39673 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.2; Silk 0.1.5; Curves 0.6.52

```
</details>

closed 2025-06-24 by github-actions[bot]

bigi111, 2026-09-21:
A Solution :
The 21.09.2026, I had the same problem of slow opening the "Open" dialog box. I had this problem in the past already. I solved this problem, without being sure which of my actions solved the problem. The problem was on version 1.1.3 of FreeCAD
I use FreeCAD on Linux Mint 22.3 with Appimage.
1) I executed the older versions 1.1.0. It can be downloaded from :
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
2) Probably the result was to clear the cache : $HOME/.cache/FreeCAD
3) If this folder is not empty, delete all files it contains.
For me, this seems to have solved the problem of slow opening dialog boxes.