FreeCAD/FreeCAD #11881: Pocket Text in ShapeString via sketch on Surface on a curved surface.

Kenfg, 2024-01-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Freecad problem.
This from help about.

In the Curves workbench
with Sketch on Surface  Property window in the Settings,
the Settings options the Thickness dropdown can be set to a positive number to Emboss or Pad the text, set from the ShapeString. But if I set a negative Thickness it does not work to produce a pocket of the text.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
Installed mods: 
  * Curves 0.6.16
  * Help 1.0.3
```

### Subproject(s) affected?

None

### Anything else?

Using a negative number of mm to on a curve, to create a Pocket does not work.
Padding the text on a curve works great.
 I could not find a related issue.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-01-02:
Maybe related to https://github.com/tomate44/CurvesWB

Kenfg, 2024-01-03:
Not being a coder I cant say it is not releted. My understanding of that section is it is not about padding or pocketing Text.

Syres916, 2024-01-03:
@Kenfg did you read the Wiki entry regarding offset https://wiki.freecad.org/Curves_SketchOnSurface ? This is an external Addon and while a lot of the code is very useful it's not the core of FreeCAD therefore you shouldn't log an issue here. What you should do is request Help from the forum first and you would be directed to raise an issue at the appropriate place if actually needed.

Roy-043, 2024-01-03:
Negative Thickness works fine. But you need to cut the generated shape from the shape the sketch is mapped onto.
[map-sketch-cut.zip](https://github.com/FreeCAD/FreeCAD/files/13817913/map-sketch-cut.zip)

Roy-043, 2024-01-03:
Closing this as a user error.

closed 2024-01-03 by Roy-043

Kenfg, 2024-01-04:
Yes by cutting the Sketch on Surface I can get the text to work on a negative thickness.
This seems like a very confusing step compaired to the ease with which the thickness works for a positive number.

Roy-043, 2024-01-04:
You misunderstand how things work here. Both a positive and a negative Thickness create a new separate shape. Your confusion stems from the fact that the shape produced by a negative value is hidden by the shape the sketch is mapped onto.

For the future: please consult the forum before creating a GitHub issue.