FreeCAD/FreeCAD #11888: [Problem] crash when adding eclipse in sketcher

greggjaskiewicz, 2024-01-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When adding ellipse in sketch , sometimes nightly/dev version of FreeCAD crashes. Other times it gives Nan constraint values. See video for way to reproduce. 
Simply said, in sketcher , click on "create conic", select eclipse by centre, start drawing the ellipse - and it does sometimes crash . 

[freecad_crash_2.txt](https://github.com/FreeCAD/FreeCAD/files/13820881/freecad_crash_2.txt)

https://github.com/FreeCAD/FreeCAD/assets/553583/33369516-2a40-4f52-93bb-305e559cb057

### Full version info

```shell
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35485 (Git)
Build type: Release
Branch: main
Hash: a662fbb2ff267fb8dd56fcc3d1ec55a5968b7f5a
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * AnimationFreeCAD 1.0.0 (Disabled)
  * A2plus 0.4.61
  * sheetmetal 0.3.13
  * freecad_metal_workbench 0.0.1
  * Curves 0.6.19
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-01-04:
I tried to reproduce and so far, I succeeded in making FreeCAD produce crashlessly this message:

```
Sketcher constraint number 2 is malformed!
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'IndexError'>: Constraint has invalid indexes
Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Constraint has invalid indexes
Sketcher constraint number 2 is malformed!
Sketcher constraint number 3 is malformed!
Traceback (most recent call last):
  File "<string>", line 4, in <module>
<class 'IndexError'>: The constraint has invalid index information and is malformed.
Unhandled exception in ViewProvider::eventCallback: The constraint has invalid index information and is malformed.
(Event type: SoMouseButtonEvent, object type: SketcherGui::ViewProviderSketch)
Sketcher constraint number 2 is malformed!
Sketcher constraint number 3 is malformed!
Sketcher constraint number 4 is malformed!
Traceback (most recent call last):
  File "<string>", line 5, in <module>
<class 'IndexError'>: The constraint has invalid index information and is malformed.
Unhandled exception in ViewProvider::eventCallback: The constraint has invalid index information and is malformed.
(Event type: SoMouseButtonEvent, object type: SketcherGui::ViewProviderSketch)
```

The message is shown if I press the left mouse button motionlessly or move the mouse cursor vertically.

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35394 (Git)
Build type: Debug
Branch: iss_11521
Hash: eaf1f88d6360d1f00c882787432c233e053a70fe
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

xtemp09, 2024-03-09:
I ought to add ― the crash report applied has nothing to do with the video. In the crash report, the user is doing chamfer. In the video you are creating an ellipse.

Even dates ― the report was created on 2023-12-26, while the video recorded on 2024-01-03.

luzpaz, 2024-07-26:
Please advise how to proceed

greggjaskiewicz, 2024-09-12:
no crashes, but it still does not always work - as shown here. 

https://github.com/user-attachments/assets/086d476c-654c-4c5a-ba0a-65adca23848b

The version is latest weekly:
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38684 (Git)
Build type: Release
Branch: main
Hash: 92e31fa7ea974fe48c372f6749916d68f42dcf95
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: //
Installed mods: 
  * AnimationFreeCAD 1.0.0
  * A2plus 0.4.62
  * sheetmetal 0.4.0
  * freecad_metal_workbench 0.0.1
  * Curves 0.6.23

greggjaskiewicz, 2024-09-12:
and the file: 
[ex16.FCStd.zip](https://github.com/user-attachments/files/16985422/ex16.FCStd.zip)

tuckerkevin, 2024-10-27:
I get a crash every time (MacOS Arm64) by creating a new file, part design body, new sketch, create ellipse by center... first click for the ellipse center on the sketch origin, next click on the sketch x axis, them move up to start giving the ellipse height and crash.

xtemp09, 2024-10-27:
@tuckerkevin, I cannot reproduce. Can you test the latest weekly, upload the crash report and _About_ info?

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39059 (Git)
Build type: Debug
Branch: master
Hash: 529779fb53085ad28bc58ece27078f0fe1a2f79d
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

```ruby
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

JohnOCFII, 2024-10-27:
> I get a crash every time (MacOS Arm64) by creating a new file, part design body, new sketch, create ellipse by center... first click for the ellipse center on the sketch origin, next click on the sketch x axis, them move up to start giving the ellipse height and crash.

Please post the full information about what version of FreeCAD you are using.  Copy and paste information from `FreeCAD menu --> About --> Copy to clipboard` and paste here.

Also, if you could include a sample file showing the problem (zipped up), that would be helpful to troubleshoot.

tuckerkevin, 2024-10-27:
@JohnOCFII 

OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
  * sheetmetal 0.5.2

There is no sample file needed, I get the crash adding a single ellipse as described in my first post as the first object to a new sketch in a new file, every time.

tuckerkevin, 2024-10-27:
@xtemp09 

Here are the relevant portions of the crash report I think.

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [79137]
Path:                  /Applications/FreeCAD.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]

OS Version:            macOS 15.0.1 (24A348)
Report Version:        12

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   freecad [79137]

Application Specific Information:
abort() called

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x18687e600 __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x1868b6f70 pthread_kill + 288
2   libsystem_c.dylib             	       0x1867c3908 abort + 128
3   libc++abi.dylib               	       0x18686d44c abort_message + 132
4   libc++abi.dylib               	       0x18685ba40 demangling_terminate_handler() + 348
5   libobjc.A.dylib               	       0x1865013f4 _objc_terminate() + 172
6   libc++abi.dylib               	       0x18686c710 std::__terminate(void (*)()) + 16
7   libc++abi.dylib               	       0x18686fcdc __cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*) + 88
8   libc++abi.dylib               	       0x18686fc84 __cxa_throw + 92
9   SketcherGui.so                	       0x16ad0456c SketcherGui::DrawSketchDefaultHandler<SketcherGui::DrawSketchHandlerEllipse, SketcherGui::StateMachines::ThreeSeekEnd, 3, SketcherGui::ConstructionMethods::CircleEllipseConstructionMethod>::getPointInfo(Sketcher::GeoElementId const&) + 520
10  SketcherGui.so                	       0x16ad03c7c SketcherGui::DrawSketchDefaultWidgetController<SketcherGui::DrawSketchHandlerEllipse, SketcherGui::StateMachines::ThreeSeekEnd, 3, SketcherGui::OnViewParameters<5, 6>, SketcherGui::WidgetParameters<0, 0>, SketcherGui::WidgetCheckboxes<0, 0>, SketcherGui::WidgetComboboxes<1, 1>, SketcherGui::ConstructionMethods::CircleEllipseConstructionMethod, true>::addConstraints() + 1196
11  SketcherGui.so                	       0x16ad52100 SketcherGui::DrawSketchDefaultHandler<SketcherGui::DrawSketchHandlerEllipse, SketcherGui::StateMachines::ThreeSeekEnd, 3, SketcherGui::ConstructionMethods::CircleEllipseConstructionMethod>::finish() + 104
12  SketcherGui.so                	       0x16ad51940 non-virtual thunk to SketcherGui::DrawSketchControllableHandler<SketcherGui::DrawSketchDefaultWidgetController<SketcherGui::DrawSketchHandlerEllipse, SketcherGui::StateMachines::ThreeSeekEnd, 3, SketcherGui::OnViewParameters<5, 6>, SketcherGui::WidgetParameters<0, 0>, SketcherGui::WidgetCheckboxes<0, 0>, SketcherGui::WidgetComboboxes<1, 1>, SketcherGui::ConstructionMethods::CircleEllipseConstructionMethod, true>>::onModeChanged() + 60
13  SketcherGui.so                	       0x16ad52fb8 SketcherGui::DrawSketchController<SketcherGui::DrawSketchHandlerEllipse, SketcherGui::StateMachines::ThreeSeekEnd, 3, SketcherGui::OnViewParameters<5, 6>, SketcherGui::ConstructionMethods::CircleEllipseConstructionMethod>::onViewValueChanged(int, double) + 232
14  libQt5Core.5.15.13.dylib      	       0x102959e84 0x102814000 + 1334916
15  libQt5Core.5.15.13.dylib      	       0x102959e84 0x102814000 + 1334916
16  libFreeCADGui.dylib           	       0x1032aedec Gui::QuantitySpinBox::validateInput() + 116
17  libFreeCADGui.dylib           	       0x1032b2310 Gui::QuantitySpinBox::focusOutEvent(QFocusEvent*) + 52
18  libQt5Widgets.5.15.13.dylib   	       0x10206f758 QWidget::event(QEvent*) + 104
19  libQt5Core.5.15.13.dylib      	       0x10293d094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
20  libQt5Core.5.15.13.dylib      	       0x10293d094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
21  libQt5Core.5.15.13.dylib      	       0x10293d094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
22  CoreFoundation                	       0x18699e3a8 __CFRunLoopDoSource0 + 176
23  CoreFoundation                	       0x18699e10c __CFRunLoopDoSources0 + 244
24  CoreFoundation                	       0x18699ccf8 __CFRunLoopRun + 840
25  CoreFoundation                	       0x18699c334 CFRunLoopRunSpecific + 572
26  HIToolbox                     	       0x191dd50cc RunCurrentEventLoopInMode + 292
27  HIToolbox                     	       0x191ddad1c ReceiveNextEventCommon + 220
28  HIToolbox                     	       0x191ddb020 _BlockUntilNextEventMatchingListInModeWithFilter + 76
29  AppKit                        	       0x18a4e0a70 _DPSNextEvent + 660
30  AppKit                        	       0x18ae067b8 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
31  AppKit                        	       0x18a4d3b7c -[NSApplication run] + 480
32  libqcocoa.dylib               	       0x1071784d0 0x10714c000 + 181456
33  freecad                       	       0x100933184 main + 5128
34  dyld                          	       0x186534274 start + 2840

tuckerkevin, 2024-10-27:
I've seen some other pretty weird behavior with centered ellipse but a little harder to describe.

There is one other confusing thing that is more of a improvement suggestion not a bug is that with either ellipse tool once you set the first dimension if you don't stay within that dimension while dragging out your second dimension your outline will disappear. In face the outline disappears after you set the first dimension until you move the cursor back in bounds. This could be improved IMO (by not having the outline disappear no matter where your cursor is with respect to the first set dimension).

JohnOCFII, 2024-10-28:
> I get a crash every time (MacOS Arm64) by creating a new file, part design body, new sketch, create ellipse by center... first click for the ellipse center on the sketch origin, next click on the sketch x axis, them move up to start giving the ellipse height and crash.

Confirmed crash.

```
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.50
 ```

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [35609]
Path:                  /Applications/Other Apps/FreeCAD-39057.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-10-27 18:47:08.1671 -0500
OS Version:            macOS 15.0.1 (24A348)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       073530D2-4024-4B3C-BDD3-794BF72D4FF5

Time Awake Since Boot: 420000 seconds
Time Since Wake:       13108 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   freecad [35609]

Application Specific Information:
abort() called

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x18632e600 __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x186366f70 pthread_kill + 288
2   libsystem_c.dylib             	       0x186273908 abort + 128
3   libc++abi.dylib               	       0x18631d44c abort_message + 132
4   libc++abi.dylib               	       0x18630ba40 demangling_terminate_handler() + 348
5   libobjc.A.dylib               	       0x185fb13f4 _objc_terminate() + 172
6   libc++abi.dylib               	       0x18631c710 std::__terminate(void (*)()) + 16
7   libc++abi.dylib               	       0x18631fcdc __cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*) + 88
8   libc++abi.dylib               	       0x18631fc84 __cxa_throw + 92
9   SketcherGui.so                	       0x308fb056c SketcherGui::DrawSketchDefaultHandler<SketcherGui::DrawSketchHandlerEllipse, SketcherGui::StateMachines::ThreeSeekEnd, 3, SketcherGui::ConstructionMethods::CircleEllipseConstructionMethod>::getPointInfo(Sketcher::GeoElementId const&) + 520
10  SketcherGui.so                	       0x308fafc7c SketcherGui::DrawSketchDefaultWidgetController<SketcherGui::DrawSketchHandlerEllipse, SketcherGui::StateMachines::ThreeSeekEnd, 3, SketcherGui::OnViewParameters<5, 6>, SketcherGui::WidgetParameters<0, 0>, SketcherGui::WidgetCheckboxes<0, 0>, SketcherGui::WidgetComboboxes<1, 1>, SketcherGui::ConstructionMethods::CircleEllipseConstructionMethod, true>::addConstraints() + 1196
11  SketcherGui.so                	       0x308ffe100 SketcherGui::DrawSketchDefaultHandler<SketcherGui::DrawSketchHandlerEllipse, SketcherGui::StateMachines::ThreeSeekEnd, 3, SketcherGui::ConstructionMethods::CircleEllipseConstructionMethod>::finish() + 104
12  SketcherGui.so                	       0x308ffd940 non-virtual thunk to SketcherGui::DrawSketchControllableHandler<SketcherGui::DrawSketchDefaultWidgetController<SketcherGui::DrawSketchHandlerEllipse, SketcherGui::StateMachines::ThreeSeekEnd, 3, SketcherGui::OnViewParameters<5, 6>, SketcherGui::WidgetParameters<0, 0>, SketcherGui::WidgetCheckboxes<0, 0>, SketcherGui::WidgetComboboxes<1, 1>, SketcherGui::ConstructionMethods::CircleEllipseConstructionMethod, true>>::onModeChanged() + 60
13  SketcherGui.so                	       0x308ffefb8 SketcherGui::DrawSketchController<SketcherGui::DrawSketchHandlerEllipse, SketcherGui::StateMachines::ThreeSeekEnd, 3, SketcherGui::OnViewParameters<5, 6>, SketcherGui::ConstructionMethods::CircleEllipseConstructionMethod>::onViewValueChanged(int, double) + 232
14  libQt5Core.5.15.13.dylib      	       0x102edde84 0x102d98000 + 1334916
15  libQt5Core.5.15.13.dylib      	       0x102edde84 0x102d98000 + 1334916
16  libFreeCADGui.dylib           	       0x103832dec Gui::QuantitySpinBox::validateInput() + 116
17  libFreeCADGui.dylib           	       0x103836310 Gui::QuantitySpinBox::focusOutEvent(QFocusEvent*) + 52
18  libQt5Widgets.5.15.13.dylib   	       0x1025f3758 QWidget::event(QEvent*) + 104
19  libQt5Core.5.15.13.dylib      	       0x102ec1094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
20  libQt5Core.5.15.13.dylib      	       0x102ec1094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
21  libQt5Core.5.15.13.dylib      	       0x102ec1094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
22  CoreFoundation                	       0x18644e3a8 __CFRunLoopDoSource0 + 176
23  CoreFoundation                	       0x18644e10c __CFRunLoopDoSources0 + 244
24  CoreFoundation                	       0x18644ccf8 __CFRunLoopRun + 840
25  CoreFoundation                	       0x18644c334 CFRunLoopRunSpecific + 572
[... cut ...]
29  AppKit                        	       0x189f90a70 _DPSNextEvent + 660
30  AppKit                        	       0x18a8b67b8 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
31  AppKit                        	       0x189f83b7c -[NSApplication run] + 480
32  libqcocoa.dylib               	       0x1077fc4d0 0x1077d0000 + 181456
33  freecad                       	       0x100eb7184 main + 5128
34  dyld                          	       0x185fe4274 start + 2840
```

maxwxyz, 2025-03-09:
Is this still an issue?

greggjaskiewicz, 2025-03-09:
can't recreate it.

closed 2025-03-09 by maxwxyz

tuckerkevin, 2025-03-09:
I can not recreate the crash on build 40504 either, I guess it's fixed.