FreeCAD/FreeCAD #11889: [Problem] techdraw - balloon page objects outside on treeview

bjnobre, 2024-01-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Why is the ballons objects stays outside its owner pages?

[image]

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Portuguese/Brazil (pt_BR)
Installed mods: 
  * Assembly4 0.50.2
  * fasteners 0.4.62
  * sheetmetal 0.2.63
[/code]
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-01-03:
Sometimes the ViewProvider does not call claimChildren() right away.   Eventually the Balloon should move itself under the Page.

WandererFan, 2024-01-15:
should be fixed by #11999.

closed 2024-02-13 by bjnobre