FreeCAD/FreeCAD #11893: [Feature Request] Sketcher: Allow Equal Constraint for Dimensions (Angles and Point Distances)

maxwxyz, 2024-01-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently the tool makes two selected lines (geometry) equal, e.g. equal length. 
It would be great if the equal constraint can be applied to dimensions as well, to make them equal
For example:
- angles
- point to point dimensions / distances

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-01-04:
Isn’t this what construction lines are for?

maxwxyz, 2024-01-04:
Construction lines are just a workaround, same as expressions.

pierreporte, 2024-01-04:
I don’t think it is a workaround, or more precisely they can be used in this manner for legitimate reasons. However I think that your feature request could be extended. I assume that you are willing to be able to edit both dimensions, so that it updates the other one. It could be useful to set a ratio between the dimensions, equals to 1 by default. You would then be able to have a dimensions which is always, say half the value of an other one. It works perfectly with expressions but they wouldn’t let you to edit the driven dimension, only the master one.

maxwxyz, 2024-01-04:
Sounds interesting, so the idea is to create a relational constraint, default to 1 to be equal?

pierreporte, 2024-01-04:
Yes, a value of 1 means equality.

FlachyJoe, 2024-01-06:
IMHO  it would be better to have a expression editor which support click on a dimension to reference it.

maxwxyz, 2024-01-08:
also fine with this idea, this would solve this issue and kind of replaces the constrain equal I guess.

pierreporte, 2024-08-30:
What I proposed is basically #8676. @maxwxyz what do you want to do?

maxwxyz, 2024-08-30:
@pierreporte yes ratio 1 is practically the same.

closed 2024-08-30 by maxwxyz