FreeCAD/FreeCAD #11894: freecad-0.21.2: fails to compile: moc_actionbox.cpp_parameters: No such file or directory[Problem]

AndrewAmmerlaan, 2024-01-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Version 0.21.2 fails to compile for me on Gentoo Linux, this version compiled successfully before but a rebuild triggered by an upgrade of boost and several other dependencies introduces this issue. Likely the issue is caused by some incompatibility with a new version of one of the dependencies, though downgrading boost did not help.

Any help in uncovering which change is responsible for this failure would be appreciated. The issue was originally reported here: https://bugs.gentoo.org/921298

```
[3096/4996] cd /tmp/portage/media-gfx/freecad-0.21.2/work/FreeCAD-0.21.2_build/src/Gui/QSint/actionpanel && @/tmp/portage/media-gfx/freecad-0.21.2/work/FreeCAD-0.21.2_build/src/Gui/QSint/actionpanel/moc_actionbox.cpp_parameters
FAILED: src/Gui/QSint/actionpanel/moc_actionbox.cpp /tmp/portage/media-gfx/freecad-0.21.2/work/FreeCAD-0.21.2_build/src/Gui/QSint/actionpanel/moc_actionbox.cpp
cd /tmp/portage/media-gfx/freecad-0.21.2/work/FreeCAD-0.21.2_build/src/Gui/QSint/actionpanel && @/tmp/portage/media-gfx/freecad-0.21.2/work/FreeCAD-0.21.2_build/src/Gui/QSint/actionpanel/moc_actionbox.cpp_parameters
/bin/sh: line 1: @/tmp/portage/media-gfx/freecad-0.21.2/work/FreeCAD-0.21.2_build/src/Gui/QSint/actionpanel/moc_actionbox.cpp_parameters: No such file or directory
```

[build.log.gz](https://github.com/FreeCAD/FreeCAD/files/13830035/build.log.gz)

### Full version info

```shell
[code]
OS: Gentoo Linux (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.21.2.Unknown
Build type: RelWithDebInfo
Python 3.11.7, Qt 5.15.11, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.0
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AndrewAmmerlaan, 2024-01-05:
The problem seems to be cmake not setting some variables somehow, fixed downstream by explicitly setting:
```
		-DQt5Core_MOC_EXECUTABLE="$(qt5_get_bindir)/moc"
		-DQt5Core_RCC_EXECUTABLE="$(qt5_get_bindir)/rcc"
```

I don't quite understand why it worked before but now needs this to be explicitly set, perhaps something changed with the upgrade from qt 5.15.11 to 5.15.12.

ionenwks, 2024-01-05:
Haven't looked too closely but maybe it just wasn't using moc and new Qt or something else made it start to, and then `cMake/FreeCAD_Helpers/SetupQt.cmake` sets `QtCore_MOC_EXECUTABLE` to an empty string (only with Qt5) as far as I can tell.

closed 2025-06-24 by github-actions[bot]