FreeCAD/FreeCAD #11895: [Feature Request] Support for Elmer Static Current Conduction solver in the FEM workbench

FEA-eng, 2024-01-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, FreeCAD FEM supports Elmer electrostatic and magnetodynamic 2D/3D solvers. One major missing Elmer electromagnetic solver is Static Current Conduction (StatCurrentSolve) described on page 113 of this manual: https://www.nic.funet.fi/pub/sci/physics/elmer/doc/ElmerModelsManual.pdf

This solver can be used to perform DC conduction (current flow) simulations with Joule heating which are commonly requested from users.

There was an attempt to implement this solver but unfortunately, the project was abandoned: https://forum.freecad.org/viewtopic.php?t=75888

New forum discussion: https://forum.freecad.org/viewtopic.php?t=84051

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35485 (Git)
Build type: Release
Branch: main
Hash: a662fbb2ff267fb8dd56fcc3d1ec55a5968b7f5a
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
Installed mods: 
  * A2plus 0.4.60n
  * CfdOF 1.24.3
  * CurvedShapes 1.0.4
  * Curves 0.6.13
  * Defeaturing 1.2.0
  * dodo 1.0.0
  * fasteners 0.4.59
  * freecad.gears 1.0.0
  * Manipulator 1.5.0
  * sheetmetal 0.2.63
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Gspohu, 2024-01-04:
I was the one who posted the message on the forum.
I'll see if I can integrate StatCurrentSolve with FreeCAD.
If all goes well, a PR will follow in the next weeks

HoWilgh, 2025-01-10:
Please also have a look at https://forum.freecad.org/viewtopic.php?t=48867&hilit=elmer+current .

PR #19835 (2025-02-28): Fem: Add support for Elmer static current solver - fixes #11895
Add support for Elmer static current solver.

A new property `NormalCurrentDensity` is added to the CurrentDensity boundary condition.
By setting property `HeatSource` to `True` this solver can work in combination with the heat equation using the Joule heating as heat source.

Fixes #11895

@FEA-eng
commit 1c91ba9a (2025-02-28): Fem: Add support for Elmer static current solver - fixes #11895
closed 2025-02-28 by maxwxyz