FreeCAD/FreeCAD #11899: [Problem] Fully constraint sketch allows movement

greggjaskiewicz, 2024-01-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Sketcher solver seems to be confused by the model I'm working with. See the video and sample project. 
Basically when I placed extra lines - they can be moved around, despite solver claiming that this sketch is fully constrained. 
Adding extra constraints to actually fully constrain it - is then met with "over constraint" message. 
See the video for reproduction. Weird one.

[ex31_sketcher_waaaa.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/13834384/ex31_sketcher_waaaa.FCStd.zip)

https://github.com/FreeCAD/FreeCAD/assets/553583/e59a86fe-3311-4be7-bbc0-fd96d824910d

### Full version info

```shell
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35485 (Git)
Build type: Release
Branch: main
Hash: a662fbb2ff267fb8dd56fcc3d1ec55a5968b7f5a
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * AnimationFreeCAD 1.0.0 (Disabled)
  * A2plus 0.4.61
  * sheetmetal 0.3.13
  * freecad_metal_workbench 0.0.1
  * Curves 0.6.19
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Jim-Holmstroem, 2024-09-19:
Not exactly the same, but probably very related and with a bit simpler reproduction.

[bug1.zip](https://github.com/user-attachments/files/17058961/bug1.zip)

It runs on a fairly recent version
```
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: makepkg
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: C/Default (C)
Installed mods: 
  * DynamicData 2.53.0
  * Assembly4 0.50.4
  * fasteners 0.4.67
```

- Open file and open Sketcher
- **redundant constraints message**
- Drag endpoints of arc and release
- **endpoints move + fully constrained message**
- Try to drag endpoints of arc and release
- **endpoints does not move + redundant constraints message**
- Repeats redundant constraints -> fully constrained on continued endpoint drags

[image]

FEA-eng, 2026-10-15:
[This](https://github.com/FreeCAD/FreeCAD/issues/11899#issuecomment-3345275566) is still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 27.1.0dev.20261013 (Git shallow)
Build date: 2026/10/13 19:55:30
Build type: Release
Branch: (HEAD detached at 50daee777)
Hash: 50daee777321b5085accd1f57b57f1a7c1781cae
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```