FreeCAD/FreeCAD #11900: [Problem] CSG import fails for a pyramid

burunduk3, 2024-01-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The following CSG file triggers the issue when tried to be imported:
```openscad
minkowski(convexity = 0) {
        sphere($fn = 100, $fa = 12, $fs = 2, r = 1);
        cylinder($fn = 4, $fa = 12, $fs = 2, h = 2, r1 = 2, r2 = 0, center = false);
}
```
_Put it into `test.csg` and run `freecad test.csg` to reproduce._

The UI shows no geometry and there is an error in the output:
```
OpenSCAD file contains both 2D and 3D shapes. That is not supported in this importer, all shapes must have the same dimensionality.
Traceback (most recent call last):
  File "…/FreeCAD/Mod/OpenSCAD/OpenSCADFeatures.py", line 598, in execute
    raise ValueError
<class 'ValueError'>: 
minkowski: 
```

This file was a result of **openscad**'s CSG export from a similar one:
```openscad
minkowski() {
  sphere(r=1, $fn=100);
  cylinder(r1=2, r2=0, h=2, $fn = 4);
}
```

_Originally I ran into this issue with a somewhat more sophisticated models, the examples above are my attempts to find a small one to reproduce._

At first I thought that the Minkowski sum causes the trouble, but apparently it's not, it's the cylinder with `r2=0, $fn=4` (basically, a four-side pyramid) which confuses **freecad**'s import. Minkowsky sum of different shapes, or with a slightly modified cylinder (see below) works fine¹.

¹ _Except that importing takes several minutes and loads CPU as **freecad** runs **openscad** to calculate the sum with some crazy precision, but that's a story for another issue maybe._

### Full version info

```shell
OS: Gentoo Linux (xmonad-wtf)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34223 (Git)
Build type: Unknown
Branch: master
Hash: 577004c1eba55641e92ea085fa6a078d36f49a06
Python 3.11.6, Qt 5.15.11, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
```
```

### Subproject(s) affected?

OpenSCAD

### Anything else?

I tried to look at the place where the exception comes from, and there are some moments which look suspicious.

First, in the [`process_ObjectsViaOpenSCADShape`](https://github.com/FreeCAD/FreeCAD/blob/1d2186b7d6f986b96f1b7932b63b124e186cfb2d/src/Mod/OpenSCAD/OpenSCADUtils.py#L645) method the message of mixed 2d and 3d appears when (at least) one of the `children` has null shape.
```python
def process_ObjectsViaOpenSCADShape(doc, children, name, maxmeshpoints=None):
    if all((not obj.Shape.isNull() and obj.Shape.Volume == 0) \
            for obj in children):
        return process2D_ObjectsViaOpenSCADShape(children,name,doc)
    elif all((not obj.Shape.isNull() and obj.Shape.Volume > 0) \
            for obj in children):
        return process3D_ObjectsViaOpenSCADShape(children,name,maxmeshpoints)
    else:
        FreeCAD.Console.PrintError( translate('OpenSCAD',\
            "OpenSCAD file contains both 2D and 3D shapes. That is not supported in this importer, all shapes must have the same dimensionality.")+'\n')
```
_Null shape will break both of the `all` checks, leading to a mixed shape error._

Apparently in my case cylinder's shape is null, so the message is misleading. It said “2D and 3D”, while it was “3D and null” actually. But I think the root cause of the issue is not a confusing error, but the reason why cylinder's shape is null.

So, second, in [`p_cylinder_action`](https://github.com/FreeCAD/FreeCAD/blob/1d2186b7d6f986b96f1b7932b63b124e186cfb2d/src/Mod/OpenSCAD/importCSG.py#L1121) there are some cases, and looks like my cylinder will fit into “frustum” category.
```python
def p_cylinder_action(p):
    …
    if h > 0:
        if ( r1 == r2 and r1 > 0):
            …
        elif (r1 != r2):
            if n < 3 or fnmax != 0 and n > fnmax:
                …
            else:
                if printverbose: print("Make Frustum")
                …
```

And then, per [`Frustum.createGeometry`](https://github.com/FreeCAD/FreeCAD/blob/1d2186b7d6f986b96f1b7932b63b124e186cfb2d/src/Mod/OpenSCAD/OpenSCADFeatures.py#L364), both radii have to be strictly positive.
```python
    def createGeometry(self,fp):
        if all((fp.Radius1,fp.Radius2,fp.FacesNumber,fp.Height)):
            …
```

I'm not sure how the openscad module works in freecad, but from the code I see that the above combined might be the reason I have the issue.

Also this is confirmed by me trying to have another shapes, or adding small `r2` to my cylinder, so it would be a proper frustum, not a full pyramid.

(I'd guess that possible solutions could be for frustum feature to accept zero as `r2`, or to have another feature, which will explicitly be a pyramid, but again I'm new to freecad and am not sure how this stuff is supposed to work.)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

KeithSloan, 2024-01-14:
The problem is the code for creating a Frustum does not handle radius of zero, so if r2=0 it fails to create a correct shape. I am looking at fixing the code in the OpenSCAD Alternate importer. Will post back when I have a fix for Fustrum. Or add a function for a n sided pyramid and use that when r2=0.

KeithSloan, 2024-01-15:
Looking at the code in OpenSCADFeatures.py I am seeing two issues here

As well as not supporting r2=0 The last section of code in createGeometry is

```
            shellperi = Part.makeLoft(wires)
            shell = Part.Shell(shellperi.Faces+faces)
            fp.Shape = Part.Solid(shell)
```
Which if my understanding is correct means a Loft will create a  **Conical Frustrum** where was what is needed in this particular is a case **Frustrum**. If sometimes a conical frustrum is required I think there needs to be an option and property to create the correct type, otherwise the code needs to create a normal frustrum rather than a conical one, as it does at present.

Think we should check with @chennes on this one.

KeithSloan, 2024-01-31:
Replacement OpenSCADFeatures.py there might be a more elegant/pythonetic way

[OpenSCADFeatures.py.txt](https://github.com/FreeCAD/FreeCAD/files/14111281/OpenSCADFeatures.py.txt)

Not sure why but I am still seeing an old tokrules.py
[tokrules.py.txt](https://github.com/FreeCAD/FreeCAD/files/14111282/tokrules.py.txt)

 Test files
[FrustrumProblem.scad.txt](https://github.com/FreeCAD/FreeCAD/files/14111295/FrustrumProblem.scad.txt)
[FrustrumWorks.scad.txt](https://github.com/FreeCAD/FreeCAD/files/14111298/FrustrumWorks.scad.txt)

luzpaz, 2024-08-30:
@KeithSloan perhaps you can make a PR so it can get reviewed in a more graphically convenient way ?

FEA-eng, 2025-06-22:
Still relevant

chennes, 2025-08-24:
@KeithSloan @luzpaz I've added an item to the next developer meeting agenda: https://github.com/FreeCAD/FreeCAD-developer-meetings/blob/main/Agendas/agenda-2025-09-07.md (it would be great if either of both of you could make that meeting).