FreeCAD/FreeCAD #11905: [Problem] Stop tree view from auto expand if model breaks

rahmanshaber, 2024-01-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently if compute failed for a feature in a body, the tree view of that body expands auto. This happens if the body is not active as well as if i import new object. 

Here you can see the treeview doesn't keep it's state, expands non-active bodies
[Screencast from 2024-01-05 16-37-04.webm](https://github.com/FreeCAD/FreeCAD/assets/30831735/124c3ce9-4cc5-4ada-8d1b-81e77e4bfeb1)

forum post
https://forum.freecad.org/viewtopic.php?p=730112#p730112

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * kicadStepUpMod 10.19.4
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rahmanshaber, 2024-01-06:
@Roy-043 just curious, why you think it's a part design workbench issue? treeview will expand any body or part no matter which workbench is used to build it, right?

ElementW, 2026-09-14:
* Duplicate of #16291

@maxwxyz #16291 has more discussion on this, I'd vote to close this one

closed 2026-09-14 by maxwxyz