FreeCAD/FreeCAD #11906: [Problem] Apostrophe placed before spreadsheet text input?

MarioElHazard, 2024-01-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

To specify the properties of a cell in the spreadsheet I copy the text in the cell left to it. If the content is pasted (in the properties window or elsewhere) a apostrophe is places before the text string (like 'modulePos_x). This only happens if I copy a text string from the FreeCAD spreadsheet. Numbers are copied correct.

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
  * A2plus 0.4.61
  * AnimationFreeCAD 1.0.0
  * Assembly4 0.50.4
  * Curves 0.6.15
  * fasteners 0.4.71
  * ThreadProfile 1.89.0
[/code]
```

### Subproject(s) affected?

Spreadsheet

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MattN92705, 2024-01-05:
This has already been reported in [https://github.com/FreeCAD/FreeCAD/issues/10058](url)

Seems to have been an intentional change - and despite several complaints, the maintainers haven't acknowledged that it should be fixed.

Roy-043, 2024-01-05:
Closing this as a duplicate.

closed 2024-01-05 by Roy-043