FreeCAD/FreeCAD #11911: [Feature request] Sketcher: implement non-construction point

pierreporte, 2024-01-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, the points in sketches are construction points. Normal points are useful for several operations, most notably hole patterns as requested in #6884, and also for master sketches. This feature has been discussed in various places, most recently in #11799 but also in #10100. Normal points would also be necessary for features that currently do not exist but may be wanted like custom patterns (not a matrix nor a crown).

Normal points should be visible outside sketches. Do not mask construction points currently visible outside sketches.

Normal points shouldn’t be displayed as a dot A cross (+ or ×) would be fine so it stands out.

In other CAD programs, it is not possible to extrude or revolve a sketch that has normal points in it. They consider them as open geometry. This is the case even if, for example, a point of a rectangle is made normal instead of construction.

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35510 (Git) AppImage
Build type: Release
Branch: main
Hash: da05f7c8e8b25ff783cc6c4eb3b73b640f134519
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * OpenDark 2023.12.1
```

### Subproject(s) affected?

Sketcher

### Anything else?

Normal points exist in all CAD software (except Alibre, that has workarounds).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-01-06:
Non-construction points are already available - just select a point and use [Sketcher ToggleConstruction](https://wiki.freecad.org/Sketcher_ToggleConstruction). But maybe they should be also created as non-construction when the non-construction geometry mode is enabled instead of having to convert each existing point.

pierreporte, 2024-01-06:
It is only possible to convert points created by `Sketcher_CreatePoint`. It is not possible to convert the extremities of a line or the vertex of a rectangle.

pierreporte, 2024-01-06:
This issue is now largely useless. I will create a new one to allow to toggle points attached to geometry into normal points, unless there is a way I don’t know.

closed 2024-01-06 by pierreporte