FreeCAD/FreeCAD #11912: [Problem] PartDesign/Hole feature threading clearance accepts only positive values

TCSMedia, 2024-01-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When trying to model a threaded rod it seems to be the easiest solution to create a threaded hole object and make a Boolean Cut with it on a proper sized cylinder. I we want to match the size of the threaded socket (or nut) and the rod, proper clearance is required.
The PartDesign/Hole feature has the clearance, but it accepts **only positive values** (which grows the hole size). I can type in negative value on the "Hole parameters" panel, and the view update shows the perfect (shrinked) result, but upon clicking the OK button the clearance sets back to 0,00.
This is a huge problem, because the shinked size threaded hole object could be used as a perfect cutting die for modelling the threaded rod.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Hungarian/Hungary (hu_HU)
Installed mods: 
  * A2plus
  * ExplodedAssembly
  * fasteners
  * fcgear
  * Manipulator
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-01-07:
Do you know the [Fasteners Workbench](https://wiki.freecad.org/Fasteners_Workbench)?

TCSMedia, 2024-01-07:
Yes, I know and used the Fasteners workbench, but for this technique it was also inappropriate.
For anyone who is interested let me explain in details:

I'm trying to model a matching M36 sized threaded rod / nut pair for use with FDM 3D printing. The new modeled thread option of the Hole feature is perfect for the nut, but the problem is that the nut is already designed and printed (I'm not the designer for that), so it's size cannot be changed (and it's a nominal M36 sized hole).

So I needed some solution for the rod to be modelled with matching dimensions which would require smaller outer diameter also to be able to fit (obviously the clearance is not zero). I was aware of the Fasteners workbench's **ScrewDie** feature which is exactly for this purpose: make a boolean cut with in on a proper sized rod. Unfortunately two issues arised:

- the size of the threaded hole is the same as the nominal M36 nut, **without any clearance** (so upon exporting and printing the rod it won't fit the nut)
- the Mesh conversion of the rod which is done with boolean cut takes several minutes and produces a gigantic file size (about 150 MB!) in STL export

The only solution right now was to shrink the modeled rod in the X and Y directions to a small amount, but this is a huge problem because it shrinks other features modeled on the rod as well.

**So in short**: it would be highly preferable for the Hole feature to handle well the negative Custom Thread Clearance parameter. And what's funny, that **it can do that right now** when I set it to negative value, the model displays the proper size, but upon pressing OK it resets to 0,00 and the model jumps back to original.

uwidev, 2024-01-19:
Trying to put in a negative number on clearances was something I also did. It was what felt intuitive to me with my rather limited understanding of CAD. I have yet to explore into addons.

My use case is very similar to @TCSMedia. I'm also FDM 3D printing, and tolerances are rather high in the positive direction (screws get bigger, holes shrink). This means that for parts to fit, the clearance of the hole needs to account for the tolerance on the screw, but by doing so, any existing parts that cannot be modified (e.g. metal screws) can no longer fit in the holes, as the holes are too much clearance.

A temporary solution is to scale the entire screw down, but you scale that entire model, which could have been trivially solved by allowing negative clearances.

pierreporte, 2024-08-30:
It would be best to implement the other way of making threads instead of having to use a workaround.

That being said, I never heard of interference fit for threads so I’m not sure if it is even needed to allow a negative value.

Chocolateandmilkwin, 2025-05-03:
I did a bit of digging on this and found a workaround.

First of all it seems the limit is arbitrary as the logic to make it smaller than 0 is present.
To show this, i used the hole editor with "live update" as you can enter negative values in the editor, and the model actually is updated with negative values.
It does seem like the limit is -0.7, so maybe the limit that is applied when the editor is closed could be changed to -0.7, that would still be very usable for 3d printing situations.
[image]
[image]

The workaround is to use a zip tool to extract the *.FCStd archive to a folder, then open the Document.xml file and find the
CustomThreadClearance and change it to what is needed, it can be hard to find the value in the document, so i first changed the value in freecad to 1.23 or some value that is not used anywhere else, then i could search it in the document.
[image]
You can then use Tools->Document Utility->Create Document to create the *.FCStd file from the extracted folder, and then you need to right click the hole feature and select "Recompute Object".

hubiscode, 2025-05-23:
I came here from a google search after I tried the same method to create a large 3D printable male thread by subtracting a female thread with negative clearance. It seems like an intuitive way to accomplish this quickly without having a direct way to create suitable male threads with custom clearance.

The current software behavior is a bit frustrating. Setting a negative offset in the hole parameter editor works and the threads are rendered correctly, but closing the editor resets the clearance to zero, which took me a while to notice.

Inspired by the comment above, I tried a different workaround which is more convenient. By tying the custom clearance value to a parameter defined in a spreadsheet alias, I was able to create threaded holes with negative clearance without editing the xml file.

Regarding the limits to the negative clearance mentioned above, this appears to be dependent on the particular thread. For an M24x2 thread, the limit appears to be around 2.1mm, while for an M18x1.5 thread the limit is around 1.6mm. The software produces the error message "Hole: TopoDS::Face" when the negative clearance becomes too large. My tests were done in FreeCAD 1.0.1 on linux x86.

I think the negative custom clearance on female threads is really useful to generate dies to cut male threads with through a boolean operation. The current behavior where negative values are set to zero when closing the hole parameter panel without informing the user is a bit undesirable, particularly since the generated geometry appears valid for negative values set via one of the workarounds.