FreeCAD/FreeCAD #11913: [Feature request] Sketcher: curves defined by equations

pierreporte, 2024-01-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Curves defined by equations are used when features follow precise mathematical definitions, like cams or involute profiles. There should be two kind of curves:

* normal functions like y = f(x) or x = f(y);
* parametric curves like x(t) = g1(t), y(t)=g2(t).

Several levels can be implemented:

1. Use the sketch origin as the equation origin, and base units.
2. Allow the user to select a point in sketch as the origin of the equation.
3. Allow the user to set the length of the unit.
4. Allow the user to have different x and y units.
5. Allow the user to define the full range of the curve using a rectangle.
6. Allow the user to define the axis of the equation (the rectangle of the previous point is rotated).
7. Allow the user to make the two axis at an arbitrary angle. I don’t know if this one is useful but who knows.

In any case:

* It should be possible to constrain geometry to this curve (make points coincident with it, lines tangent to it, etc).
* It should be possible to use variables in the equation, like named dimensions or values from a spreadsheet.
* It should be possible to edit the curve after it is created.
* It should be possible to re-limit the curve to use just a portion of it (the rest remains construction geometry or is hidden).

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35510 (Git) AppImage
Build type: Release
Branch: main
Hash: da05f7c8e8b25ff783cc6c4eb3b73b640f134519
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * OpenDark 2023.12.1
```

### Subproject(s) affected?

Sketcher

### Anything else?

Solidworks screenshots:

[image]

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-01-07:
Two macros that may be of interest:
https://wiki.freecad.org/Macro_3D_Parametric_Curve
https://wiki.freecad.org/Macro_Parametric_Curve_FP

pierreporte, 2024-02-14:
@PaddleStroke what do you think of this?

PaddleStroke, 2024-02-15:
Sounds interesting but it is a big feature that needs a lot of work.