FreeCAD/FreeCAD #11914: [Feature request] Edit multiple documents at a time (= make tabs independent editors)

pierreporte, 2024-01-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD has been designed to edit one document at a time only. When two files are opened and an edition is occurring in one of them (for example a sketch is opened), the program stays in this edit mode when switching to the other document (the task panel is still the same). If this second document has a sketch that is also opened, it closes the one opened in the other document. In other words, tabs are not independent editor.

This behavior is a limitation. It makes working on several documents in parallel difficult. A workaround is to open as much as FreeCAD sessions as needed, but it is impossible to detach a tab into its own FreeCAD session.

This would be a big change because the architecture would be altered. Though it would make FreeCAD more predictable as basically all software with tabs make them independent editors. 

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35510 (Git) AppImage
Build type: Release
Branch: main
Hash: da05f7c8e8b25ff783cc6c4eb3b73b640f134519
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * OpenDark 2023.12.1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-01-26:
This may be related: when inserting parts in an assembly (using the new worbench), it opens tabs for the files. That’s annoying so you close it and then you can’t edit them separately while in single document mode.

PaddleStroke, 2024-01-29:
I already pointed out this very weird behavior of freecad.
And the counter argument 'yes but if you used spreadsheet and you need to access the spreadsheet while editing a sketch'.

IMO this weird behavior should be killed. It is weird, error-prone etc.

One solution could be to make an exception for spreadsheet.

I seem to recall there was other arguments. Like the link creation. But nothing that can't be solved imo

pierreporte, 2024-06-04:
I stumbled upon a side effect of this issue: the undo stack is common for the whole program.

pierreporte, 2024-08-10:
@PaddleStroke The problem with spreadsheets could be solved with using two kind of tabs. Everything in a document that currently uses the whole screen (like spreadsheets) would open in an document-internal tab, while other documents are opened in normal tabs. Anyway, the undo stack and tasks panels should be tab specific.

Here is a UI proposal for these tabs. It could be similar to Opera 11/12 tab groups. When spreadsheets are opened, their tabs are in the group. When the document is active, the group is expanded. When it is inactive, it is collapsed, showing only the document name as today. For easier access, I propose a drop down menu to access grouped tabs from another document when the user clicks on the group border.

[image]

[image]

[image]

luzpaz, 2024-08-11:
> I stumbled upon a side effect of this issue: the undo stack is common for the whole program.

I'm sure there is also impact to the Macro Recording functionality.

pierreporte, 2024-12-29:
I'm changing the priority to P1 because it makes working with assembly a pain. When an assembly is open, and you work on another unrelated assembly in a second tab, the component insertion panel lists all the part of the other assembly.

Also, the undo stack should be per tab, so that there is no risk of undoing something made in other tabs, even if it's in the same document.

evan-king, 2025-01-12:
Speaking of spreadsheet exceptions, I really wish spreadsheets were in some way able to be viewed/edited _alongside_ other workflows so I can see the effect of edits in "real time," catch model breakage as early as possible, and maybe even directly compare the sheets of two different documents.  Some form of multi-pane splitting to the primary view panel (also meaning UI context would follow pane focus as well as per-pane tab selection) would be my ideal solution.

I mention it here because I suspect such a feature in any form would also rely heavily on the UI architecture changes _this_ feature request requires.  Perhaps that's something an architectural update might plan to accommodate (in the abstract, meaning just avoid implementation shortcuts likely to interfere).

pierreporte, 2025-01-19:
During [18th January 2025 developer meeting](https://github.com/FreeCAD/FreeCAD-developer-meetings/blob/main/Minutes/minutes-2025-01-18.md), there was consensus that the tabs should be treated independently, especially the undo stack.

Ping @chennes

PaddleStroke, 2025-01-19:
One thing to note about this, is when you edit an object from an external file. ie you change a part from an assembly for instance.
So I think the undo stack should not be tab specific. But document specific.

pierreporte, 2025-01-19:
@PaddleStroke This is actually a different thing, see #19139.

pierreporte, 2025-04-29:
A **1000 € reward** will be given to the contributor that writes a pull request, solving this issue, that gets merged. It will be paid by the FreeCAD Project Association. It is to be considered an incentive. Grant applications are welcome, but the total amount will exclude the reward.

leoheck, 2025-05-07:
Ah, this feature would be great. It is annoying having to close the widgets to measure or edit other files.

leoheck, 2025-05-07:
I realized today that even when launching two instances of FreeCAD (AppImage, in my case), having heavy activity in one of them, like marking everything in a large project for recalculation and then performing the recalculation, prevents me from interacting with the UI in the other instance. That’s crazy.