FreeCAD/FreeCAD #11915: [Problem] Freecad 0.21.2 hangs a few seconds after loading file

nicodemus26, 2024-01-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

OS: NixOS 23.11
Freecad version: 0.21.2

Freecad is hanging a few seconds after I load the attached file: [control_enclosure.zip](https://github.com/FreeCAD/FreeCAD/files/13851535/control_enclosure.zip)

I can sometimes make a few changes in the th_ins_holes sketch (which I'm actively working on) and save again before it hangs again.

I have a gdb core dump available if helpful. It's 156MB in a tarball though, so I'll need to get it to you some other way than attachment.

Thread state:
```
(gdb) info threads
  Id   Target Id                                           Frame 
* 1    Thread 0x7fffee681bc0 (LWP 10981) "..FreeCAD-wrapp" 0x00007ffff2918e6f in write ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  2    Thread 0x7fffec9d26c0 (LWP 10984) "QDBusConnection" 0x00007ffff2917d8f in poll ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  3    Thread 0x7fffe7fff6c0 (LWP 10985) "QXcbEventQueue"  0x00007ffff2917d8f in poll ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  4    Thread 0x7fffe62886c0 (LWP 10986) "pool-spawner"    0x00007ffff292363d in syscall ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  5    Thread 0x7fffe5a876c0 (LWP 10987) "gmain"           0x00007ffff2917d8f in poll ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  6    Thread 0x7fffe52866c0 (LWP 10988) "gdbus"           0x00007ffff2917d8f in poll ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  7    Thread 0x7fffe48376c0 (LWP 10989) "freecad:cs0"     0x00007ffff289fc96 in __futex_abstimed_wait_common ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  8    Thread 0x7fffd5ebe6c0 (LWP 10990) "freecad:disk$0"  0x00007ffff289fc96 in __futex_abstimed_wait_common ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  9    Thread 0x7fffd56bd6c0 (LWP 10991) "freecad:sh0"     0x00007ffff289fc96 in __futex_abstimed_wait_common ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  10   Thread 0x7fffd4ebc6c0 (LWP 10992) "freecad:shlo0"   0x00007ffff289fc96 in __futex_abstimed_wait_common ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  11   Thread 0x7fffc75ff6c0 (LWP 10993) "freecad:gdrv0"   0x00007ffff289fc96 in __futex_abstimed_wait_common ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  12   Thread 0x7fffc6dfe6c0 (LWP 10994) "freecad:gl0"     0x00007ffff289fc96 in __futex_abstimed_wait_common ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  22   Thread 0x7fff85ee16c0 (LWP 11006) "sandbox_ipc_thr" 0x00007ffff2917d8f in poll ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  23   Thread 0x7fff856e06c0 (LWP 11012) "..FreeCAD-wrapp" 0x00007ffff290f187 in wait4 ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  24   Thread 0x7fff83bff6c0 (LWP 11013) "ThreadPoolServi" 0x00007ffff2925a26 in epoll_wait ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  26   Thread 0x7fff82bfd6c0 (LWP 11015) "Chrome_IOThread" 0x00007ffff2925a26 in epoll_wait ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  28   Thread 0x7fff81bfb6c0 (LWP 11017) "ThreadPoolForeg" 0x00007ffff289fc96 in __futex_abstimed_wait_common ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  29   Thread 0x7fff6bfff6c0 (LWP 11018) "ThreadPoolForeg" 0x00007ffff289fc96 in __futex_abstimed_wait_common ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  30   Thread 0x7fff6b7fe6c0 (LWP 11019) "inotify_reader"  0x00007ffff29230ac in select ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  32   Thread 0x7fff6a7fc6c0 (LWP 11021) "CompositorTileW" 0x00007ffff289fc96 in __futex_abstimed_wait_common ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  33   Thread 0x7fff69ffb6c0 (LWP 11022) "Chrome_InProcGp" 0x00007ffff289fc96 in __futex_abstimed_wait_common ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  34   Thread 0x7fff697fa6c0 (LWP 11023) "VideoCaptureThr" 0x00007ffff289fc96 in __futex_abstimed_wait_common ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  35   Thread 0x7fff68ff96c0 (LWP 11024) "Chrome_ChildIOT" 0x00007ffff2925a26 in epoll_wait ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  40   Thread 0x7fff4bfff6c0 (LWP 11029) "VizCompositorTh" 0x00007ffff289fc96 in __futex_abstimed_wait_common ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  41   Thread 0x7fff4b7fe6c0 (LWP 11030) "freecad:gdrv0"   0x00007ffff289fc96 in __futex_abstimed_wait_common ()
   from /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
  42   Thread 0x7fff4affd6c0 (LWP 11031) "freecad:gl0"     0x00007ffff289fc96 in __futex_abstimed_wait_common ()
[... 126 lines cut ...]

### Full version info

```shell
[code]
OS: NixOS 23.11 (Tapir) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.2.Unknown
Build type: Release
Python 3.11.6, Qt 5.15.11, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.2
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Anything else?

I am also getting this error: `Failed to compute left/right minimum bearings for "cursor.pcf"` observed in https://github.com/FreeCAD/FreeCAD/issues/10514

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

nicodemus26, 2024-01-06:
`s/cursor.pcf/cursor/` does appear to have fixed this hang. This is probably a duplicate of #10514 . Closing this issue. Maybe the debug info will help?

closed 2024-01-06 by nicodemus26

timor, 2024-02-01:
> `s/cursor.pcf/cursor/` does appear to have fixed this hang.

Could you elaborate on how exactly you fixed this?  Thanks!

nicodemus26, 2024-02-02:
@timor, in nixos 23.11 something like this fixed the issue:
```zsh
export FC_CFG="~/.config/FreeCAD/user.cfg";
grep -rn 'cursor.pcf' "$FC_CFG";
cp "$FC_CFG" "${FC_CFG}.bak";
sed -i 's/cursor\\.pcf/cursor/g' "$FC_CFG";
grep -rn 'cursor' "$FC_CFG"*;
```

deftdawg, 2024-05-21:
> @timor, in nixos 23.11 something like this fixed the issue:
> 
> ```shell
> export FC_CFG="~/.config/FreeCAD/user.cfg";
> grep -rn 'cursor.pcf' "$FC_CFG";
> cp "$FC_CFG" "${FC_CFG}.bak";
> sed -i 's/cursor\\.pcf/cursor/g' "$FC_CFG";
> grep -rn 'cursor' "$FC_CFG"*;
> ```

Thanks for this, there's a small typo:

```sh
export FC_CFG="${HOME}/.config/FreeCAD/user.cfg"; # or export FC_CFG=~/.config/FreeCAD/user.cfg (no quotes)
grep -rn 'cursor.pcf' "$FC_CFG";
cp "$FC_CFG" "${FC_CFG}.bak";
sed -i 's/cursor\\.pcf/cursor/g' "$FC_CFG";
grep -rn 'cursor' "$FC_CFG"*;
```