FreeCAD/FreeCAD #11918: [Feature Request] Sketcher: Improve Dimension between Circle/Arc (Distance Constraints)

maxwxyz, 2024-01-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, it is only possible to constrain the minimum distance between two circles/arcs with the dimension tool:
[image]

The tool should be improved to behave like the distance tools on lines, **depending on the cursor position of the mouse**

1. Minimum distance between circle/arcs, already implemented
2. Maximum distance between circles/arcs
3. Minimum horizontal distance between circles/arcs
4. Maximum horizontal distance between circles/arcs
5. Minimum vertical distance between circles/arcs
6. Maximum vertical distance between circles/arcs

What is meant:
[image]
[image]

Building on top of @FlachyJoe #9166
This FR is includes https://github.com/FreeCAD/FreeCAD/issues/11412 but includes additional cases to support all distances.
Also includes https://github.com/FreeCAD/FreeCAD/issues/5864

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-16:
@FlachyJoe I tried to implement something based on your c2c implementation but failed. This is above my skills.

FlachyJoe, 2024-02-16:
@maxwxyz have you something to share?

maxwxyz, 2024-03-23:
@FlachyJoe I've stumbled across your [TODO list](https://github.com/users/FlachyJoe/projects/1/views/1) would you mind adding this as you are so skilled with constraints?

maxwxyz, 2025-07-18:
@FlachyJoe are you still planning on working on this? If not, would you add a completely new constraint type or just hack the distance constraint depending on mouse position within the dimension tool? I think a new constraint for maximum distance would be good to work with circle2circle (incl. arcs), circle2line and circle2point, but maybe even a max distance for ellipses later?
Could you take a look at my messing around and trial here: https://github.com/maxwxyz/FreeCAD/commit/1ffef30678f8eae054cf2a6d1e86884bae8a970e if this is going in the right direction?

FlachyJoe, 2025-09-09:
@maxwxyz I've rethought the problem and there are currently workarounds with constraints already implemented, so IMHO the solver shouldn't change.

[image]

[image]

[image]

We can write a command to automatize the construction and constraints creation.

maxwxyz, 2025-09-09:
@FlachyJoe Yes, the workaround was always there, but the same could be said for the currently implemented minimum distance. This could also be done with additional geometry but was solved in the solver. Not sure if the auto-creation of new geometry is ideal, as if you want to remove the dimension you need to manually delete the construction geometry.
Why is it better to not implement it in the solver in your opinion (I am not that familiar)?

FlachyJoe, 2025-09-10:
The solver currently has 36 constraints. With your proposal it will increase to 41, I think it's too much for peripheral usage.
And we already have functions which add multiple constraints e.g. [SketcherGui::makeTangentToEllipseviaNewPoint](https://github.com/FreeCAD/FreeCAD/blob/58a8d589a734199c0caef9f1dfd6a78087028006/src/Mod/Sketcher/Gui/CommandConstraints.cpp#L371)

maxwxyz, 2025-09-10:
So is the idea to create new geometry which remains if the circle is deleted or something like "internal" geometry like an ellipse?

Reqrefusion, 2026-05-01:
I think there are two problems here. The distances given for the arcs are from non-existent parts, and the maximum distance for the circles cannot be specified. I think the first problem could even be considered a bug. Even if it made sense, it shouldn't have created any distance based on things that didn't exist.

belinski-roman, 2026-08-30:
> I think there are two problems here. The distances given for the arcs are from non-existent parts, and the maximum distance for the circles cannot be specified. I think the first problem could even [...]

Why for non‑existent ones? Formally, we have an option to enable the display of conditional arc continuation. So why not link to these “objects”?