FreeCAD/FreeCAD #11920: [Problem] Sketcher: points of lines, circles, etc. cannot be turned into non-construction geometry

pierreporte, 2024-01-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This prevents from creating patterns of points that can be picked from outside the sketch, without the need to keep the lines in normal mode.

### Correct behaviour

1. Create a point using `Sketcher_CreatePoint`.
2. Select the point.
3. Click on `Sketcher_ToggleConstruction`.
4. The point turns white, as expected.

### Wrong behaviour

1. Create geometry (line, circle, whatever).
2. Select one of the points of the geometry.
3. Click on `Sketcher_ToggleConstruction`.
4. The points remains red.

### Workaround

It is possible to get the desired result by creating and converting standalone points first then connect them with construction geometry and eventually constraining them.

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35510 (Git) AppImage
Build type: Release
Branch: main
Hash: da05f7c8e8b25ff783cc6c4eb3b73b640f134519
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * OpenDark 2023.12.1
```

### Subproject(s) affected?

Sketcher

### Anything else?

This is for #6884

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-01-08:
I do not understand the reasoning. You can create a pattern from a point. Isn't it advisable to use PartDesign patterns instead of patterns (arrays) in sketches? And the endpoints of edges can already be accessed outside the sketch.

pierreporte, 2024-02-27:
@PaddleStroke is this bug easy to fix? It’s not very annoying right now but it will make normal points easier to use and the UX more consistent.

maxwxyz, 2024-03-02:
Related: https://github.com/FreeCAD/FreeCAD/issues/12582

pierreporte, 2025-04-12:
@PaddleStroke What do you think about this issue? Does my solution for duplicate point is a good idea for solving it. See this comment: https://github.com/FreeCAD/FreeCAD/issues/13391#issuecomment-2585909362

PaddleStroke, 2025-11-07:
@pierreporte I think #13391 should be closed because it is a misconception.

Now that does not mean that this ticket here is not valid. I need to investigate this

pierreporte, 2025-11-07:
@PaddleStroke The screenshot is excatly what I’m looking for so you understood correctly. There is a style issue but it’s out of topic.

> I'm not sure to understand the usecase. Could you explain better what's the goal with a clear example where the value of the feature request can be seen? It's easy to add defining points on top of construction geometry, the only thing that currently can't really be done is to hide points of normal geometry (ie defining edge and construction endpoints). Is that the main thing you want?

There are several use cases but they are basically a variation of the following. When you want to prepare a sketch for making holes with the PD Hole feature. Say you want to put holes equally spaced on a circle: you draw a regular pentagon with the dedicated feature, turn it to construction geometry, then turn the vertex as defining geometry. Currently you would need to put a standalone point manually at each of those vertex. This use case is very common in commercial CAD programs. Basically it reduces the need to manually put defining points on the sketch. Doing so is very easy but it’s quite cumbersome.

> Although I don't know if we can hide points of edges easily outside of editmode.

I think they shouldn’t be hidden. It’s great to make them visible like they are now if their supporting geometry is defining. It allows users to pick them and they don’t make the sketch busy. It’s what I love in Catia and Solidworks, on the other hand in Solid Edge and Alibre those points are hidden and it’s a bit less practical.

> Adds a lot of complexity for users who may get a mess of unwanted construction endpoints

It is true that users that only know FreeCAD will have issues. However it will be the opposite: since construction vertex will be the default (as they de facto are now), they may tend to turn them to defining instead of keeping them construction. However I believe that the benefits outweigh this temporary inconvenience (users eventually learn).

> I think https://github.com/FreeCAD/FreeCAD/issues/13391 should be closed because it is a misconception.

If two vertex are coincident (say a line ends at a circle center), then there would be two vertex that could be turned into construction or defining state. How do you make sure that users don’t change the wrong one? It is why I thought that current vertex should be internal so that the duplicates are hidden and standalone points are created on the fly as needed so that there is only one point to modify in this situation.

belinski-roman, 2026-09-01:
I don’t quite understand what problem this request is supposed to solve. In arrays for Part Design, we will soon have a function for suppressing instances. A simple (single point) can change the geometry type to constructive. If we add this for the centers of circles and the ends of lines, this may add difficulties when we are dealing with several vertices of different objects that are overlaid on each other.

belinski-roman, 2026-09-01:
Simply, you need to allow the array to use the vertices of lines or the centers of circles (arcs). By analogy with the Hole tool. Another option. This is to use two sketches — one with a pentagon, and the other with the projection of its vertices. Or the coordinate method — add full‑fledged parametric arrays to the sketch. Then your points could simply be distributed in a circle using an array within the sketch.

pierreporte, 2026-09-01:
@belinski-roman Simly, for whatever reasons if you want a set of points to be positioned on an existing shape like a polygon, you need to manually add points on the vertices. If you could simply switch them to defining geometry, you don’t need to manually add anything. It’s not necessarily related to holes or arrays, although they would heavily make use of it. Besides, there is not always circle centers available.

belinski-roman, 2026-09-01:
A pentagon consists of five lines and 10 vertices. If you claim that you need this functionality for new arrays in Part Design (array by points in sketches), then you will have 10 instances, 5 of which will coincide in pairs in space. Is not it so?

belinski-roman, 2026-09-01:
> Simly, for whatever reasons if you want a set of points to be positioned on an existing shape like a polygon, you need to manually add points on the vertices. If you could simply switch them to defining geometry, you don’t need to manually add anything. 

It seems that the issues you mentioned should have been resolved by the new style of displaying sketches outside. I came across a mention in Discord, but I don’t know the details.

pierreporte, 2026-09-01:
5 is not much but it can be more complex. At the end of the day, from experience, switching state is quicker than adding standalone points. In Alibre, adding points is a major PITA.

The problem is not going to be resolved by displaying construction lines outside edit mode, by definition of construction geometry.

belinski-roman, 2026-09-01:
Individual points in a sketch for a Hole were only recently allowed. Previously, only circles and arcs (in closed shapes) were allowed. This can be circumvented if another type is allowed — the vertices of lines and arcs (at least, that’s how it works in Solidworks).

belinski-roman, 2026-09-01:
Solving the problem with the pentagon

https://github.com/user-attachments/assets/38ee790d-e834-4f46-94ea-4d9c392aa322

pierreporte, 2026-09-01:
It’s not just about holes, and your workaround involves extra steps that would be unnecessary if this issue is resolved.

belinski-roman, 2026-09-01:
This is how Solidworks processes lines and arcs. 

https://github.com/user-attachments/assets/7b3905c8-e57e-46cb-97b8-e45c59d06972

I believe that the Freecad approach is more flexible, and what you call a problem can be easily solved using the Offset function (for any closed or open contour)

belinski-roman, 2026-09-01:
> It’s not just about holes, and your workaround involves extra steps that would be unnecessary if this issue is resolved.

I believe your solution adds more problems. There’s an unclosed complex wire that I need to convert into auxiliary geometry. Right now, I can simply double‑click on it and convert it into auxiliary geometry. After your change, I’ll have to select everything using a frame, because the line itself and its vertices can be independently converted into auxiliary geometry, and it’s easy to accidentally skip one vertex.

belinski-roman, 2026-09-01:
You’re solving the problem from the wrong end. It would be better to modify the Hole tool by adding another option — the vertices of lines and arcs.

pierreporte, 2026-09-01:
It is not the wrong end. Both approach can be implemented. In fact Solidworks has both.

belinski-roman, 2026-09-01:
In SolidWorks, there is only one type of point, and their state cannot be changed.

pierreporte, 2026-09-01:
I have used Solidworks and was able to switch between construction points and defining points. It is true that hole points are unique, but then again this is not just about holes.

belinski-roman, 2026-09-02:
If you make it so that switching the line mode and its two vertices occurs independently, rather than the way it is now (I can select one of the vertices, both, or just specify the line, and all three will switch), you will turn the user’s life into hell. Because one unswitched point will prevent you from creating a Pad (or Pocket) in Part Design. Imagine that there are two lines whose vertices coincide, and the user moved one line into auxiliary geometry, but missed one point that is hidden under the vertex of the other line. Finding such an error will instantly wear the user down. Your problem should be solved differently, perhaps indeed using layers within the sketch, as suggested @PaddleStroke

pierreporte, 2026-09-02:
IIRC the Pad feature is already clever enough to ignore defining points. From experience the proposed behavior rarely cause any problems.

What you propose is longer while the solution proposed here works in more than one commercial programs without any issue. We can add modes to use all kind of vertex but it doesn't mean we shouldn't improve points.

Have you considered that not allowing vertex to be defining, sketch would look more busy while outside edit mode, with no other reason? Readability is important and the proposal goes in this direction as well.

belinski-roman, 2026-09-02:
If we make it so that the vertices of lines (or arcs) always (always!) remain regular geometry, regardless of the mode in which the “parent” line or arc is located (but without causing problems when creating a Pad or Pocket), then this still makes some sense. But if you’re saying that we need the ability to independently switch the mode for each of the vertices and the line itself, then this will create more problems than it solves.