FreeCAD/FreeCAD #11929: [Problem] Inconsistency in activating documents leads to no document active and a crash

lukash, 2024-01-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In a complex project with multiple documents depending on each other, I observe inconsistencies in selecting documents. Some of open documents don't activate when clicked their name. They only activate when clicked on an object inside them. It's possible to rightclick such a document when it's not active and click Close document. This results in a state when no document is active. Then, at least with some of the other documents, they don't activate when either their name or any object inside is clicked on. Interacting with their objects (e.g. opening a sketch) then leads to a crash.

Steps roughly:
1. open a complex project with multiple documents
2. right click on a certain document and select Close document (normally, this is not possible right away, as that item is not in the context menu, but in this special case it is there)
3. click on a plus next to another document that doesn't in this state aactivate when its name is clicked
4. open a sketch inside this expanded, but inactive document

I have a 100% reproducer for this in a project I don't want to share publicly, but can share privately.

Traceback:
```
(gdb) bt
#0  0x00007ffff65a7ccf in Gui::Document::setEdit (this=0x0, p=0x7065b70, ModNum=0, subname=0x0) at /home/lu/dev/nix-freecad/source/src/Gui/Document.cpp:297
#1  0x00007ffebd706481 in SketcherGui::ViewProviderSketch::doubleClicked (this=0x7065b70) at /home/lu/dev/nix-freecad/source/src/Mod/Sketcher/Gui/ViewProviderSketch.cpp:2777
#2  0x00007ffff6932097 in Gui::TreeWidget::mouseDoubleClickEvent (this=0xb70cf0, event=0x7ffffffd65e0) at /home/lu/dev/nix-freecad/source/src/Gui/Tree.cpp:1443
#3  0x00007ffff1bb33d4 in QWidget::event(QEvent*) () from /nix/store/0acd4cnpgm06lhq4v95lw5qy4bipnha1-qtbase-5.15.11/lib/libQt5Widgets.so.5
#4  0x00007ffff1c5e7ee in QFrame::event(QEvent*) () from /nix/store/0acd4cnpgm06lhq4v95lw5qy4bipnha1-qtbase-5.15.11/lib/libQt5Widgets.so.5
#5  0x00007ffff0ecadd3 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () from /nix/store/0acd4cnpgm06lhq4v95lw5qy4bipnha1-qtbase-5.15.11/lib/libQt5Core.so.5
#6  0x00007ffff1b7102e in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /nix/store/0acd4cnpgm06lhq4v95lw5qy4bipnha1-qtbase-5.15.11/lib/libQt5Widgets.so.5
#7  0x00007ffff1b79500 in QApplication::notify(QObject*, QEvent*) () from /nix/store/0acd4cnpgm06lhq4v95lw5qy4bipnha1-qtbase-5.15.11/lib/libQt5Widgets.so.5
#8  0x00007ffff6663c0a in Gui::GUIApplication::notify (this=0x7ffffffd6fa0, receiver=0xb962c0, event=0x7ffffffd65e0) at /home/lu/dev/nix-freecad/source/src/Gui/GuiApplication.cpp:85
#9  0x00007ffff0ecb058 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /nix/store/0acd4cnpgm06lhq4v95lw5qy4bipnha1-qtbase-5.15.11/lib/libQt5Core.so.5
#10 0x00007ffff1b7768e in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) ()
   from /nix/store/0acd4cnpgm06lhq4v95lw5qy4bipnha1-qtbase-5.15.11/lib/libQt5Widgets.so.5
#11 0x00007ffff1bcb87e in QWidgetWindow::handleMouseEvent(QMouseEvent*) () from /nix/store/0acd4cnpgm06lhq4v95lw5qy4bipnha1-qtbase-5.15.11/lib/libQt5Widgets.so.5
#12 0x00007ffff1bced20 in QWidgetWindow::event(QEvent*) () from /nix/store/0acd4cnpgm06lhq4v95lw5qy4bipnha1-qtbase-5.15.11/lib/libQt5Widgets.so.5
#13 0x00007ffff1b7103e in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /nix/store/0acd4cnpgm06lhq4v95lw5qy4bipnha1-qtbase-5.15.11/lib/libQt5Widgets.so.5
#14 0x00007ffff6663c0a in Gui::GUIApplication::notify (this=0x7ffffffd6fa0, receiver=0x1ebb120, event=0x7ffffffd6b30) at /home/lu/dev/nix-freecad/source/src/Gui/GuiApplication.cpp:85
#15 0x00007ffff0ecb058 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /nix/store/0acd4cnpgm06lhq4v95lw5qy4bipnha1-qtbase-5.15.11/lib/libQt5Core.so.5
#16 0x00007ffff132fe6d in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () from /nix/store/0acd4cnpgm06lhq4v95lw5qy4bipnha1-qtbase-5.15.11/lib/libQt5Gui.so.5
#17 0x00007ffff130332c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /nix/store/0acd4cnpgm06lhq4v95lw5qy4bipnha1-qtbase-5.15.11/lib/libQt5Gui.so.5
#18 0x00007fffeaa981ea in xcbSourceDispatch(_GSource*, int (*)(void*), void*) ()
   from /nix/store/s94nki7hs6dbq96jmxbwaw4mywkxs78z-qtbase-5.15.11-bin/lib/qt-5.15.11/plugins/platforms/../../../../../jiprinddjxbksh18gamf3c5yk1d7pbrr-qtbase-5.15.11/lib/libQt5XcbQpa.so.5
#19 0x00007fffecd120d7 in g_main_dispatch () from /nix/store/5qxy21wbd1n391y8wrf2djyjkiz1bs16-glib-2.78.1/lib/libglib-2.0.so.0
#20 0x00007fffecd14ff7 in g_main_context_iterate_unlocked.constprop () from /nix/store/5qxy21wbd1n391y8wrf2djyjkiz1bs16-glib-2.78.1/lib/libglib-2.0.so.0
#21 0x00007fffecd155cc in g_main_context_iteration () from /nix/store/5qxy21wbd1n391y8wrf2djyjkiz1bs16-glib-2.78.1/lib/libglib-2.0.so.0
#22 0x00007ffff0f22bf6 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /nix/store/0acd4cnpgm06lhq4v95lw5qy4bipnha1-qtbase-5.15.11/lib/libQt5Core.so.5
#23 0x00007ffff0ec9a53 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /nix/store/0acd4cnpgm06lhq4v95lw5qy4bipnha1-qtbase-5.15.11/lib/libQt5Core.so.5
#24 0x00007ffff0ed1f56 in QCoreApplication::exec() () from /nix/store/0acd4cnpgm06lhq4v95lw5qy4bipnha1-qtbase-5.15.11/lib/libQt5Core.so.5
#25 0x00007ffff64d6934 in Gui::Application::runApplication () at /home/lu/dev/nix-freecad/source/src/Gui/Application.cpp:2379
#26 0x000000000042859f in main (argc=1, argv=0x7ffffffd7c48) at /home/lu/dev/nix-freecad/source/src/Main/MainGui.cpp:268
(gdb) f 1
#1  0x00007ffebd706481 in SketcherGui::ViewProviderSketch::doubleClicked (this=0x7065b70) at /home/lu/dev/nix-freecad/source/src/Mod/Sketcher/Gui/ViewProviderSketch.cpp:2777
2777        Gui::Application::Instance->activeDocument()->setEdit(this);
(gdb) p Gui::Application::Instance->activeDocument()
[Thread 0x7fff88bfa6c0 (LWP 721868) exited]
$1 = (Gui::Document *) 0x0
```

This particular traceback is of 0.21.2 (that was easier for me to build), but the bug reproduces the same in current weekly (version in version info).

Code in question:
```
bool ViewProviderSketch::doubleClicked()
{
    Gui::Application::Instance->activeDocument()->setEdit(this);
    return true;
}
```

It's apparent the crash happens because there is no active document and a null pointer is derefenced.

### Full version info

```shell
OS: NixOS 23.05 (Stoat) (none+awesome/none+awesome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35485 (Git)
Build type: Release
Branch: main
Hash: a662fbb2ff267fb8dd56fcc3d1ec55a5968b7f5a
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Curves 0.6.5
  * Render 2023.1.0
  * Assembly4 0.50.6
```

### Subproject(s) affected?

None

### Anything else?

https://forum.freecad.org/viewtopic.php?t=83998

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-01-10:
I don't understand how to reproduce it.

> (normally, this is not possible right away, as that item is not in the context menu, but in this special case it is there)

?

> click on a plus next to another document that doesn't in this state aactivate when its name is clicked

Where is this plus sign?

lukash, 2024-01-10:
> I don't understand how to reproduce it.

You need the reproducer project, as I've stated in the report. Or, potentially, just a number of very big documents, I'm not sure.

> > (normally, this is not possible right away, as that item is not in the context menu, but in this special case it is there)
> 
> ?

Normally, when you right-click a document that isn't active, the Close document option is not in the context menu. In this case it is.

> > click on a plus next to another document that doesn't in this state aactivate when its name is clicked
> 
> Where is this plus sign?

To the left of the document name, the tree marker that expands the tree.

@bgbsww already contacted me for the project reproducer and is working, or planning to work on this issue.

bgbsww, 2024-01-10:
I am able to reproduce using the (very large) example.  I'm currently trying to generate a simpler public test case, as the shortcut path of looking directly for the cause of the broken project did not lead me to quick results.

lukash, 2024-01-10:
You'll likely know better than me, but as I've said in the message to you, it happened to me once that following the steps, the document activation worked correctly. Hence, my current theory is the cause is some sort of a race stemming from the documents being big.

bgbsww, 2024-01-11:
I think this is a nice simplified version of this crash: [CrashBoxes.zip](https://github.com/FreeCAD/FreeCAD/files/13896877/CrashBoxes.zip)

1. Open BoxAssem.FCStd
2. Right click on any of Box3, Box2, or Box4 and select Close Document from the menu.  You will see no activeDocument.
3. Single Click the Open Arrows for BoxAssem, then Fusion, then Extrude.   Still no activeDocument.
4. Double Click on Sketch and the crash happens.

Note that if you take the focus out of FreeCAD and back in* , then the BoxAssem document is selected and no crash occurs.  
*Because maybe you were writing a github comment at the time.

I am not going to ask to be assigned and take this further, because it looks to me like some discussion on what the correct behaviors should be needs to happen before a fix is applied.  So open season for someone else.

Thanks, @lukash for the example, I will destroy my copies of your documents tomorrow.

lukash, 2024-01-11:
@bgbsww thanks a lot for the analysis and the public reproducer. Sent you a PM on the forums.

luzpaz, 2024-08-17:
@bgbsww any idea who needs to be part of the continued discussion ?

bgbsww, 2024-08-17:
Unfortunately, not really.  But as a UI/UX question I would start with the design working group to get a suggestion on what the behavior should be ( obviously not crash, but what should be selected, should a warning dialog come up, etc ).

Once that's known, this might be a good starter bug.

luzpaz, 2024-08-17:
@FreeCAD/design-working-group :point_up_2:

maxwxyz, 2024-10-07:
can someone try this in the RC / latest dev version? IIRC the document activation on left/right click in the tree view was removed.

wwmayer, 2024-10-13:
I can confirm the crash with v0.21 but with v1.1 I can't.

maxwxyz, 2024-10-13:
thanks, closing this as fixed then.

closed 2024-10-13 by maxwxyz