FreeCAD/FreeCAD #11934: [Problem] Cannot create Groove and Revolution from ShapeBinder and SubShapeBinder sketch

filippor, 2024-01-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

create a revolution or grrove from a ShapeBinder sketch give  Revolution: SketchBased: No face selected error

Alse get error on opening file created with stable version 0.21.0

Part Design workbench
Step to reproduce:
- create a sketch
- create a ShapeBinder on that sketch
- using shapebinder create a Revolution

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35510 (Git)
Build type: Release
Branch: main
Hash: da05f7c8e8b25ff783cc6c4eb3b73b640f134519
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * BIM 2021.12.0
  * Curves 0.6.8
  * Defeaturing 1.2.0
  * freecad.gears 1.0.0
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jcoalson, 2024-02-25:
@AjinkyaDahale could you take a look and see if there is a quick fix? The problem is this commit from your PR#7193: https://github.com/FreeCAD/FreeCAD/pull/7193/commits/6b1a1e1508ce010675a2c739f1f7fdf6b3a2dcd8#diff-cd409b830506c22cc2520df5e57b70f576a7bed50aaaebab9e1ad8b053545c7bR166

^ specifically, calling `getSupportFace()` before checking the method.

AjinkyaDahale, 2024-02-28:
> @AjinkyaDahale could you take a look and see if there is a quick fix?

Sorry I'd been focused on some other stuff so getting back into partdesign may take me some time. Please remind me if I don't get back by Friday.

> The problem is this commit from your PR#7193: [6b1a1e1#diff-cd409b830506c22cc2520df5e57b70f576a7bed50aaaebab9e1ad8b053545c7bR166](https://github.com/FreeCAD/FreeCAD/commit/6b1a1e1508ce010675a2c739f1f7fdf6b3a2dcd8#diff-cd409b830506c22cc2520df5e57b70f576a7bed50aaaebab9e1ad8b053545c7bR166)
> 
> ^ specifically, calling `getSupportFace()` before checking the method.

So was this working before this got merged?

jcoalson, 2024-02-29:
Thanks, yes if you check out even the previous commit https://github.com/FreeCAD/FreeCAD/commit/a7da1003fc it is working.

AjinkyaDahale, 2024-03-07:
Hi @jcoalson sorry this will take some time. Some urgent stuff came up.

maxwxyz, 2024-06-24:
@AjinkyaDahale any news on this?

AjinkyaDahale, 2024-07-02:
@maxwxyz it's best to not assign blockers to me at this moment. I might not be reliably available until the end of July.

AjinkyaDahale, 2024-08-17:
I was able to avoid this particular bug (#15920), but seems "up to face" for revolution no longer works, even when not using shapebinder (IIRC that was supported in #7193). This may be a topo-naming issue.

maxwxyz, 2024-08-17:
@bgbsww ^

AjinkyaDahale, 2024-08-18:
As expected, the up-to-face stuff is not working after the TNP fix stuff was added.

```
#ifdef FC_USE_TNP_FIX
            // revolve the face to a solid
            //            TopoShape result(0);
            try {
                result = base.makeElementRevolution(gp_Ax1(pnt, dir),  supportface, upToFace);
            }
            catch (Standard_Failure&) {
                return new App::DocumentObjectExecReturn("Could not revolve the sketch!");
            }
#else
            RevolMode mode = RevolMode::None;
            generateRevolution(result,
                               base.getShape(),
                               sketchshape.getShape(),
                               supportface,
                               upToFace,
                               gp_Ax1(pnt, dir),
                               method,
                               mode,
                               Standard_True);
#endif
```
The part after `#else` is the new stuff from #7193. So now 2-directional revolve works but up to face no longer does.

bgbsww, 2024-08-24:
The general error here appears to be that PropertyLinkSub::getValue does not return a usable face and thus the revolution (and other examples) then fail with a "SketchBased: No face selected".  You can replicate this with a real face from something like a pad too.  When we make a SubShapeBinder, we can no longer get to the face ( or sketch that gets converted into a face ) via the link in the SubShapeBinder.  This will probably be one little fix that solves all cases.

bgbsww, 2024-08-24:
I suspect this is missing link visibility code in PartFeature.cpp calling missing  getHiddenSubnames code in link.cpp.  We'll see ...

bgbsww, 2024-08-25:
Yeah, that code wasn't needed.  What it did need was to stop arbitrarily failing getFaceFromLinkSub when the subelements list has an empty entry, since for things like a sketch, the link has an empty sub reference - the sketch is the reference.

PR #16036 (2024-08-26): Update to Toposhape versions and fix shapebinder references with test
Fix #11934.
closed 2024-08-26 by chennes