FreeCAD/FreeCAD #11936: [Problem] Feature Request: A file format that integrates well with git.

konkers, 2024-01-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The `.FCStd` format is a zip of multiple files.  This produces a binary file which git is notoriously poor at handling.  For example the [lumenpnp ](https://github.com/opulo-inc/lumenpnp)repo, a not terribly complex mechanical project, contains almost 1GB worth of git objects at this point.

Some format where the data is written out in plain text would make managing projects that FreeCAD with git much easier.

One possible approach would be to support an alternate `.FCDir` format which would be this same collection of files in a directory instead of a single zip.   The various `*Color*` files are binary which, in this approach, would benefit from being stored in a textual representation.  This will allow git to manage a set of text files rather than monolithic binary files.

One downside is that this will increase the size of the currently checked out files.  In the above lumen repo this would be ~1136MB vs ~120MB.  These files will be compressed by git when stored in the object database so the size is not an issue there. 

The majority of the size here looks to be in brep files.  According to https://wiki.freecad.org/File_Format_FCStd, these are often cached calculations from `Document.xml`.  Much of the above checkout bloat could be resolved by having FreeCAD maintain a local, possibly compressed, cache of this geometry rather than keep the cache in the files.  This will also remove the need for git to track these files resulting in a smaller object db.  This trades off reduction in repo size for extra computation time when pulling new changes from git.

### Full version info

```shell
[code]
OS: macOS 13.6.3
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kadet1090, 2024-01-09:
Maybe for such git friendly format we should not store the brep representation? it is non diffable, it takes a lot of space, will create a lot of conflicts and it is possible to recompute based on the document tree. Such format could then specify brep files in the .gitignore file so they could be reused for performance reasons but after chedkong out the file it will require full recompute.

FlachyJoe, 2024-01-09:
Because of the huge size of BREP data, sharable documents shouldn't include re-computable geometries. The everyday limiting 1Mo max file size of the forum is a good proof of the need for a such format.

 I had a try for a brep removing macro: https://forum.freecad.org/viewtopic.php?t=78965

pierreporte, 2024-01-09:
Does removing the BREP increase load time? If so, it could significantly increase the loading time of large assemblies. The files themselves would load a lot faster but the reconstruction of the models will take more time and depend on hardware performance. Is there any performance benchmark on this?

Besides, the version control of CAD parts differs from source code version control. I’d rather have a CAD-friendly git alternative. Something like [SCCS](https://en.wikipedia.org/wiki/Source_Code_Control_System) (tracking individual files with versions 1, 2, 3, etc. with locks) but more modern.

pierreporte, 2024-01-09:
My experience with CAD cache systems hasn’t been very good. In all companies I’ve worked for, it was basically overridden every single time. It could work for one-user projects, even big ones but collaboration will always lead to a lot of edited parts which needs updating in the cache. In the industry, it is not considered acceptable that an assembly containing thousands of parts (not tenth of thousands) needs half an hour to fully load.

konkers, 2024-01-09:
```My experience with CAD cache systems hasn’t been very good. In all companies I’ve worked for, it was basically overridden every single time. It could work for one-user projects, even big ones but collaboration will always lead to a lot of edited parts which needs updating in the cache.```

This sounds like the perfect application for a distributed cache.  Something that massive software projects routinely use.

pierreporte, 2024-01-09:
This is starting to look like a white elephant. Git is just the wrong tool for that.

hasecilu, 2024-01-10:
This is an example of project with files more git-friendly: https://github.com/alexneufeld/FreeCAD_modelmania
At least making little changes should not make you have another big blob. If I'm not wrong it's the FreeCAD files unziped, maybe @alexneufeld can correct me.

alexneufeld, 2024-01-10:
This is the 'save as directory' feature implemented by @realthunder in the linkdaily builds. One could most likely write [git hook(s)](https://git-scm.com/book/en/v2/Customizing-Git-Git-Hooks) to accomplish something similar. As mentioned previously, git(hub) has limits on file *sizes* - some of the files in the repo you linked have data omitted in order to fit under the cap. 

some more relevant reading:
- [JojoDiff - diff utility for binary files](https://jojodiff.sourceforge.net/)
- [Git large file storage](https://git-lfs.com/)

Git LFS in particular is specifically pitched for use with graphics/video content, which is similar to our use case. See also [https://github.com/alexneufeld/lfstest](https://github.com/alexneufeld/lfstest)

konkers, 2024-01-10:
Oh nice!  "Save as directory" sounds essentially like my proposed solution.

I'm not a huge fan of LFS unless absolutely necessary as it breaks the git paradigm of each clone being a completely independent instance.

yorikvanhavre, 2024-01-11:
My 2 cents here: 

We've been go and round this conversation a lot already. Any way we look at it, we need the BREP files. A workflow without it is simply not efficient at all. Storing them outside the file is a possibility, but would introduce a wealth of other problems (synchronisation, version checks, etc).

Storing a file as a directory, using the Link3 branch works, but your git repo finds itself cluttered with a bunch of files, most of them, although they are text inside, behaving like blobs (diffs are meaningless). So the interest of Git here is mitigated.

All in all, zip-based formats are something increasingly more common, many FOSS applications use them. So I believe our best course of action here is to wait for proper support for these kinds of files in Git. This, certainly, will be the killer solution.

In the meantime, there is a small utility that can be used to make Git work better with FreeCAD files: https://github.com/FreeCAD/FreeCAD/blob/main/src/Tools/fcinfo It doesn't solve everything but at least gives meaningful diffs...

WinlCheng, 2024-01-29:
If you want to save like OpenSCAD, the user mentioned the issue of slow loading speed. Explaining the text requires a certain amount of time. Without explaining the text, git cannot function properly. Therefore, attempting to use git for drawing management is likely to fail.

However, it would be great if FreeCAD could have its own drawing management system. Or, more simply, a drawing information system.

johnsonm, 2024-03-15:
@yorikvanhavre 

> Storing a file as a directory, using the Link3 branch works, but your git repo finds itself cluttered with a bunch of files, most of them, although they are text inside, behaving like blobs (diffs are meaningless).

When I have used Realthunder's branch save-as-directory and Git, I have found great value in `git diff` — there's enough structure in the data to often remind me what I have changed in the commit. When I started wanting more upstream features and he got behind on merging into his branch, losing save-as-directory made me very sad.

I don't understand the "cluttered" comment either. It assumes that people who want save-as-directory would even agree that it's "cluttered" — this seems like an unwarranted assumption.

The assertions that his approach are useless continue to leave me nonplussed. :sob: 

You don't have to want to be able to `git merge` to want this feature.

yorikvanhavre, 2024-03-18:
It was only my humble opinion. If the feature is useful to some, I of course have no objection to have it included...

johnsonm, 2024-05-22:
Is there a better open issue here for discussing simply bringing in Realthunder's save-as-directory work?

I'm worried that the description on this issue, by being subjectively valued ("well"), will have the unintended side effect of encouraging bike-shedding, and make it take longer to reach consensus on what could be done.

My experience is that save-as-directory integrates *well enough* with git to be worth implementing to resolve this issue.

I understand that there might be concern that some users might not recognize the limitations of save-as-directory, so it could make sense to **hide it behind an explicitly experimental parameter,** enabled only by pasting in a name and not merely clicking a button in the GUI. But merging that feature into upstream FreeCAD would give a starting point for suggestions for changes that would improve the interaction with git in concrete ways.

Since save-as-directory is basically creating a directory that looks like the zip file, making changes later to improve storage to improve the experience should not break backward compatibility; you'd still need to be able to read the original format to be able to read old FCStd files.

tobiasfalk, 2024-05-22:
Yes, save-as-directorysound like a good solution for advanced users, and a good integrated Version controll sound even better.
But an other simpler solution could be that FreeCAD just saves a change log file(`.FCpv`) directly next to the FreeCAD file. This could be a simple text file witch a list of all the changes(and from who they were done) to the FreeCAD file and maybe even more(things like part description or so).
This would be a simple and easy to understand way for users to look up the changes to a part.

An other thing that would be nice if FreeCAD had the ability to save the FreeCAD file also automatically as a 3D model file, that github or other git Web management programs can display. This way it is easy to see the changes that were done by someone.
https://docs.github.com/en/repositories/working-with-files/using-files/working-with-non-code-files#3d-file-viewer

https://gitlab.com/gitlab-org/gitlab/-/issues/17610

johnsonm, 2024-05-22:
Both of those ideas (changelog, model file) are sidecars to the core consideration of avoiding big binary blobs of zip files. Considering them as enhancements independently of this issue would make sense. But I think they are distractions from the core point of "just don't make things needlessly bad for using git to store your files."

Realthunder figured this out years ago, and It. Just. Works.

FlachyJoe, 2024-05-23:
FYI you can now dramatically reduce file size of models created with PartDesign by Suppressing some feature. A Suppressed feature doesn't save shape and is fully recalculate when unsuppressed.

recursivenomad, 2024-07-10:
Hopping in to mention that #9432 is also discussing this topic.  Would be a great feature to natively have in FreeCAD (both the addition of unzipped source files, as well as optional removal of any recomputible files)

FlachyJoe, 2026-01-28:
Not all the BRep can be deleted, some objects are unable to recreate their own shape i.e. the non-parametric objects.
For the others we also need to be sure the toponaming is the same after recreation.

bubble-07, 2026-01-28:
Ah, yeah,that's a good point. Is the stability of toponaming only able to be guaranteed with stored BRep? My best (uninformed about the inner workings of toponaming) mental model there is that there's no FreeCAD version stability for BRep, but for a given FreeCAD version and the same file, it should be the same. If so, would the work there come down to being able to pin down a permanent topological naming scheme? Or is there also nondeterminism in toponaming for the same parametric (cached-object-free) file in the same version?

MikeOpsGit, 2026-01-28:
AFAIK imported geometries such as STLs cannot have their BREPs recompiled, because the geometry wasn't made in FreeCAD so it has no reference point.

pieterhijma, 2026-03-13:
So, I've been looking into discardable BRep files with the goal to make only small changes to the FreeCAD code, backwards compatibility and even forward compatibility.