FreeCAD/FreeCAD #11942: [Problem] Some of the checkboxes in Tasks pane inside Sketch editing view are not scaled properly

grudir, 2024-01-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The checkboxes in Constraints and Elements lists in Tasks pane inside Sketch editing view are displayed properly when using Windows Display scale percentage different from 100%. Additionally the arrows used to expand and collapse the sections inside the Tasks pane as well as the icons in Edit > Preferences are not displayed properly.

Pre-requisites:
- Right click Windows Desktop and choose Display Settings. Set Scale to 125%.

Steps to reproduce:
1. Start FreeCAD 0.21.1.
2. Click File > New.
3. Click Part Design > Create Body.
4. Click Sketch > Create Sketch.
5. Choose XY-plane (and click OK).
6. Draw a rectangle.

Actual result:
The checkboxes in Constraints and Elements lists inside Tasks pane are a bit jagged (probably due to scaling).

Expected result:
The checkboxes in Constraints and Elements lists inside Tasks pane are clear similar to the checkbox on the left of the Filters button.

Example screenshot:
[image]

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Bulgarian/Bulgaria (bg_BG)
Installed mods: 
  * Assembly4 0.50.6
  * CurvedShapes 1.0.5
  * fasteners 0.4.75
  * freecad.gears 1.0.0
  * InventorLoader 1.4.0
  * Manipulator 1.5.5
  * Pyramids-and-Polyhedrons
  * QuickMeasure 2022.10.28
  * Render 2023.8.18
  * workfeature
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-01-10:
Related?: #11766

grudir, 2024-01-11:
They are in similar areas, but the issues are separate. This one is related to Windows text scaling feature that is used for higher density displays (for example 27" display with 2560 x 1440 or 4K resolution).

closed 2024-01-23 by chennes

grudir, 2024-02-20:
Thank you for the fix @kadet1090! I checked against weekly build 36082 and the checkboxes are drawn as expected.

PR #12054 (2024-01-23): Sketcher: Apply styling to elements and constraints UI
This ensures that checkboxes drawn in the Elements UI of the sketcher can be styled using the correct style.

Three changes are introduced:
1. paint() method uses style provided by the widget, not application.
2. Indicator is drawn using `PE_IndicatorCheckBox` not `CE_CheckBox` - it allows to use `::indicator` in qss to style it properly. This works the same as TreeView.
3. Fix for minor issues with text alignment and extension of selected items background to cover checkbox. This is how other controls of this kind work.

It also completely [... cut]