FreeCAD/FreeCAD #11944: [Problem] Clipping Plane isn't accessible anymore when panel is closed

greggjaskiewicz, 2024-01-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Open model, open Clipping Plane , adjust, etc, then press X to close the panel instead of pressing "close" .
The model will stay clipped, and clipping plane won't be available again unless you restart freeCAD ! 

https://github.com/FreeCAD/FreeCAD/assets/553583/32cbe650-084a-4627-baac-f36883193a36

### Full version info

```shell
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35510 (Git)
Build type: Release
Branch: main
Hash: da05f7c8e8b25ff783cc6c4eb3b73b640f134519
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * AnimationFreeCAD 1.0.0 (Disabled)
  * A2plus 0.4.61
  * sheetmetal 0.3.13
  * freecad_metal_workbench 0.0.1
  * Curves 0.6.19
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

arossato, 2024-02-20:
I can reproduce this on linux. The same problem affects View -> Persistent section cut:

```
OS: Slackware 15.0 x86_64 (/usr/share/xsessions/xwmconfig)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36082 (Git) AppImage
Build type: Release
Branch: main
Hash: 7a025e4e60a607bdbdd1599627ce13bb8eeda071
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.7
  * A2plus 0.4.62
  * Assembly3 0.12.2
  * ExplodedAssembly
  * fasteners 0.5.10
```

wwmayer, 2024-03-03:
This  is a bug of the overlay stuff from realthunder. If it is disabled in the preferences then the dock widget behaves correctly because it doesn't show the X.

So, the problem is that this overlay mechanism bypasses too much Qt logic and then causes this kind of side-effects.

commit 1bcefaf9 (2024-03-04): GUI: Follow the widget indications when docking them (#12530)

* GUI: Follow the widget indications when docking them

The overlay manager handles the creation and drawing of the title bars
for the docked widgets. When adding a QDockWidget the manager ignored
the set of features (closable, movable, floatable) provided by the
widget and always showed all the actions.

Amend the code to take into account the features specified by the
widget.

Closes #11944
closed 2024-03-04 by chennes