FreeCAD/FreeCAD #11951: Null Geometry in isBsplinePole

GitHub4MillBill, 2024-01-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Sketch, selected angle constrain and get the exception message. Was working on a Triangle Polygon. Bounding circle deleted.  Horizontal axis corner was filleted to Radius 8mm. trying to angle constrain line to fillet to 15 degrees relative to horizontal axis. As soon as Angle constrain is selected, the exception happens.
[image]

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Anything else?

[Null geometry in isBsplinePole.zip](https://github.com/FreeCAD/FreeCAD/files/13895491/Null.geometry.in.isBsplinePole.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-01-11:
I can't reproduce this. Can you retest with the current dev version?
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35554 (Git)
Build type: Release
Branch: main
Hash: 8e2ab9f76888ac4da971f9d5069cec7bcc2b572f
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

luzpaz, 2024-02-01:
@GitHub4MillBill :point_up_2:

maxwxyz, 2024-06-13:
I'll close this. I cannot reproduce and OP is not responsive

closed 2024-06-13 by maxwxyz