FreeCAD/FreeCAD #11953: [Problem] [Feature] Respect Clipping Plane of a view when using 'Insert View' from TechDraw workbench

akrpic77, 2024-01-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I want to limit view that is inserted using 'Insert View' from TechDraw workbench. Actual 3d view can be clipped using Clipping Plane, but 'Insert View' doesn't respect that clipping.

Is this intentional, or can it be made to respect currently applied clipping ?

I am trying to generate techdraw views of a front and back panel of an enclosure. But through holes in the front panel one can see holes in the back panel, and this is also captured by 'Insert View' (see attached picture). By placing clipping plane in a way to split enclosure to front path (and back part with flip) I get nice 3d view, but, as written above, 'Insert View' doesn't respect the clipping.

Or can this be solved in some other way?

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (KDE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35554 (Git) AppImage
Build type: Release
Branch: main
Hash: 8e2ab9f76888ac4da971f9d5069cec7bcc2b572f
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * sheetmetal 0.2.63
  * fasteners 0.4.56
  * ExplodedAssembly
  * lattice2 1.0.0
  * Manipulator 1.5.0
  * kicadStepUpMod 10.19.4
  * ModernUI 0.4.56
```
```

### Subproject(s) affected?

None

### Anything else?

![image](https://github.com/FreeCAD/FreeCAD/assets/894186/42259c35-35fb-4270-9817-b16e96d29d52)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-01-11:
You can use a [Part_Cut](https://wiki.freecad.org/Part_Cut).

akrpic77, 2024-01-16:
While these two work-around could probably work on simpler object, they break down in my case as this enclosure is more complicated due to holes being done via shape binders bind to other object etc... 

but plane clipping works on the view very fast, without additional shapes (both work-around require or produce additional artifacts). This issue got marked as "Feature", although it could be Bug as well, as "Insert View" doesn't actually insert the _current_ view.

akrpic77, 2024-01-16:
I solved it by first making copy of object with Simple Copy, and then using Part Cut.

WandererFan, 2024-03-06:
If I understand the clipping plane correctly, it is not persistent.  So we could use the clipping plane only to set up an initial view.  If the clipping plane dialog is turned off later, should the view recognize that and display the whole piece or should it continue to display the cut piece?

I don't use clipping plane so I don't easily see how this is to work.

WandererFan, 2024-03-06:
- I solved it by first making copy of object with Simple Copy, and then using Part Cut.

This is how it would be done in a new view creation tool.

akrpic77, 2024-03-06:
Yes, I later realized, as it is not persistent, it's not right thing. Is this "view creation" tool in the making? 

It would be nice if we can link it (forum, issue, pull request) from here, and then this issue can be closed.

WandererFan, 2024-06-10:
Since the clipping plane is not persistent, we can't really implement this as requested.