FreeCAD/FreeCAD #11957: [Problem] Reverses the direction of sketch elements.

kaiwas, 2024-01-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

At the moment there is a problem when constructing walls, when the lines of one sketch are drawn in different directions.
I understand that you need to do everything correctly at once, but if there was a tool that changed the direction of the selected lines, this would allow you to work more freely.
For an analogy, I give an example of how a similar tool from the Corel Draw program works. Where you can change direction with one button.
Is it possible to implement something similar for freecad? Maybe this would be useful not only for sketches, but also for adding lines to the draft.
[image]
[image]
[image]

### Full version info

```shell
[code]
OS: Windows 7 build 7601  ///
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * BIM 2021.12.0
  * CADExchanger
  * CurvedShapes 1.0.5
  * Curves 0.6.21
  * DynamicData 2.60.0
  * frame 0.1.0
  * FreeCAD_Info
  * IconThemes
  * Manipulator 1.5.6
  * OpenDark 2023.12.17
  * PieMenu 1.2.7 (Disabled)
  * PitchedRoof
  * QuickMeasure 2022.10.28
  * Reinforcement
[/code]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-03-30:
Related forum discussion as an example of the lack of this feature causing confusion to newcomers: https://forum.freecad.org/viewtopic.php?t=86546

paullee0, 2024-04-06:
There could be different approaches to this situation, for discussion:

1. Tool to reverse the direction of the edge by a simple click
2. Tool to make the particular Edge Align to be Center, Left, or Right

The 2nd one is already available.

kaiwas, 2024-04-15:
> The 2nd one is already available.

If you are talking about “Overright Align”, then this works poorly, because there is no way to quickly identify the desired edge (if there are many of them).

In addition, creating a list for alignment is a separate difficult task (white text on a yellow background. Capital letters are required. And you can’t make a mistake in a single character or accidentally put a space)
[image]

paullee0, 2024-04-16:
> If you are talking about “Override Align”, then this works poorly, because there is no way to quickly identify the desired edge (if there are many of them).
> 
> In addition, creating a list for alignment is a separate difficult task (white text on a yellow background. Capital letters are required. And you can’t make a mistake in a single character or accidentally put a space) 

Indeed, there are a lot to do :) 

1. There is a simple tool in the experimental Add-on user just need to click on the edge so that it toggle between Left - Center - Right : no need to input by typing.
2. Visualisation is a thing that needs to help user to identify the exact Align.  This https://forum.freecad.org/viewtopic.php?p=750643&sid=c9a6a1b743d5de48573285c07b57e7c7#p750643 is one suggestion.  Asked on forum the code how to add some visual clue but no response yet.  Anyone would like to try?
3. And this Add-on had overcome the 'topological-naming' problem, I am trying to expose this feature on all tools (available for automatic placement only at the moment).
4. With the tool and ability to switch Align of each edge / wall segment, does it need another tool to switch the direction of the sketch's edge?  I am not sure.  If yes, could try implement this.

Hope there are more advanced developers to help :D

kaiwas, 2024-04-16:
I know about the “align” option (for the entire wall as a whole).
But sometimes you need to change the alignment for a separate fragment (that’s why I did it through a list).
I just wanted to say that there is no way to identify which list line will affect a specific wall fragment. (this is already a topic not related to this topic on GitHub).

This is the situation. What we can do is apply the alignment to the entire wall as a whole, but change the direction of the sketch.
or we can make a list (difficult because there are no identifiers). 
Example in the picture.
[image]

maxwxyz, 2024-04-16:
Related:
- https://github.com/FreeCAD/FreeCAD/issues/12436

paullee0, 2024-04-18:
> When I started mastering freecad, I began to greatly regret that I did not have programming skills. I would really like to make changes, but I can only “identify” shortcomings (((

I am not trained as a programmer either, just picked up basic python and FreeCAD knowledge and had been able to make some simple fixes and tools hope it would suit architectural practitioner.

The simple tool to edit the Align of each individual wall segment / edges is demonstrated in some youtube video, like this recent one; welcome for comment and idea to improve :)

[ArchSketch + ArchWall - - - Edit Align Demo](https://www.youtube.com/watch?v=eDFVWo0-jes)

paullee0, 2024-04-18:
> Related:
> 
>     * [Modify custom vertex mapping of sketches for loft commands #12436](https://github.com/FreeCAD/FreeCAD/issues/12436)

Thanks for the link.  'Technically', it is something different from what ArchWall's algorithm; in other words, any tool to fix/improve workflow in either workbench would not help the other one regrettably in this case :)

paullee0, 2024-07-20:
This should be fixed (if not earlier) in https://forum.freecad.org/viewtopic.php?p=771954#p771954, can anybody further test other similar case?  Thanks :)

furgo16, 2024-07-20:
@paullee0 thanks. Is there a specific PR that fixes the bug? From the forum post, I can see https://github.com/FreeCAD/FreeCAD/pull/15530, but that one is still open.

paullee0, 2024-07-20:
> @paullee0 thanks. Is there a specific PR that fixes the bug? From the forum post, I can see #15530, but that one is still open.

There seems to be a few issues and forum threads on similar problem.  I follow another one and submit #15530 ; I have a quick look and that should fix this one.

And yes, that one is yet to be reviewed and accepted ?currently.

paullee0, 2024-12-19:
Can use your images below to put in wiki of ArchWall?  

Thanks.

>  ![image744](https://private-user-images.githubusercontent.com/32273833/296037345-3b8dab6b-9591-4d5b-b1cc-030d351528b7.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ1Njc3MTAsIm5iZiI6MTczNDU2NzQxMCwicGF0aCI6Ii8zMjI3MzgzMy8yOTYwMzczNDUtM2I4ZGFiNmItOTU5MS00ZDViLWIxY2MtMDMwZDM1MTUyOGI3LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjE5VDAwMTY1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJmMDNhZDRkZWQyNzcyODZhYjA4NjA3OGNhZWY0MmY5MDZmYjJjZTRkMmU2MjE4ZTAyYTZjMDViODc2YTlmMTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.YBE7RaU1_Rydt7XnoAamaNfbsO0Fb7qPQBFwVla569E) 

![2024-01-11 22_19_05](https://private-user-images.githubusercontent.com/32273833/296037381-596bd007-d289-49c6-86c2-86afbf6dce90.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ1Njc3MTAsIm5iZiI6MTczNDU2NzQxMCwicGF0aCI6Ii8zMjI3MzgzMy8yOTYwMzczODEtNTk2YmQwMDctZDI4OS00OWM2LTg2YzItODZhZmJmNmRjZTkwLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjE5VDAwMTY1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVhYTIxYmQwZDM3NzU4NzI4ZjgwNDZlYjZkNjI3NmM0ZjU2YzdmYTNkODMyMjJjZGM2Njg0NWI5ZTQ0MmIwZjUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.UyAp2k0ulY5VMVbHmZHghcGyQIW3P2wGe15T6HHhdYQ)

kaiwas, 2024-12-19:
> Can use your images below to put in wiki of ArchWall?

If you are satisfied with the quality, then of course yes.

kaiwas, 2024-12-19:
@paullee0 It will probably look more presentable this way.
[image]

paullee0, 2024-12-19:
> @paullee0 It will probably look more presentable this way. 

Thanks :)

psommerfeld, 2025-02-20:
Can we add a note to the tooltip of Wall:Align property?
Even reading the Arch_Wall wiki, I would not have caught that direction matters with having no prior familiarity with BIM Wall.
In the tooltip, something like, "NOTE: Wall alignment is dependent on the direction that the dependent lines were drawn" or something like that. It would have saved me a lot of searching around and confusion.

psommerfeld, 2025-02-21:
I would say that what is already in the tooltip plus just one line mentioning that the base object's line direction affects left/right alignment, would be sufficient. One extra line like:
NOTE: Left/right wall alignment will depend on which direction each line in the base object was drawn.
would be my suggestion.

> It is a bit long in the wiki, any suggestion to the whole tooltip that should be shown?

furgo16, 2025-02-21:
@psommerfeld  @paullee0 reworded the `Alignment` property as per the feedback => https://wiki.freecad.org/Arch_Wall#Data

Let me know if it looks ok to you.

furgo16, 2025-02-21:
Oh, are we getting GitHub notifications every time we edit the wiki now? Where does this come from? ^^

kaiwas, 2025-02-21:
> Let me know if it looks ok to you.

Maybe this picture should be moved?

[image]

furgo16, 2025-02-21:
I thought so as well, but I'm not sure if it would break the wiki formatting. I'm not a wiki expert, I hadn't done any Mediawiki markup for years before starting to contribute to FreeCAD about a year ago :-)

kaiwas, 2025-02-21:
> break the wiki formatting

You can write a phrase in brackets, in the block about alignment, for example (See the image below).

Reqrefusion, 2025-02-21:
> Oh, are we getting GitHub notifications every time we edit the wiki now? Where does this come from? ^^

One of the extra features of the my project is that I try to create a synchronized version of the wiki on Github. It works when you github quote something in the wiki description section. I thought it was a pretty cool thing to do. What's your opinion?

psommerfeld, 2025-02-22:
I would agree to move the picture up as well though.

> Psommerfeld suggested to update the tooltip in the FreeCAD command though :)

paullee0, 2025-02-22:
I just login the wiki and found Roy_043 had moved the diagram to the top :)