FreeCAD/FreeCAD #11958: New sketch in a link does not open at correct location.

ppemawm, 2024-01-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

We seem to have lost the ability to assemble (Assembly4) an empty body to another body LCS, select the link, add a PartDesign sketch, and begin design at that LCS origin. The sketch incorrectly opens at the global or assembly origin. It has to be closed and re-opened for it to properly locate to its assembled LCS location.  

Steps to reproduce the bug using Assembly4:

1. Create an empty Assembly4 Body
2. Insert the body into the assembly and attach it another body using an LCS not located at the assembly origin.
3. Double click the body link which takes you to PartDesign and add a sketch.
4. It does not open at the proper LCS location. It opens at the assembly or global origin.
5. Close the sketch. It jumps to the intended LCS location! Now, you can reopen the sketch and proceed with in-context design work.

You get the same unwanted behavior if you create a link to an empty body in PartDesign, change its placement properties, and add a sketch. The sketch opens at the global origin rather than the body origin. If you close and reopen it is located properly.

According to @Syres initial investigation [git commit a3c4f4b](http://github.com/FreeCAD/FreeCAD/commit/a3c4f4b) does not appear to handle Links.

### Full version info

```shell
Works as expected before V0.21.2.33771 but not after
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.6
  * fasteners 0.4.15
  * freecad.gears 1.1.0
  * QuickMeasure 2022.10.28
[/code]

OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35510 (Git)
Build type: Release
Branch: main
Hash: da05f7c8e8b25ff783cc6c4eb3b73b640f134519
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.6
  * fasteners 0.4.15
  * freecad.gears 1.1.0
  * QuickMeasure 2022.10.28
```
```

### Subproject(s) affected?

Sketcher

### Anything else?

Please see forum discussion here:  https://forum.freecad.org/viewtopic.php?p=731682#p731682 and 
https://forum.freecad.org/viewtopic.php?p=731723#p731723

Select the link in the attached file and delete the sketch in the link.  Add a new sketch in the link and open it.  Close and reopen.
[link_sketch-bug.zip](https://github.com/FreeCAD/FreeCAD/files/13908002/link_sketch-bug.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-02-11:
> According to @syres initial investigation [git commit a3c4f4b](https://github.com/FreeCAD/FreeCAD/commit/a3c4f4b) does not appear to handle Links.

@wmayer seems it was code you committed :point_up_2:

ppemawm, 2024-04-24:
Any progress on this bug?  This hamstrings in-context modelling and especially the assembly design work process.  Or, is that the intent until the new assembly workbench is better established?

3x380V, 2024-05-10:
@ppemawm, since Assembly4 is not working with current version of FreeCAD, this bug will hardly see any progress. I just tried again [link_sketch-bug.zip](https://github.com/FreeCAD/FreeCAD/files/13908002/link_sketch-bug.zip) with current FreeCAD at commit 7d738f161a64b1f and my [Assembly4 fork](https://github.com/3x380V/FreeCAD_Assembly4/tree/devel) and could not reproduce it. Perhaps I'm doing something wrong, so I would appreciate if you could verify that.

ppemawm, 2024-05-11:
@3x380V @luzpaz
Please note that this is not particular to Assembly4:

_"You get the same unwanted behavior if you create a link to an empty body in PartDesign, change its placement properties, and add a sketch. The sketch opens at the global origin rather than the body origin."_

Apparently, no one is doing in-context modelling and assembly design or we would hear more complaints.  I suspect it will not be fixed until someone needs that capability with the new assembly workbench in progress.

I am at V0.21.1 which works well enough for all my needs.  I  have no incentive to update to the development version so I have no idea if this bug has been taken care of.

3x380V, 2024-05-11:
@ppemawm, I cannot reproduce that one either. Sketch opens at Body origin.

ppemawm, 2024-05-11:
@3x380V As a courtesy I tried the same procedure in this development version with the same undesired result.
Did you change the placement of the body link before adding the sketch?  Did you activate the link before adding the sketch?

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37249 (Git)
Build type: Release
Branch: main
Hash: 3acbc561f299b4cddea8215c387a52f3d5132cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.12
  * DynamicData 2.60.0
  * fasteners 0.5.15
  * freecad.gears 1.2.0
  * MnesarcoUtils 0.2.5
  * QuickMeasure 2022.10.28

3x380V, 2024-05-11:
@ppemawm, I see, _activating link_ is what breaks it. It is even written in the code...

3x380V, 2024-05-17:
That suspicious commit noted in the issue description is misleading to say the least. With 
```
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.20.2.
Build type: Release
Python 3.11.1, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
Installed mods: 
  * fasteners 0.5.20
  * Assembly4 0.50.12
  * sheetmetal 0.4.13
```
and my previous example you get after creating sketch:
```
Gui.runCommand('PartDesign_NewSketch',0)
Gui.Selection.clearSelection()
Gui.Selection.addSelection('LinkTest','Link','Origin001.XZ_Plane001.')
```
while
```
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
Installed mods: 
  * fasteners 0.5.20
  * Assembly4 0.50.12
  * sheetmetal 0.4.13
```
gives you
```
Gui.runCommand('PartDesign_NewSketch',0)
Gui.Selection.clearSelection()
Gui.Selection.addSelection('LinkTest','Body001','Origin001.XZ_Plane001.')
```
so you are in the trouble much earlier than @wwmayer's change hits the code path.

ppemawm, 2024-05-17:
@3x380V Thanks for continuing to look into this.

3x380V, 2024-05-22:
https://github.com/FreeCAD/FreeCAD/blob/ff40ef3586bfa041e01d0d9e3ad176382f4ccfc5/src/Mod/PartDesign/Gui/Utils.cpp#L92 is calling
https://github.com/FreeCAD/FreeCAD/blob/ff40ef3586bfa041e01d0d9e3ad176382f4ccfc5/src/Gui/ActiveObjectList.cpp#L40-L58 with `resolve = true`, thus _Link_ is resolved to a _Body_ it is pointing to. This was not a problem prior #9557 as _Link_ itself was returned in `parent` parameter of `getActiveObject`, so a minimal fix would look like
```
--- a/src/Mod/PartDesign/Gui/Utils.cpp
+++ b/src/Mod/PartDesign/Gui/Utils.cpp
@@ -89,12 +89,12 @@ bool setEdit(App::DocumentObject *obj, PartDesign::Body *body) {
         return false;
     App::DocumentObject *parent = nullptr;
     std::string subname;
-    auto activeBody = activeView->getActiveObject<PartDesign::Body*>(PDBODYKEY);
+    auto activeBody = activeView->getActiveObject<PartDesign::Body*>(PDBODYKEY, &parent);
     if (activeBody != body) {
         parent = obj;
     }
     else {
-        parent = getParent(obj, subname);
+        getParent(obj, subname);
     }
```
But given the amount of guesses around previous code change in this function and especially @0penBrain's https://github.com/FreeCAD/FreeCAD/pull/9863#issuecomment-1624215138 which also suggests that #9557 only revealed some other problem, I have no clue if above change does not cause some other damage.

3x380V, 2024-05-23:
...above does not fix original problem in [link_sketch-bug.zip](https://github.com/FreeCAD/FreeCAD/files/13908002/link_sketch-bug.zip)
`<Gui> Document.cpp(386): Invalid sub object 'link_sketch_bug#Assembly.Sketch002.'`
While reason is obvious, solution is not ;-)

wwmayer, 2024-06-19:
It was confirmed that v0.21 works as expected but my commit (of https://github.com/FreeCAD/FreeCAD/pull/9557)  you blamed is part of it. It was merged at the beginning of June 2023 and the release of v0.21 was end of July 2023. You can see that https://github.com/FreeCAD/FreeCAD/blob/releases/FreeCAD-0-21/src/Mod/PartDesign/Gui/Utils.cpp includes the changes of the PR.

So, I wonder how it can be the culprit or part of it.

3x380V, 2024-06-19:
@wwmayer, I thought we already proved your commit just revealed other problems, although it is the one after described behavior began to manifest itself. Now the real question is if those who have deeper insight into this code agree that [link_sketch-bug.zip](https://github.com/FreeCAD/FreeCAD/files/13908002/link_sketch-bug.zip) indeed shows a bug and whenever anyone is able to fix is. My knowledge is clearly insufficient here.

maxwxyz, 2024-08-05:
what is the progress on this issue?

ppemawm, 2024-08-07:
@maxwxyz @3x380V 
IMHO this regression since 0.21.1 should be labeled a blocker to 1.0 since it greatly hinders basic assembly design by making in-context modelling more difficult.

maxwxyz, 2024-08-17:
@3x380V could you weigh in if this is a feasible fix for the 1.0 release?

3x380V, 2024-08-21:
@maxwxyz, if _feasible_ means I can fix it before release, I'm not really sure. It requires pretty deep understanding of surrounding code, which means a few days of focused reading. And I hadn't accumulated such amount of continuous spare time since this issue popped up.
I would appreciate if someone working in this area could push me the right direction, then I can attempt to cook some solution for review.

maxwxyz, 2024-08-21:
@PaddleStroke mabe you could help with the experience from assembly/sketcher?

ppemawm, 2024-08-22:
@maxwxyz @3x380V @PaddleStroke @Zolko-123 

Perhaps @Zolko-123 can weigh in with his valued insight as well.

It would be interesting to hear from @PaddleStroke how the integrated assembly workbench handles critical assembly design and in-context modelling requirements without encountering this bug.

There is a rather awkward workaround that may add some clue as to what is the problem.  Instead of assembling an empty body, if you first create a "dummy" **feature**, when the body is assembled (e.g. using Assembly4 default assembly LCS), it and the feature sketch are located at the correct LCS joint location.

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38429 (Git)
Build type: Release
Branch: main
Hash: 362cf1c80edb89957d81c0dbe6e50ce35daaa6db
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods:
  * Assembly4 0.50.15
  * DynamicData 2.60.0
  * fasteners 0.5.15
  * freecad.gears 1.2.0
  * MnesarcoUtils 0.2.5
  * QuickMeasure 2022.10.28
  
I have not tried it with the latest 0.22 weekly nor with the integrated assembly workbench.

The same workaround can be used if you make a link of a body that has a dummy feature, change its placement, and then add a sketch at its origin.  It is correctly located.

maxwxyz, 2024-09-13:
@chennes any idea who could help on this issue?

chennes, 2024-09-13:
It basically already pings all of the "usual suspects" -- we could branch out and see if @bgbsww has any insight.

bgbsww, 2024-09-13:
I'm just confused.  Opening link_sketch-bug gives me:

```
...
15:51:22  <App> PropertyTopoShape.cpp(394): Pending recompute for generating element map: link_sketch_bug#Sketch002
15:51:22  pyException: Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ModuleNotFoundError'>: No module named 'Asm4_libs'

```  in
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38644 (Git)
Build type: Debug
Branch: bgbsww-Multithreading
Hash: 7ed1e9380acccd686a0e3573f883ac793aec3299
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * sheetmetal 0.4.25
  * Curves 0.6.47
  * DynamicData 2.62.0
  * BIM021 2021.12.0
  * parts_library

Is Assembly4 the assembly workbench built into 1.0?  Should I be getting this module issue? Is there suppose to be an Assembly4 in the Addons (there isn't for me)?  In my installed Mods?

I need an ELI5 set of instructions and assertion of what is wrong to be able to help, unfortunately I just don't get what's going on here.

3x380V, 2024-09-13:
@bgbsww, you do not need Assembly4 installed, just follow https://github.com/FreeCAD/FreeCAD/issues/11958#issuecomment-2106006688
In case anything is still unclear, feel free to ask. Thank you.

bgbsww, 2024-09-13:
Thanks!  Sometimes I can be dense.  Okay, I can replicate, kinda:

In my own debug builds, I see exactly the problem.   However, in the daily I get a completely different thing:

17:25:52  Cannot create object 'Link': (Cannot add static property)

And there is no link in the drawing!

Code shows difference in debug mode (code is unreachable there):

```
void PropertyData::addProperty(OffsetBase offsetBase,const char* PropName, Property *Prop, const char* PropertyGroup , PropertyType Type, const char* PropertyDocu)
{
#ifdef FC_DEBUG
    if(!parentMerged)
#endif
    {
        short offset = offsetBase.getOffsetTo(Prop);
        if(offset < 0)
            throw Base::RuntimeError("Invalid static property");
        auto &index = propertyData.get<1>();
        auto it = index.find(PropName);
        if(it == index.end()) {
            if(parentMerged)
                throw Base::RuntimeError("Cannot add static property");
            index.emplace(PropName, PropertyGroup, PropertyDocu, offset, Type);
        } else{
#ifdef FC_DEBUG
            if(it->Offset != offset) {
                FC_ERR("Duplicate property '" << PropName << "'");
            }
#endif
        }
    }

```

Daily:
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38702 (Git)
Build type: Release
Branch: main
Hash: 77b3311f574dbc985da8ec6a62e6277ca90d4651
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * sheetmetal 0.4.25
  * Curves 0.6.47
  * DynamicData 2.62.0
  * BIM021 2021.12.0
  * parts_library

Now adding some debug and investigating further...

ppemawm, 2024-09-13:
@bgbsww 
Please be advised that this regression is NOT peculiar to Asembly4.  It is a general problem of adding a sketch to a body link after its placement has been changed.:
"You get the same unwanted behavior if you create a link to an empty body in PartDesign, change its placement properties, and add a sketch. The sketch opens at the global origin rather than the body origin. If you close and reopen it is located properly."

bgbsww, 2024-09-14:
Okay, so when you are in the initial "Select attachment" dialog provided by TaskFeaturePick.cpp and you choose a plane, you get something like 'LinkT001#Body001.(XZ-plane001)`  which is clearly the original object and not the link.  That drives the display, what happens in the sketch edit dialog etc.   And it makes sense _because there is no sketch object in the model hierarchy to find at this point during a select_  After you first close the sketch, it is created in the model, and exists both in the BodyAtLeft at the left side origin, and in the link at BodyLinkAtRight at the right side origin, and all is fine.

So what we are trying to deal with here is a situation where the tip points to a link, but the select searches default to matching preview geo that is in the link target.

That leads me to question how the selection is working in TaskFeaturePick::onItemSelectionChanged when we call Gui::Selection().addSelection ...

SIDE RANT:  OPAQUE QSTRINGS IN THE DEBUGGER.  ARGH!!!!

...  So we're trying to find XZ_Plane001  inside of the "LinkTest" document.  And our feature is "Body001" not "Link"  And our SubName is "Origin001.XZ_Plane001" [correct]

That had to be setup by SelectionSingleton::checkSelection getting it from Document::getObject("XZ_Plane001").

And that just summarily chooses the first object that find() locates in the document, doesn't care about the rest of the iterator..   But that doesn't matter because there's only one in the document.

So the question here is when and where to have extra logic to see if we are referring to an object that is in an origin that appears in a link.   I got nothing here - no idea of the conditions.

More coming ...

bgbsww, 2024-09-14:
SketchWorkflow::tryFindBasePlanes from SketchWorkflow::findAndSelectPlanes.

`App::Origin *OriginGroupExtension::getOrigin () const {
    App::DocumentObject *originObj = Origin.getValue ();`

Ultimately, I think  '''    else {
        SketchRequestSelection requestSelection{guidocument, activeBody};
        requestSelection.findSupport();
    }
'''  inside of 'SketchWorkflow::createSketchWithModernWorkflow(' is not aware of the link.  Why?  Hopefully one of you is an expert here.

chennes, 2024-09-16:
@FreeCAD/cad-advisory-group do you have any opinion about our support of this workflow? The devs at the review meeting this morning don't have a sense for what the path forward on this is.

ppemawm, 2024-09-16:
The Integrated Assembly Workbench has yet to add the "hook" or means for top down assembly design and in-context modeling, but when they do it is likely that this regression will become a serious issue for that workbench as well.

bgbsww, 2024-09-17:
Okay, more brute force looking:  If you are here:

https://github.com/bgbsww/FreeCAD/blob/b4923d1c584a4dce794d91a4b090a5293569223a/src/Mod/PartDesign/Gui/SketchWorkflow.cpp#L529-L540

You can see that we are getting the planes to populate the sketcher selector from [this.]activeBody.   But in the failure case here, we presumable want the planes from the link object.  Unfortunately, there doesn't seem to be anything in activeBody, guidocument, appdocument, PartDesignGui:: or anywhere that allows me to get the Link instead of Body001.  Given that it is shown as highlit in the tree after activation, there must be something that knows this that we can read from.  Anyone got a hint?

bgbsww, 2024-09-17:
A:  You provide a parent DocumentObject parameter to getObject.  It comes back null except for when it comes back with the link.  Logic should then be to apply the placement from that parent, I think.   Implementing, testing ...

3x380V, 2024-09-17:
@bgbsww, btw, tried something similar in https://github.com/FreeCAD/FreeCAD/issues/11958#issuecomment-2125254112 and while if fixed [LinkTest](https://github.com/FreeCAD/FreeCAD/files/15285321/LinkTest.zip) case, the original one reported by @ppemawm did not and I'll happily test your fix with Asm4.

bgbsww, 2024-09-17:
@3x380V https://github.com/bgbsww/FreeCAD/tree/bgbsww-SketchCorrectLocation

3x380V, 2024-09-17:
I'm a bit confused by `pdBody and topParent->isLink()` so I made it `&&`, now building. As I switched a branch, it would take about half an hour.

bgbsww, 2024-09-17:
That's a context switching from Python issue.  However, apparently it is now legit:  https://stackoverflow.com/questions/47174920/difference-between-and-and-operators

3x380V, 2024-09-17:
For a reference, this is a patch used for testing: https://github.com/3x380V/FreeCAD/commit/5aed4e848a1194852ac0a9c868494e7aa633a590 (apart from what you fixed already there are some unimportant nits polished)
Both problem seems fixed. Unfortunately I cannot provide binary for @ppemawm as he seems to be on Windows, but what else can be done now, than opening a PR? Thank you, analysis you've done would take me ages.

bgbsww, 2024-09-17:
Thanks for testing, I will revise branch and make a PR.  And go party.  This one was **tough.**

ppemawm, 2024-09-17:
@bgbsww @3x380V 
>... I will revise branch and make a PR. And go party. This one was **tough.**

Thanks to all for helping to resolve this.  I look forward to testing the fix once available for Windows.  Party on.

3x380V, 2024-09-18:
@adrianinsaval, as CI is running on all platforms supported, how hard would to make build results downloadable by general public? Probably not each and every build, but controlled by some extra commit in PR of choice for example.
Sometimes people able to test issue are running FreeCAD on another platform and do not build binaries themselves.
(I'll open issue for that if it is not completely stupid idea, which could be expressed by a single simple answer :))

adrianinsaval, 2024-09-18:
from what I'm reading [here](https://docs.github.com/en/actions/writing-workflows/choosing-what-your-workflow-does/storing-and-sharing-data-from-a-workflow) it looks like storage space for artifacts on github is free for public repos like this, so it might actually be doable

PR #16606 (2024-09-18): PartDesign: When creating Sketch inside a Body in a link, use link position
Fix #11958
closed 2024-09-18 by chennes