FreeCAD/FreeCAD #11961: [Problem] 0.22dev - regression with the additive pipe tool

phtlnx, 2024-01-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello,

There is a regression with the additive pipe tool.
When an additive pipe is applied, the reference planes are rotated.
The result after applying the tool, left version 0.21 and right 0.22dev the result rotates around the X axis.
[image]

### Full version info

```shell
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35554 (Git) AppImage
Build type: Release
Branch: main
Hash: 8e2ab9f76888ac4da971f9d5069cec7bcc2b572f
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * freecad_metal_workbench 0.0.1
  * CurvedShapes 1.0.5
  * dodo 1.0.0
  * sheetmetal 0.3.13
  * Curves 0.6.19
  * Pyramids-and-Polyhedrons
  * ExtremeProDark 2.9.0
  * A2plus 0.4.61
  * fasteners 0.4.75
  * 3D_Printing_Tools
```

### Subproject(s) affected?

PartDesign

### Anything else?

[exemple.zip](https://github.com/FreeCAD/FreeCAD/files/13916527/exemple.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-01-12:
Git Bisect in progress

Syres916, 2024-01-12:
@bgbsww I'm sorry to raise this but https://github.com/FreeCAD/FreeCAD/commit/f2d5e07b4ead2209544f80a9f5f1872e7882fc04 is causing IMHO a more serious regression than the bug that it was intended to fix. It's applying a rotation of 90 degrees on the X axis to perfectly sound model. Please let me know if you want me to raise the PR to revert it as I think this should be done fairly quickly.

bgbsww, 2024-01-12:
@Syres916  Wow.  I agree, that can't stand, and while it's easy to replicate, it isn't clear why that change is causing the error.

So, reverting is in order, and then determining another path to fix it.

closed 2024-01-13 by chennes