FreeCAD/FreeCAD #11965: [Problem] no proper `PropertyExpressionEngine.execute()` for cross-property references

Andrei-Pozolotin, 2024-01-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Recently resolved issue #11825 with PR #11937 removed previously occurring spurious exceptions,
and exposed next-level problem:  

PROBLEM:
in a use case when one property has a simple expression referencing another property (see  #11825)
the `value` part of the source quantity is propagated as-is to the target,  without any regard to the `unit` information

specifically, as code tracing shows, that when  `PropertyExpressionEngine::execute()` gets to processes 
the expression constructed elsewhere by the `ExpressionParser::parse()`, 
https://github.com/FreeCAD/FreeCAD/blob/3ba0c3d7958dad891e125e065a41d95faac47c6e/src/App/PropertyExpressionEngine.cpp#L607

the type of the expression corresponding to the `source property`  here `std::shared_ptr<App::Expression>`:
https://github.com/FreeCAD/FreeCAD/blob/3ba0c3d7958dad891e125e065a41d95faac47c6e/src/App/PropertyExpressionEngine.cpp#L677

is actually `VariableExpression`
https://github.com/FreeCAD/FreeCAD/blob/3ba0c3d7958dad891e125e065a41d95faac47c6e/src/App/Expression.cpp#L2896

and not the `UnitExpression` as one would expect
https://github.com/FreeCAD/FreeCAD/blob/3ba0c3d7958dad891e125e065a41d95faac47c6e/src/App/Expression.cpp#L1222

which explains the lack of `value/unit` information conversion

Here is a proposed test case, which is currently failing:
https://github.com/FreeCAD/FreeCAD/compare/main...random-cadder:FreeCAD:issue_11825_A

SOLUTION:
it is not clear how to proceed:
* should this be resolved in `ExpressionParser::parse()` ?
* should this be handled in the `PropertyExpressionEngine::execute()`?
* is there perhaps an expression language syntax with meaning: `TreatMeAsUnitExpression(<<10 mm>>)` ?

@chennes  : please advise

### Full version info

```shell
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35567 (Git)
Build type: None
Branch: makepkg
Hash: 6a54591bac6c8843f12c6971623437083d017ab3
Python 3.11.6, Qt 5.15.11, Coin 4.0.1, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2024-01-12:
> it is not clear how to proceed:
* should this be resolved in ExpressionParser::parse() ?
* should this be handled in the PropertyExpressionEngine::execute()?
* is there perhaps an expression language syntax with meaning: TreatMeAsUnitExpression(<<10 mm>>) ?

Probably better answered by @WandererFan or @yorikvanhavre than by me.

Andrei-Pozolotin, 2024-01-13:
SOLUTION A: PR #11971

this solution is based on two ideas:
* use magic reference syntax: `prop_name + 0mm`
* add `Quantity::parse()`  to expression evaluation path

why this solution works:
* the magic reference syntax: `prop_name + 0mm` forces expression type to be `UnitExpression`
* and  `QuantityPyImp::pyToQuantity()` is using `Quantity::parse()` similar to existing approach in:
https://github.com/FreeCAD/FreeCAD/blob/5ee16df6b74f55e88f9f00747a6c096f7f60e1a8/src/App/PropertyUnits.cpp#L64

@WandererFan  @yorikvanhavre @chennes  : please review

Andrei-Pozolotin, 2024-02-01:
SOLUTION B: PR https://github.com/FreeCAD/FreeCAD/pull/12225

this solution is based on this idea:
* add `parsequant()` function to the expression language

commit c5ad7b50 (2024-02-19): Resolve #11965 - no proper execute() for cross-property references

Solution B: parsequant() function
closed 2024-02-19 by chennes