FreeCAD/FreeCAD #11970: [Problem] Return values of FreeCAD.get*Dir and FreeCAD.get*Path functions have inconsistent path separators

Roy-043, 2024-01-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Return values of FreeCAD.get*Dir and FreeCAD.get*Path functions have inconsistent path separators. On Windows some use a mix of forward and backslashes. Some do not end in a path separator. The path separators in the return value of `getUserMacroDir` depend on the optional argument.

```
FreeCAD.getHelpDir()
# => 'D:/Tmp/FreeCAD_0.22/doc/'
# Is this still useful?

FreeCAD.getHomePath()
# => 'D:/Tmp/FreeCAD_0.22/'

FreeCAD.getLibraryDir()
# => 'C:/bld/freecad_1704741003218/_h_env/Library/lib'
# Not sure what this is for.

FreeCAD.getResourceDir()
# => 'D:/Tmp/FreeCAD_0.22/data/'

FreeCAD.getTempPath()
# => 'C:/Users/Roy/AppData/Local/Temp\\'

FreeCAD.getUserAppDataDir()
# => 'C:\\Users\\Roy\\AppData\\Roaming\\FreeCAD\\'

FreeCAD.getUserCachePath()
# => 'C:/Users/Roy/AppData/Local/cache\\FreeCAD\\Cache\\'

FreeCAD.getUserConfigDir()
# => 'C:\\Users\\Roy\\AppData\\Roaming\\FreeCAD\\'

FreeCAD.getUserMacroDir()
# => 'C:\\Users\\Roy\\AppData\\Roaming\\FreeCAD\\Macro\\'

FreeCAD.getUserMacroDir(True)
# => 'C:/Users/Roy/AppData/Roaming/FreeCAD/Macro'
```

### Full version info

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35554 (Git)
Build type: Release
Branch: main
Hash: 8e2ab9f76888ac4da971f9d5069cec7bcc2b572f
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-03-20:
Fixed with https://github.com/FreeCAD/FreeCAD/pull/13068

Roy-043, 2024-04-26:
The following paths still have issues:

```
FreeCAD.getLibraryDir()
# => 'C:/bld/freecad_1713816304749/_h_env/Library/lib'

FreeCAD.getTempPath()
# => 'C:/Users/Roy/AppData/Local/Temp\\'

FreeCAD.getUserCachePath()
# => 'C:/Users/Roy/AppData/Local/cache\\FreeCAD\\Cache\\'

FreeCAD.getUserMacroDir(True)
# => 'C:\\Users\\Roy\\AppData\\Roaming\\FreeCAD\\Macro'
```

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36958 (Git)
Build type: Release
Branch: main
Hash: a041129090b8dcbb8367636840d3c668ef30a933
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

Roy-043, 2025-02-08:
This is still an issue.

luzpaz, 2026-10-10:
Is this related to ..?
* #18027

commit 2295cba5 (2024-03-22): Fixes #11970: Return values of FreeCAD.get*Dir and FreeCAD.get*Path functions have inconsistent path separators